pop-os/nvidia-graphics-drivers #10: Issue with Nvidia driver 420 causing monitors to go to sleep

bmarshall1985, 2019-07-18:
After the 420 update I experienced the following;
1. System would boot up, 2-3 seconds after logging in and arriving at gnome desktop all monitors would go blank and report no signal. Power cycling one of the monitors multiple times would sometimes cause all 3 to reconnect. System was running, I could connect remotely via ssh.
2. Assuming I could get the system working in point 1 if I tried to open gnome display manager all monitors again would go blank and report no signal. Power cycling multiple times again would often get it back. Gnome displays had not opened when the system came back 
3. Downloading a file in Firefox (newest version from pop shop 17 Jul), going to the downloads icon in the toolbar and clicking to open the file also causes monitors to go blank. Again power cycling multiple times can recover. 

In all these cases, while the monitors are off something is stressing the cpu as the fans spin up. Using htop via ssh I could see it was some xorg process pinning core 2 on my cpu to 100%. I cannot remember the detail of this process

Hardware config as follows
Intel core i7 8700k
ASUS Strix Z370-F D4 motherboard
48gb Corsair vengeance LPX 3200mhz ram
KFA2 GeForce 770 2gb gpu 
Triple ASUS B276 HUL 1440p monitors all connected by DisplayPort 
Samsung 960 EVO nvme 500gb m.2 ssd 
Other stuff that probably isn’t relevant but I can provide detail if needed.

Software running was pop_os 19.04 fully up to date as of the morning of July 17th ( UK time ). Some snap packages and some flatpaks were installed. 
The only non standard config I can think of that I had done was to install a non standard version of open razor for black widow elite support which has since been merged but I hadn’t switched back to the main branch yet so I was running a version from around 1 month ago 
https://github.com/openrazer/openrazer/commit/03a48901499ef922dfe9969330ca9808a76b62ab

Apologies for the vagueness, I have since reformatted the machine and reinstalled pop 19.04 to see if something I had done to the machine was the cause so I don’t have access to the problem config anymore. I will be retrying the upgrade to 420 at the weekend.

jackpot51, 2019-07-20:
I replicated the issue and reported to NVIDIA

bmarshall1985, 2019-07-21:
I took a screenshot from my phone logged in via ssh of the process that eats CPU while trying to detect monitors if it helps
[image]

jackpot51, 2019-07-23:
Does the issue go away if `/lib/modprobe.d/nvidia-graphics-drivers.conf` has the line:

```
options nvidia-drm modeset=1
```

Changed to:

```
options nvidia-drm modeset=0
```

kartikvashistha, 2019-07-25:
@jackpot51 I can confirm that I still experience the monitor sleep bug (on a fresh install) after applying the suggested fix. I can confirm the bug reported by @bmarshall1985 on firefox when clicking the download manager button. 
Another time I have noticed the display going to sleep is when trying to access system settings via the app drawer and via the gnome-shell top bar. The only way I can wake my screen up is by switching user sessions from tty2 to tty1/tty3 and back to tty2. Could gnome shell settings be somehow interfering with the NVIDIA driver?
Also, in the meantime is it possible to rollback to a previous working version of the NVIDIA driver? Thanks!
System specs:
i7 6700k, NVIDIA GTX 1080, 16GB RAM, POP!_OS 19.04 kernel 5.0.0-21 with all the latest updates as of 25th July 2019 BST.

ArnoldDeRuiter, 2019-07-26:
Changing the  nvidia-drm modeset value to 0 didn't work for my GTX1080 setup.

Is there a way to downgrade the driver for now perhaps?

WatchMkr, 2019-07-30:
@bmarshall1985 @kartikvashistha @ArnoldDeRuiter are your displays daisy chained?

WatchMkr, 2019-07-30:
We are able to confirm this bug occurs without daisy chaining. We have identified a Viewsonic model that fails with older firmware and works with newer firmware. We've reported the new finding and EDID data to nvidia.

bmarshall1985, 2019-07-30:
No daisy chaining here, all connected direct to gpu by dp when I had the issue

gkaissis, 2019-07-31:
booting with the ```nosplash``` flag fixed it for me despite the 430 driver

onais, 2019-08-28:
Neither `nosplash`, nor 'modeset=0' did not solve the issue for me.

jackpot51, 2019-09-02:
Please try the new 435.21 driver

jeremyjenkins5, 2019-09-03:
I’m still getting a blank screen when launching “settings” with 435.21 driver. 

Ive edited the /lib/modprobe.d/nvidia-graphics-drivers.conf as above also.

bmarshall1985, 2019-10-20:
Has there been any progress with this issue? I just put my 1070 in assuming it as gone now in 19.10 and its still there as before. Only now /lib/modprobe.d/nvidia-graphics-drivers.conf doesn't seem to exist. Are there any more ideas?

jackpot51, 2019-10-24:
@bmarshall1985 what is the output of `nvidia-smi`?

isamu1024, 2019-10-28:
Exactly the same problem, i can see the login page, then i see the desktop half a second and my monitor goes to sleep mode. (GTX1070).

andcarnivorous, 2020-03-12:
OS: Pop!_OS 18.04.
CPU: i7-6700K
GPU: GTX 950M 4GB

Every since last summer I can only log into Wayland sessions, and some Steam games really perform bad on Wayland. If I try to log in with Xorg I get a blank screen immediately and cannot do anything. I've always updated & upgraded the drivers in hopes of a fix. I've also tried modifing the 
```/lib/modprobe.d/nvidia-graphics-drivers.conf``` with no luck.

Is the situation resolved in 19.10?

jackpot51, 2020-03-21:
This issue was fixed a while ago with NVIDIA 435 and later

closed 2020-03-21 by jackpot51