pop-os/nvidia-graphics-drivers #11: Nvidia driver 430 breaks graphics

GabrielMorris, 2019-07-23:
I upgraded to the Nvidia 430 driver version today and now my resolution is stuck at 1024x768, my second monitor is not recognized, and nothing shows up within the Nvidia X Settings

[image]

I have tried downgrading to 418 with `sudo apt remove nvidia-driver-430 && sudo apt autoremove && sudo apt install nvidia-driver-418`, but this appears to just install 430 again. Using `ubuntu-drivers list` I can only see the 430 drivers available for my system. Additionally, no drivers show up in the Pop_Shop at all and the System76 driver states "All of the drivers for this system are provided by Ubuntu."

mmstick, 2019-07-23:
This happens if the graphics drivers aren't being loaded. How did you perform this upgrade? It would have been received as a regular system update, with no interaction required.

GabrielMorris, 2019-07-23:
I just did `sudo apt update && sudo apt upgrade` as usual.

How do I get the graphics drivers loaded?

GabrielMorris, 2019-07-23:
This is the error I get when opening `nvidia-settings` from the terminal:

```
[~]$ nvidia-settings                                                        

ERROR: An internal driver error occurred

ERROR: Unable to load info from any available system

(nvidia-settings:18487): GLib-GObject-CRITICAL **: 14:03:34.171: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

** (nvidia-settings:18487): WARNING **: 14:03:34.173: PRIME: Failed to execute child process “/usr/bin/prime-supported” (No such file or directory)
** Message: 14:03:34.173: PRIME: is it supported? no
```

mmstick, 2019-07-23:
The drivers will function by default if they are installed. Do you have `system76-driver-nvidia` installed? Any issues reported by `sudo apt install --fix-broken` or `sudo dpkg --configure -a`? You could try reinstalling the driver with `sudo apt install --reinstall nvidia-dkms-430`.

GabrielMorris, 2019-07-23:
`system76-driver-nvidia` is installed. The only error that seems to be thrown from any of those is during the reinstall of the drivers, in which it says: `rmdir: failed to remove 'updates/dkms': Directory not empty`, but it appears to complete install successfully.

mmstick, 2019-07-23:
Having reinstalled the driver, is NVIDIA functioning after a reboot? `nvidia-smi` should report 430 as being installed.

GabrielMorris, 2019-07-23:
After a reboot the drivers are not working. `nvidia-smi` throws:

```
[~]$ nvidia-smi                      
Failed to initialize NVML: Driver/library version mismatch
```

mmstick, 2019-07-23:
Sounds like you have packages installed for two different version of the NVIDIA driver. I would recommend running `sudo apt purge *nvidia*` to remove all NVIDIA packages. `dpkg --list *nvidia*` should return no packages afterwards. Then you can install `system76-driver-nvidia`.

GabrielMorris, 2019-07-23:
I think the regex for purge and dpkg are incorrect, as they do not find anything to purge/list. Could you double check that?

GabrielMorris, 2019-07-23:
Actually I found the correct regex I believe ('.*nvidia.*') but I still get the version mismatch even after reinstalling the s76 driver

mmstick, 2019-07-23:
It should work, but if you were in a directory that had a nvidia file, the shell may have expanded it. This is what I get from the purge command:

```
The following packages will be REMOVED:
  libnvidia-cfg1-430* libnvidia-common-430* libnvidia-compute-418* libnvidia-compute-430* libnvidia-decode-430* libnvidia-encode-430*
  libnvidia-fbc1-430* libnvidia-gl-430* libnvidia-ifr1-430* nvidia-compute-utils-430* nvidia-dkms-430* nvidia-driver-418* nvidia-driver-430*
  nvidia-kernel-common-430* nvidia-kernel-source-430* nvidia-settings* nvidia-utils-430* system76-driver-nvidia* xserver-xorg-video-nvidia-430*
0 upgraded, 0 newly installed, 19 to remove and 29 not upgraded.
After this operation, 323 MB disk space will be freed.
```

After reinstalling the nvidia driver, and doing a reboot, it should work.

GabrielMorris, 2019-07-23:
If I use *nvidia* I get:

```
[~/Videos/Film]$ sudo apt purge *nvidia*                
zsh: no matches found: *nvidia*
[~/Videos/Film]$ dpkg --list *nvidia*
zsh: no matches found: *nvidia*
```
(This is an empty folder)

If I do `sudo apt-get purge '.*nvidia.*' -s` I get:

```
The following packages will be REMOVED:
  libnvidia-cfg1-430*
  libnvidia-common-430*
  libnvidia-compute-430*
  libnvidia-decode-430*
  libnvidia-encode-430*
  libnvidia-fbc1-430*
  libnvidia-gl-430*
  libnvidia-gl-430:i386*
  libnvidia-ifr1-430*
  libnvidia-ifr1-430:i386*
  nvidia-compute-utils-430*
  nvidia-dkms-430*
  nvidia-driver-418*
  nvidia-driver-430*
  nvidia-kernel-common-430*
  nvidia-kernel-source-430*
  nvidia-settings* nvidia-utils-430*
  system76-driver-nvidia*
  xserver-xorg-video-nvidia-430*
0 upgraded, 0 newly installed, 20 to remove and 0 not upgraded.
```
dpkg doesn't list anything with either strings, but going ahead with that purge and then reinstalling the driver does not resolve the issue.

mmstick, 2019-07-23:
> going ahead with that purge and then reinstalling the driver does not resolve the issue.

And you rebooted after installing `system76-driver-nvidia`? That would be weird if there's still a driver/library mismatch. Perhaps you need to run `sudo update-initramfs -c -k all`?

GabrielMorris, 2019-07-23:
Yeah, I rebooted after installing it. I ran `sudo update-initramfs -c -k all` and rebooted and it still does not work.

romanze87, 2019-07-29:
Hi not sure if this is related, but when I try to run 'nvidia-settings' in terminal, it takes a few minutes to open and spits out this in the terminal:
`(nvidia-settings:16739): GLib-GObject-CRITICAL **: 21:33:13.563: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

** (nvidia-settings:16739): WARNING **: 21:33:13.712: PRIME: Failed to execute child process “/usr/bin/prime-supported” (No such file or directory)
** Message: 21:33:13.713: PRIME: is it supported? no

`

PrzemekSkw, 2019-08-01:
I also noticed that nvidia driver don't work properly after update on my Dell laptop. Glx Gears work but shows only:
```
337 frames in 5.0 seconds = 67.313 FPS                                                                                                                                                                                                      
315 frames in 5.0 seconds = 62.924 FPS                                                                                                                                                                                                      
301 frames in 5.0 seconds = 60.013 FPS
```
so nothing works on that driver.   
I install `system76-driver-nvidia` and reinstall `sudo apt install --reinstall nvidia-dkms-430` but don't work.

PrzemekSkw, 2019-08-01:
> Sounds like you have packages installed for two different version of the NVIDIA driver. I would recommend running `sudo apt purge *nvidia*` to remove all NVIDIA packages. `dpkg --list *nvidia*` should return no packages afterwards. Then you can install `system76-driver-nvidia`.

I also do this and still don't work.
```
przemek@pop-os:~$ glxgears                                                                                                                                                                                                                  
Running synchronized to the vertical refresh.  The framerate should be                                                                                                                                                                      
approximately the same as the monitor refresh rate.                                                                                                                                                                                         
322 frames in 5.0 seconds = 64.385 FPS                                                                                                                                                                                                      
XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":1"                                                                                                                                                                 
      after 55 requests (55 known processed) with 0 events remaining.                                                                                                                                                                       
przemek@pop-os:~$ nvidia-smi                                                                                                                                                                                                                
Thu Aug  1 11:43:05 2019                                                                                                                                                                                                                    
+-----------------------------------------------------------------------------+                                                                                                                                                             
| NVIDIA-SMI 430.34       Driver Version: 430.34       CUDA Version: 10.1     |                                                                                                                                                             
|-------------------------------+----------------------+----------------------+                                                                                                                                                             
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |                                                                                                                                                             
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |                                                                                                                                                             
|===============================+======================+======================|                                                                                                                                                             
|   0  GeForce GTX 1050    Off  | 00000000:01:00.0 Off |                  N/A |                                                                                                                                                             
| N/A   49C    P0    N/A /  N/A |    355MiB /  4042MiB |      1%      Default |                                                                                                                                                             
+-------------------------------+----------------------+----------------------+                                                                                                                                                             
                                                                                                                                                                                                                                            
+-----------------------------------------------------------------------------+                                                                                                                                                             
| Processes:                                                       GPU Memory |                                                                                                                                                             
|  GPU       PID   Type   Process name                             Usage      |                                                                                                                                                             
|=============================================================================|                                                                                                                          [... cut ...]      
drm                   475136  13 drm_kms_helper,nvidia_drm,i915                                                                                                                                                                             
ipmi_msghandler       102400  2 ipmi_devintf,nvidia                                                                                                                                                                                         
przemek@pop-os:~
```

jackpot51, 2019-09-02:
Please try the new 435.21 driver

alessandrojcm, 2020-06-05:
I'm having the same issues with the 440.82 driver and kernel 5.4. I tried everything in this thread and even tried downgrading to kernel 5.3; no luck. Sadly I cannot say if it was working before, because until now I didn't need the NVIDIA card, but today I tried connecting an external monitor via HDMI and 3 hours later no luck. I have an HP laptop with a dual Intel 630 UHD and a NVIDIA 1050 ti running the latest version of Pop!_OS.

[image]

alessandrojcm, 2020-06-06:
I managed to get it working: I took the xorg generated by `nvidia-xconfig` and added the dGPU `BusID` (it was missing IDK why); running smoothly on hybrid mode. Although I think this is weird, I've found plenty of people that said this worked out of the box :thinking:

kingcheng2000, 2020-07-30:
I got the same issue and not got it fixed yet.

robot@robot-Inspiron-5593:~$ neofetch
            .-/+oossssoo+/-.               robot@robot-Inspiron-5593 
        `:+ssssssssssssssssss+:`           ------------------------- 
      -+ssssssssssssssssssyyssss+-         OS: Ubuntu 18.04.4 LTS x86_64 
    .ossssssssssssssssssdMMMNysssso.       Host: Inspiron 5593 
   /ssssssssssshdmmNNmmyNMMMMhssssss/      Kernel: 4.15.0-112-generic 
  +ssssssssshmydMMMMMMMNddddyssssssss+     Uptime: 20 mins 
 /sssssssshNMMMyhhyyyyhmNMMMNhssssssss/    Packages: 2883 
.ssssssssdMMMNhsssssssssshNMMMdssssssss.   Shell: bash 4.4.20 
+sssshhhyNMMNyssssssssssssyNMMMysssssss+   Resolution: 1920x1080 
ossyNMMMNyMMhsssssssssssssshmmmhssssssso   DE: GNOME 3.28.4 
ossyNMMMNyMMhsssssssssssssshmmmhssssssso   WM: GNOME Shell 
+sssshhhyNMMNyssssssssssssyNMMMysssssss+   WM Theme: Adwaita 
.ssssssssdMMMNhsssssssssshNMMMdssssssss.   Theme: Ambiance [GTK2/3] 
 /sssssssshNMMMyhhyyyyhdNMMMNhssssssss/    Icons: Ubuntu-mono-dark [GTK2/3] 
  +sssssssssdmydMMMMMMMMddddyssssssss+     Terminal: gnome-terminal 
   /ssssssssssshdmNNNNmyNMMMMhssssss/      CPU: Intel i7-1065G7 (8) @ 3.900GHz 
    .ossssssssssssssssssdMMMNysssso.       GPU: Intel Integrated Graphics 
      -+sssssssssssssssssyyyssss+-         GPU: NVIDIA NVIDIA Corporation Devic 
        `:+ssssssssssssssssss+:`           Memory: 2187MiB / 7745MiB 
            .-/+oossssoo+/-. 
                                                                   

robot@robot-Inspiron-5593:/etc/X11$ sudo nvidia-settings

ERROR: Unable to load info from any available system

(nvidia-settings:15001): GLib-GObject-CRITICAL **: 09:30:13.714: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
** Message: 09:30:13.717: PRIME: No offloading required. Abort
** Message: 09:30:13.717: PRIME: is it supported? no

//  HDMI can NOT be detected 
robot@robot-Inspiron-5593:/etc/X11$ xrandr
xrandr: Failed to get size of gamma for output default
Screen 0: minimum 1920 x 1080, current 1920 x 1080, maximum 1920 x 1080
default connected primary 1920x1080+0+0 0mm x 0mm
   1920x1080     77.00* 
robot@robot-Inspiron-5593:/etc/X11$ 

robot@robot-Inspiron-5593:/etc/X11$ nvidia-smi
Thu Jul 30 09:32:26 2020       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 410.104      Driver Version: 410.104      CUDA Version: 10.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce MX230       Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   54C    P8    N/A /  N/A |      0MiB /  4042MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|  No running processes found                                                 |
+-----------------------------------------------------------------------------+
robot@robot-Inspiron-5593:/etc/X11$