pop-os/nvidia-graphics-drivers #13: Increased VRAM usage with 430

IceAmaura, 2019-07-25:
Using an NVIDIA GTX 1050 in Optimus mode.

430.34 uses 60%+ VRAM on idle vs 33% VRAM on idle on 418.56. Measured using the NVIDIA X Server panel. It also seems to have issues clearing the VRAM when closing programs as it doesn't fully go back to previous VRAM state like in 418.56.

WatchMkr, 2019-07-29:
We're unable to replicate this problem on three machines with different nvidia graphics card setups.

RTX 2080
GTX 1050 Ti
GTX 1650 in Optimus mode

Idle dedicated graphics memory usage is under 10%, increases under benchmarking load, and decreases again below 10% at the end of benchmarking.

Is there something you're doing that might help us replicate the bug?

IceAmaura, 2019-07-30:
I do have an extension installed called Material-shell. That's the only
thing I have different from a vanilla install (that and having my nvidia
driver held back for now).

closed 2026-01-07 by leviport