pop-os/nvidia-graphics-drivers #15: Nvidia 430.34 Driver graphical errors in game.

romanze87, 2019-07-29:
Hello, I am on Nvidia 430.34 trying to play Arma 3 on the Linux branch and the graphics are really terrible on any settings. Here is a list of the problems:
1. There is a weird grid-like texture over top of everything
2. When running or turning the graphics drop to really low quality and I can see some black boxes pop up very briefly.
3. Grass is flickering
4. Unrelated to graphics, but since I updated Pop a few days ago whenever I try to open Nvidia X Server Settings it takes minutes to open. This is usually not the case.
I am on i7 8700K, Nvidia 1660 Ti, 16 GB 3000 Mhz RAM and all the latest Pop!_OS updates. I will say I have not run Arma 3 on this system prior to this, but on another Ubuntu system, I have had no graphical issues. 
I tried to display the issue in this picture, but if you look at the grass it's very blurry with a grid-like texture.

[image]

WatchMkr, 2019-07-29:
Thanks for your report.

Do performance issues occur in other games? Are you using a custom 20-nvidia.conf in /etc/X11/xorg.conf.d/? If yes, what are its contents?

jackpot51, 2019-09-02:
Please try the new 435.21 driver

VoQZHD, 2020-09-08:
> Please try the new 435.21 driver

How to install it? This inbuild driver absolutely breaks everything and no one cares. Just open issues

VoQZHD, 2020-09-08:
Most people installed Pop Os as their first Linux experience and this happens

leviport, 2020-09-08:
This issue hasn't had any activity for over a year. The current Nvidia driver in 20.04 is 440.100. Are you running 20.04 with 440.100 and continuing to see this issue? Also, are you also using a 1660Ti like the original poster was using?

mmstick, 2020-09-08:
@VoQZHD You should create a new issue with your specs and specific issue. But know that we can't do much about rendering issues in certain games. That's primarily up to the game developer or NVIDIA to fix.

closed 2020-09-08 by mmstick

VoQZHD, 2020-09-08:
The thing is pop os is based in on ubuntu yet ubuntu works splendid on both systems. I guess il install pop shell for now and wait for fixes. I just really love pop os. Made me to finally look into linux