pop-os/nvidia-graphics-drivers #17: 430 drivers break all external monitors

josephxsxn, 2019-08-01:
Also started having issues after upgrading to the latest drivers for nvidia 430, I can no longer use any external monitors. Below are my system specs. Also I have noticed that lspci sometimes will not always show the nvidia card anymore and only show the Intel one, I have had the vga device show back up and this allowed 1 external monitor to work when I was uninstall and reinstall the drivers, but most of the time it doesn't work; plugging in 2 monitors forces the laptop screen to go black.

Model: oryp4 [oryx pro]
Current Nvidia version : 430
Pop os version : 19.04
Pop os driver version : 19.04.9

```
dkms status
nvidia, 430.34, 5.0.0-21-generic, x86_64: installed
system76, 1.0.4~1561669629~19.04~6ea4ee8, 4.18.0-25-generic, x86_64: installed
system76, 1.0.4~1561669629~19.04~6ea4ee8, 5.0.0-21-generic, x86_64: installed
system76-io, 1.0.1~1559663713~19.04~ea5f61a, 4.18.0-25-generic, x86_64: installed
system76-io, 1.0.1~1559663713~19.04~ea5f61a, 5.0.0-21-generic, x86_64: installed

```

PS -- I also have another Oryx Pro model  oryp3-ess on 19.04 with the 430 drivers which has no issues at all.

WatchMkr, 2019-08-05:
Hi Joseph, please open a ticket with our support team. They'll be able to help you out.

closed 2019-08-05 by WatchMkr

josephxsxn, 2019-08-08:
just wanted to follow up in here that in the end I reformatted and installed 19.04 clean, and then upgraded to 430.34 drivers and have no problems anymore. This machine before had been upgraded from 18.04 to 19.04 with a do-release-upgrade