pop-os/nvidia-graphics-drivers #22: How to install nvidia 430.40 driver on PopOS!?

PrzemekSkw, 2019-08-17:
There's still broken 430.34 driver, why there's no update? Please tell me how to install last nvidia drivers, I don't want to migrate to another distro.

mmstick, 2019-08-18:
The 430.40 driver did not fix the issue, so it was not pushed. If you still want to upgrade to 430.40, then you can do so by cloning the `pop-os/pop` repository and running `./scripts/apt add 430.40` instead the repository.

PrzemekSkw, 2019-08-19:
@mmstick thanks for response. So 430.40 will still be broken on my graphics?In that case I have to still block to upgrade drivers (apt-hold nvidia) and use 430.26?

OMGtechy, 2019-12-16:
@PrzemekSkw on a related note, if you want to install a driver manually you can do so.

The high level steps are:

1. Disable the current driver
2. Install the new one
3. Load the new driver

Here's each step fleshed out a bit (all of this is in the terminal):

**Disable the current driver**

First, you need to blacklist the nvidia modules by creating a file at /etc/modprobe.d/blacklist-nvidia.conf with the following content (you'll need to run your text editor as root or use sudo):

**/etc/modprobe.d/blacklist-nvidia.conf**
```

blacklist nvidia_drm
blacklist nvidia
blacklist nvidia_modeset

```
Next you'll need to regenerate the initial root filesystem:
`sudo update-initramfs -u`

Now reboot either as you would normally or by running the following command:
`reboot`

**Install the new one**

Download the driver for your card here:
`https://www.nvidia.com/Download/index.aspx`

Find the file and mark it as executable. For example, if the file is in your Downloads folder and is called `NVIDIA-Linux-x86_64-440.44.run`:

```
cd ~/Downloads
chmod +x ./NVIDIA-Linux-x86_64-440.44.run
```

Install the driver:
`sudo ./NVIDIA-Linux-x86_64-440.44.run`

**Load the new driver**

You need to remote the blacklist file you made earlier
`sudo rm /etc/modprobe.d/blacklist-nvidia.conf`

Regenerate the root filesystem:
`sudo update-initramfs -u`

Reboot the system:
`reboot`

**Done!**

You should have the new driver up and running now. You can double check this by running `nvidia-smi`:

```
$ nvidia-smi 
Mon Dec 16 03:55:51 2019       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.44       Driver Version: 440.44       CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+

Hope this is helpful to you!

```

mmstick, 2019-12-16:
@OMGtechy Completely necessary, and fairly dangerous. It's much easier to add a staging branch.

```
git clone https://github.com/pop-os/pop
./scripts/apt add 440.44
sudo apt upgrade
```

brs17, 2019-12-17:
The staging branch has been deleted and 440.44 will be released today.

closed 2019-12-17 by brs17