pop-os/nvidia-graphics-drivers #24: Provide Nvidia drivers for cuda while using the desktop with Intel graphics

KIC, 2019-09-07:
As a developer i really like Pop!_OS. Since I need to develop a lot of cuda stuff I have bought a Notebook with Intel+Nvidia GPU. However, currently if I boot into into Intel Graphics I can not use cuda :.-(

```
kic@pop-os:~$ nvidia-smi 
NVIDIA-SMI has failed because it couldn't communicate with the NVIDIA driver. Make sure that the latest NVIDIA driver is installed and running.
```

This is really a pity since I actually don't need any graphics from Nvidia GPU :-) So I want to propose two options for an enhancement:

1. either allow cuda while running the desktop using Intel GPU
1. introduce a third option in the menu "Use Intel Graphics with Headless Nvidia"

Now since I have already set my system up as I like it to be. Is there some workaround how I can get Nvidia drivers loaded for cuda while I am using the desktop on Intel? Like is it enough to run `sudo apt install nvidia*headless*`?

robobenklein, 2020-10-03:
@KIC does the new system76-power work?

I got compute-only mode working with https://github.com/pop-os/system76-power/pull/162

jacobgkau, 2020-10-09:
Compute mode is precisely this, it allows keeping the NVIDIA GPU powered on and available for compute workloads (such as CUDA) without using it to render the desktop. Closed by https://github.com/pop-os/system76-power/pull/162.

closed 2020-10-09 by jacobgkau