pop-os/nvidia-graphics-drivers #26: NVIDIA [435.21] - gnome-shell process consuming 100% CPU - interfering with NVIDIA.

xNarkon, 2019-10-01:
`+-----------------------------------------------------------------------------+
| NVIDIA-SMI 435.21       Driver Version: 435.21       CUDA Version: 10.1     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 1650    Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   43C    P8     5W /  N/A |    507MiB /  3911MiB |     10%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1654      G   /usr/lib/xorg/Xorg                            27MiB |
|    0      2636      G   /usr/lib/xorg/Xorg                           137MiB |
|    0      2796      G   /usr/bin/gnome-shell                         196MiB |
|    0      3400      G   ...quest-channel-token=3161121399846160830     6MiB |
|    0      5803      G   ...quest-channel-token=4539356392965985163    84MiB |
+-----------------------------------------------------------------------------+`

`             /////////////                narkon@thinkpad-x1-extreme 
         /////////////////////            ------------------------------- 
      ///////*767////////////////         OS: Pop!_OS 19.04 x86_64 
    //////7676767676*//////////////       Host: 20QV001GPB ThinkPad X1 Extreme 2nd 
   /////76767//7676767//////////////      Kernel: 5.2.15-050215-generic 
  /////767676///*76767///////////////     Uptime: 18 mins 
 ///////767676///76767.///7676*///////    Packages: 2271 (dpkg), 13 (snap) 
/////////767676//76767///767676////////   Shell: zsh 5.5.1 
//////////76767676767////76767/////////   Resolution: 1920x1080 
///////////76767676//////7676//////////   DE: GNOME 3.32.2 
////////////,7676,///////767///////////   WM: GNOME Shell 
/////////////*7676///////76////////////   WM Theme: Pop 
///////////////7676////////////////////   Theme: Pop-slim-dark [GTK2/3] 
 ///////////////7676///767////////////    Icons: Pop [GTK2/3] 
  //////////////////////'////////////     Terminal: gnome-terminal 
   //////.7676767676767676767,//////      CPU: Intel i7-9750H (12) @ 4.500GHz 
    /////767676767676767676767/////       GPU: NVIDIA NVIDIA Corporation Device 1f91 
      ///////////////////////////         GPU: Intel UHD Graphics 630 
         /////////////////////            Memory: 2687MiB / 31807MiB 
             /////////////
                                                                  

`
Let's get to the point. Usually I'm connected onto two external displays so I need to use my dedicated graphic card (nvidia). I was thinking that everything is working flawlessly, but it isn't. When I have disconnected external displays and only using my internal display with enabled NVidia I got very annoying and even dangerous problem. On locked screen with black display (I mean with sleeping display) gnome-shell uses 100% CPU what causes high temp's WHEN I'm not doing anything because my computer is locked! You can ask, what the Nvidia has to this problem - answer is pretty clear and easy - this situation only happens when I don't have any external displays connected so I'm using only my internal display and only when I'm working on NVidia GPU - on my integrated Intel GPU everything is fine.

Do you have any idea what I can do? Is it driver specific problem?

I was trying everything like changing linux kernel, reinstalling nvidia drivers etc.

mmstick, 2019-10-01:
It's usually a bad extension that causes high CPU usage. Have you tried disabling all third party extensions?

GNOME Shell writes its logs to journald, which you can read by running `journalctl /usr/bin/gnome-shell`. You can follow the logs with the `-f` argument flag.

xNarkon, 2019-10-01:
Unfortunately, I don't have any additional extensions except those installed by default by PopOS. But, I disabled everything non-standard and there is still the same situation. Only one core is going up to 100% load by gnome-shell - as was said at related topic I linked in the last message probably the drivers are problematic in this case:

> Unfortunately I cannot help beyond pointing out that the graphics drivers are bonkers.
> But we can now get an idea of why it's loading so high.
> It's running the llvmpipe drivers, which run everything through the cpu.
> llvmpipe is a fallback driver that runs when the proper drivers have some issue or another that prevents them from running cleanly.

```Oct 01 22:48:11 thinkpad-x1-extreme org.gnome.Shell.desktop[2796]: [1:1:1001/224811.379911:ERROR:child_process_sandbox_support_impl_linux.cc(81)] FontService unique font name matching request 
Oct 01 22:48:11 thinkpad-x1-extreme org.gnome.Shell.desktop[2796]: [1:1:1001/224811.380413:ERROR:child_process_sandbox_support_impl_linux.cc(81)] FontService unique font name matching request 
Oct 01 22:49:29 thinkpad-x1-extreme org.gnome.Shell.desktop[2796]: [1:1:1001/224929.677860:ERROR:child_process_sandbox_support_impl_linux.cc(81)] FontService unique font name matching request 
Oct 01 22:49:29 thinkpad-x1-extreme org.gnome.Shell.desktop[2796]: [1:1:1001/224929.678282:ERROR:child_process_sandbox_support_impl_linux.cc(81)] FontService unique font name matching request 
Oct 01 22:50:17 thinkpad-x1-extreme org.gnome.Shell.desktop[2796]: [1:1:1001/225017.694151:ERROR:child_process_sandbox_support_impl_linux.cc(81)] FontService unique font name matching request 
Oct 01 22:50:17 thinkpad-x1-extreme org.gnome.Shell.desktop[2796]: [1:1:1001/225017.694537:ERROR:child_process_sandbox_support_impl_linux.cc(81)] FontService unique font name matching request 
Oct 01 22:57:27 thinkpad-x1-extreme gnome-shell[2796]: JS WARNING: [resource:///org/gnome/gjs/modules/signals.js 128]: Too many arguments to method Clutter.Actor.destroy: expected 0, got 1
Oct 01 22:57:27 thinkpad-x1-extreme gnome-shell[2796]: JS WARNING: [resource:///org/gnome/gjs/modules/signals.js 128]: Too many arguments to method Clutter.Actor.destroy: expected 0, got 1
Oct 01 22:57:27 thinkpad-x1-extreme gnome-shell[2796]: JS WARNING: [resource:///org/gnome/gjs/modules/signals.js 128]: Too many arguments to method Clutter.Actor.destroy: expected 0, got 1
Oct 01 23:05:50 thinkpad-x1-extreme gnome-shell[2796]: gnome-shell-extension-system76-power: power profile was set: 'Balanced'
Oct 01 23:05:50 thinkpad-x1-extreme gnome-shell[2796]: [AppIndicatorSupport-DEBUG] Registering StatusNotifierItem :1.79/StatusNotifierItem
Oct 01 23:05:50 thinkpad-x1-extreme gnome-shell[2796]: [AppIndicatorSupport-DEBUG] Registering StatusNotifierItem :1.70/StatusNotifierItem
Oct 01 23:05:50 thinkpad-x1-extreme gnome-shell[2796]: [AppIndicatorSupport-DEBUG] Registering StatusNotifierItem :1.73/StatusNotifierItem
Oct 01 23:05:50 thinkpad-x1-extreme gnome-shell[2796]: [AppIndicatorSupport-FATAL] unable to update overlay icon
Oct 01 23:05:50 thinkpad-x1-extreme gnome-shell[2796]: [AppIndicatorSupport-FATAL] unable to update overlay icon
Oct 01 23:05:50 thinkpad-x1-extreme gnome-shell[2796]: [AppIndicatorSupport-FATAL] unable to update overlay icon
Oct 01 23:05:50 thinkpad-x1-extreme gnome-shell[2796]: [AppIndicatorSupport-FATAL] unable to update overlay icon
Oct 01 23:05:50 thinkpad-x1-extreme gnome-shell[2796]: [AppIndicatorSupport-FATAL] unable to update overlay icon
Oct 01 23:05:50 thinkpad-x1-extreme gnome-shell[2796]: [AppIndicatorSupport-FATAL] unable to update overlay icon
Oct 01 23:05:50 thinkpad-x1-extreme gnome-shell[2796]: [AppIndicatorSupport-FATAL] unable to update overlay icon
Oct 01 23:05:50 thinkpad-x1-extreme gnome-shell[2796]: [AppIndicatorSupport-FATAL] unable to update overlay icon
Oct 01 23:05:50 thinkpad-x1-extreme gnome-shell[2796]: [AppIndicatorSupport-FATAL] unable to update overlay icon
Oct 01 23:05:51 thinkpad-x1-extreme org.gnome.Shell.desktop[2796]: Window manager warning: Overwriting existing binding of keysym 36 with keysym 36 (keycode f).
Oct 01 23:05:51 thinkpad-x1-extreme org.gnome.Shell.desktop[2796]: Window manager warning: Overwriting existing binding of keysym 39 with keysym 39 (keycode 12).
Oct 01 23:05:51 thinkpad-x1-extreme org.gnome.Shell.desktop[2796]: Window manager warning: Overwriting existing binding of keysym 38 with keysym 38 (keycode 11).
Oct 01 23:05:51 thinkpad-x1-extreme org.gnome.Shell.desktop[2796]: Window manager warning: Overwriting existing binding of keysym 32 with keysym 32 (keycode b).
Oct 01 23:05:51 thinkpad-x1-extreme org.gnome.Shell.desktop[2796]: Window manager warning: Overwriting existing binding of keysym 33 with keysym 33 (keycode c).
Oct 01 23:05:51 thinkpad-x1-extreme org.gnome.Shell.desktop[2796]: Window manager warning: Overwriting existing binding of keysym 35 with keysym 35 (keycode e).
Oct 01 23:05:51 thinkpad-x1-extreme org.gnome.Shell.desktop[2796]: Window manager warning: Overwriting existing binding of keysym 34 with keysym 34 (keycode d).
Oct 01 23:05:51 thinkpad-x1-extreme org.gnome.Shell.desktop[2796]: Window manager warning: Overwriting existing binding of keysym 37 with keysym 37 (keycode 10).
Oct 01 23:05:51 thinkpad-x1-extreme org.gnome.Shell.desktop[2796]: Window manager warning: Overwriting existing binding of keysym 31 with keysym 31 (keycode a).
[... cut ...]
Oct 01 23:05:51 thinkpad-x1-extreme gnome-shell[2796]: Error setting metadata to desktop files , Gio.IOErrorEnum: Setting attribute metadata::nautilus-icon-position not supported
Oct 01 23:05:51 thinkpad-x1-extreme gnome-shell[2796]: Error setting metadata to desktop files , Gio.IOErrorEnum: Setting attribute metadata::nautilus-icon-position not supported
Oct 01 23:05:51 thinkpad-x1-extreme gnome-shell[2796]: Error setting metadata to desktop files , Gio.IOErrorEnum: Setting attribute metadata::nautilus-icon-position not supported```

mmstick, 2019-10-01:
If you're using LLVMpipe, then NVIDIA drivers aren't installed. You should try purging the NVIDIA packages and reinstalling them.

```sh
sudo apt purge nvidia*
sudo apt install nvidia-driver-435
```

xNarkon, 2019-10-01:
I'm not using it, it's probably use as a fallback for nvidia drivers. I was trying this before and it not helped, unfortunately.

mmstick, 2019-10-01:
Is your system set to use hybrid graphics, or is it in discrete-only mode? The fallback should be the Intel iGPU if the NVIDIA driver was not installed, and the nouveau driver has been blacklisted.

xNarkon, 2019-10-02:
Hybrid graphics

essagl, 2020-02-04:
Hi,i had exactly the same problem. Installing the System76 Power Management shell extension solved it for me.