pop-os/nvidia-graphics-drivers #33: Nvidia 440.31 update breaks obs-studio and TF2

Lockdonnen, 2019-11-22:
After updating to the 440.31 Nvidia drivers I'm unable to launch obs-studio, it gives me an error;
"Failed to intialize video. your GPU may not be supported, or graphics drivers may need to be updated."
With TF2 it will play the intro valve music and bald guy but when it attempts to load the main menu it will complain about not enough video memory, or just "menuui load error"

Strangely these are the only two things so far that don't work. Other things like Blender, CS:GO or other games work fine.

Using a GTX1080 [NVIDIA GP104]

mmstick, 2019-11-22:
I've had no issues opening TF2 or OBS Studio with NVIDIA on 440.31. Do you get an output from `nvidia-smi` indicating that the driver is loaded?

Lockdonnen, 2019-11-22:
$ nvidia-smi
Fri Nov 22 13:53:39 2019       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.31       Driver Version: 440.31       CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 1080    Off  | 00000000:01:00.0  On |                  N/A |
| 10%   52C    P0    42W / 180W |    922MiB /  8116MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1773      G   /usr/lib/xorg/Xorg                            30MiB |
|    0      2805      G   /usr/lib/xorg/Xorg                           228MiB |
|    0      3028      G   /usr/bin/gnome-shell                         251MiB |
|    0      6472      G   ...m/debian-installation/ubuntu12_32/steam    90MiB |
|    0      6496      G   ./steamwebhelper                               2MiB |
|    0      6513      G   ...AAAAAAAAhAAAgAAAAAAAAYAAAAAAAEAAAAAAAAA   230MiB |
|    0      8026      G   /usr/lib/firefox/firefox                       1MiB |
|    0      8232      G   /usr/lib/firefox/firefox                       1MiB |
|    0      8263      G   /usr/lib/firefox/firefox                       1MiB |
|    0      8271      G   /usr/lib/firefox/firefox                       1MiB |
|    0     26558      G   ...uest-channel-token=18363615394935335766    29MiB |
+-----------------------------------------------------------------------------+

Seems to all be there. Also I tested this on my T480s which has the intel+nvidia combo. On the Intel GPU tf2 and obs loads but not with nvidia.
Also forgot some details if needed, running on Pop!_OS 19.10 with kernel,
5.3.0-22-generic #24+system76~1573659475~19.10~26b2022-Ubuntu

mmstick, 2019-11-22:
Are you in hybrid or NVIDIA graphics mode?

mmstick, 2019-11-22:
Also, are you using OBS Studio from the Ubuntu repos?

Lockdonnen, 2019-11-22:
On my big tower machine I'm running the 1080 by itself, the intel gpu is disabled via bios.
On the laptop it's NVIDIA graphics mode via system76s power switcher, the hybrid mode is to fussy for me.

I have obs-studio installed via flatpak, forgot about that. I'll try via straight from ubuntu repo.

Also this was recently posted on Lutris discord:

> It appears that the new 440.xx proprietary Nvidia drivers have an issue with shaders - they get discarded each time you reopen a game.
It may not affect all games, but it should affect a lot of them, especially the ones with a big number of shaders, like Overwatch.
For now, I suggest to rollback to 435.xx or 430.xx series of drivers. I'll make another announcement when the bug gets resolved.

Lockdonnen, 2019-11-22:
OBS works from the ubuntu repos.

mmstick, 2019-11-22:
NVENC works on Pop!_OS with OBS Studio from the Ubuntu repositories because we repackage ffmpeg with NVENC support enabled.

mmstick, 2019-11-22:
We did just package the 440.36 driver which released today.

mmstick, 2019-11-22:
Have you performed a flatpak update recently? It does need to have the matching drivers installed.

Lockdonnen, 2019-11-22:
> NVENC works on Pop!_OS with OBS Studio from the Ubuntu repositories because we repackage ffmpeg with NVENC support enabled.

Well this is pretty much why I was using the flatpak version, thanks!
And yes, I was dumb and didn't update the flatpak version, works now...not that I need it any more.

TF2 however is still crashing on menu load. I've gone through and cleared the shader cache for it as well just to check but it seems to be the only thing complaining. apt update isn't showing 440.36 yet, but once I get that I'll see if that fixes it.

Lockdonnen, 2019-11-28:
Discovered an issue report in ValveSoftware/Source-1-Games that sounded a lot like my problem.
This solved it for me.

https://github.com/ValveSoftware/Source-1-Games/issues/2866#issuecomment-553748300

closed 2019-11-28 by Lockdonnen