pop-os/nvidia-graphics-drivers #35: 440.31 breaks external monitors

MyMirelHub, 2019-11-25:
Hello, I'm running pop 19.10 on Lenovo P1, on a quadro p1000.

When upgrading from 435 to 440 I am unable to use my external monitors. They are detected by the gnome display manager and the Nvidia x-server GUI settings, but the monitors have the `no signal detected` error. I tried this with two different brand monitors and different cables.

Unfortunately, I can't give any logs as I ended up reinstalling the OS and pinning the nivida packages to 435 which fixed the problem again. 

Wondering if anyone else is experiencing this?

haydenhughes, 2019-12-12:
I wonder if it could potentially be because of #37?
To install the driver are you using the `system76-driver-nvidia` package?

MyMirelHub, 2019-12-12:
Yes, it was through the `system76` package. Thanks, next time I'll try doing a purge first and post it here.

nmehitabel, 2020-01-06:
I am experiencing this issue as well on Lenovo P1 / quadro p1000. Fresh install of 19.10, have tried purge and reinstall of 440 driver as suggested but to no avail. If I hot plug HDMI is not recognised at all, mini dpi appears to be accepted but monitor remains blank. If I boot with cable plugged in then HDMI is found but still no display on external monitor.

MyMirelHub, 2020-01-16:
Updating to the latest `440.44` has fixed the issue for me

Xceno, 2020-03-19:
I'm trying to fix this issue on a Thinkpad X1 Extreme (Gen1) for several weeks now, but have absolutely no luck. There are many threads on reddit, a related forum post on [pop-planet](https://pop-planet.info/forums/threads/state-of-the-art-of-nvidia-prime.408), and various blog-posts in the wild.

I'm not sure if this is a POP!_OS problem at all, or rather an nvidia issue that happens on hybrid intel/nvidia  systems. For example, here's someone describing the same issue [on gentoo](https://bugs.gentoo.org/670340#c8).

In my case, the nvidia driver isn't even loaded properly. I purged and reinstalled different versions (currently running nvidia-driver-440), to no avail.
I tried my luck with installing `nvidia-prime` and using it instead of `system76-power` but this only resulted in a black screen.

Whenever I ran `nvidia-smi` I get:
`NVIDIA-SMI has failed because it couldn't communicate with the NVIDIA driver. Make sure that the latest NVIDIA driver is installed and running.`

My relevant system info:
```
OS: Pop!_OS 19.10 x86_64 
Host: 20MFCTO1WW ThinkPad X1 Extreme 
Kernel: 5.3.0-7642-generic 
DE: GNOME 3.34.3 
CPU: Intel i7-8750H (12) @ 4.100GHz 
GPU: NVIDIA GeForce GTX 1050 Ti Mobile 
GPU: Intel UHD Graphics 630 
Memory: 31569MiB 
```

I'm using the latest firmware for everything, the latest BIOS, disabled secure boot, and tried all the other things the internet had to offer. To me it seems that everybody's just trying random things at this point. I sure do.

Maybe a complete re-install fixes it, that's the only thing I didn't try yet.

jackpot51, 2020-03-19:
@Xceno it looks like you have disabled the nvidia driver in either nvidia-prime or system76-power. I recommend a reinstall. Please make a new issue if that doesn't work.

closed 2020-03-19 by jackpot51