pop-os/nvidia-graphics-drivers #41: Is updating directly from NVIDIA site is a bad idea?

aasutossh, 2020-02-12:
I saw that there is an update for NVIDIA graphics driver for 1050 Ti.
https://www.nvidia.com/Download/driverResults.aspx/156772/en-us
I was also having issues with games lag, hang, so on.
I also notices youtube getting stuck for a second after playing 3-4 seconds after I put my laptop in sleep all night(if it matters).

mmstick, 2020-02-12:
Yes, it is a bad idea.

The 440.59 branch is already available in Pop!_OS through the `440.59` staging branch.

https://github.com/pop-os/nvidia-graphics-drivers/tree/440.59

aasutossh, 2020-02-12:
Thanks for replying.
I got it.

aasutossh, 2020-02-12:
BTW, how do I get it? From the staging branch.

closed 2020-02-12 by aasutossh

jaxn, 2020-02-19:
@aasutossh I found this in [another thread](https://github.com/pop-os/nvidia-graphics-drivers/issues/22)

```
git clone https://github.com/pop-os/pop
./scripts/apt add 440.59
sudo apt upgrade
```
that worked for me :)

aasutossh, 2020-02-20:
Sadly now I cant boot after updating the driver.
How do I revert?

aasutossh, 2020-02-20:
I mean I got black screen during boot.

aasutossh, 2020-02-20:
Solved at https://www.reddit.com/r/pop_os/comments/f6src4/black_screen_just_after_booting_and_showing_hp/.