pop-os/nvidia-graphics-drivers #45: How to install latest 440.64 drivers?

acrovia, 2020-03-29:
Hello,
I love pop-os because of it's stability, but dudes, we all want to play latest games with Vulkan. Specially now in quarantine. 
Can you upgrade it faster ?
Or can you give a solution on how to purge the pop-os drivers an use the latest ones? Will that destroy my system?
Thanks!

chuckyz, 2020-04-13:
440.64 is already out with 440.82 in PR.  just `sudo apt-get upgrade nvidia-driver-440` should do the trick.  I believe this is also possible through the pop shop but I'm not sure what the right package is.

Vyrr, 2020-06-20:
IIRC Nvidia driver number does not make sense and beta drivers are not following the same notation as public drivers.
See here: https://developer.nvidia.com/vulkan-driver
Can confirm that driver 440.82 does not support Vulkan 1.2, and beta driver 440.66 (as of today) does support Vulkan 1.2.

thierrydev, 2020-07-09:
I have to say that it seems to be a bit of a struggle to install or update  Vulkan Beta Driver

thierrydev, 2020-07-09:
this feature  seem to be ignored by design to favor[ System76 products ](https://support.system76.com/articles/system76-driver/)

jacobgkau, 2020-07-27:
We are currently shipping NVIDIA driver version 440.100.

closed 2020-07-27 by jacobgkau