pop-os/nvidia-graphics-drivers #47: Upgrading to 440.64 broke mouse and vulkan/lutris

ProtonLover432, 2020-04-03:
I updated to 440.64 today and restarted my computer. Now my mouse moves sluggishly or really laggy. I also tried to play Guild Wars 2 using Lutris, which worked fine yesterday. When I hit play I got a  popup from lutris saying I didn't have a Vulkan Compatible card, I have a 2080ti so it is. Again this was all working fine yesterday before I did the update.

mmstick, 2020-04-03:
Check that the NVIDIA driver is being loaded, and not reporting a mismatch (`nvidia-smi`).

Ensure that the initramfs is updated, and copied to the EFI partition.

```
sudo update-initramfs -c -k all
sudo kernelstub -v
```

ProtonLover432, 2020-04-03:
nvidia-smi returns `Failed to initialize NVML: Driver/library version mismatch` how would I go about fixing that.
I see in dmesg
`NVRM: API mismatch: the client had the version 440.64, but this kernel module has the version 440.44`

ProtonLover432, 2020-04-03:
I just remembered I am not using the default kernel that ships with pop. I needed a new one for my wifi card. I switched back to the current and after a restart it is working fine. Sorry for the trouble, thanks for the help!

closed 2020-04-03 by ProtonLover432