pop-os/nvidia-graphics-drivers #49: Multiple GPU's with Multiple Monitors not working.

OliverGee, 2020-05-04:
Hi there,

I posted this issue on Reddit but it bizarrely got downvoted so I am reposting here in the hope that someone can point me in the right direction.

Hi all,

I ordered a new system in anticipation for 20.04 release. It arrived on release day and works great.

However, I am having issues configuring multiple displays. I cannot for the life of me get my 2 side vertical monitors to  work via the second GPU.

I apologise in advance for the Windows dual boot. I have to have it for After Effects unfortunately.

**Firstly, my configuration:**

* AMD Ryzen 3950x
* Asus WS X570 ACE Pro
* 128gb Corsair Vengeance DDR 3600
* GPU 1 (center screen): Asus Dual Evo RTX 2070 SUPER
* GPU 2 (side monitors): EVGA GTX 970 SC
* Boot Drive 1 (Windows): 500gb Samsung Evo 970 M2
* Boot Drive 2 (Pop OS): 500gb Samsung Evo 970 M2
* Additional Drives: 2x 500gb Samsung Evo 870 SSD (RAID 0). 2x1tb Seagate storage drives.

**Monitor Configuration:**

* Left Screen: 24" 1920x1200 - HP LCD - Vertical
* Center Screen: 43" 4K - LG 43" UD79 - Portrait
* Right Screen: 24" 1920x1200 - HP LCD - Vertical

**Software Config:**

* Pop OS 20.04 - Up to date as of 09:00 GMT on 04/05/2020. Nvidia version installed.
* Nvidia 440 drivers.
* Realtek and Intel bonded in Pop OS and teamed in Windows. Works fine in both OS.
* No additional software installed currently.

**The issue:**

1. Pop OS does not recognise the two side monitors, despite seeing the 970.
2. Nvidia xserver does see the external monitors. When I enable them in xserver (and write my changes to the x11 config), they show the mouse but nothing else. Pop OS still won't see them in display configuration.
3. During troubleshooting I tried Xinerama on a whim to see if it would work. Now it won't launch at all after login. I assume i'll now need to boot a live USB to fix my X11 config?

&#x200B;

Perhaps i'm being dim but this configuration worked fine on my old system with a similar card configuration (dual 970) in Ubuntu 18.04 with the same Nvidia 440 drivers without any messing about.

Can anyone point me in the right direction?

Thanks!

EDIT: xorg.conf

    # nvidia-settings: X configuration file generated by nvidia-settings
    # nvidia-settings:  version 440.64
    
    
    Section "ServerLayout"
        Identifier     "Layout0"
        Screen      0  "Screen0" 1200 0
        Screen      1  "Screen1" 5040 0
        Screen      2  "Screen2" 0 0
        InputDevice    "Keyboard0" "CoreKeyboard"
        InputDevice    "Mouse0" "CorePointer"
        Option         "Xinerama" "0"
    EndSection
    
    Section "Files"
    EndSection
    
    Section "Module"
        Load           "dbe"
        Load           "extmod"
        Load           "type1"
        Load           "freetype"
        Load           "glx"
    EndSection
    
    Section "InputDevice"
    
        # generated from default
        Identifier     "Mouse0"
        Driver         "mouse"
        Option         "Protocol" "auto"
        Option         "Device" "/dev/psaux"
        Option         "Emulate3Buttons" "no"
        Option         "ZAxisMapping" "4 5"
    EndSection
    
    Section "InputDevice"
    
        # generated from default
        Identifier     "Keyboard0"
        Driver         "kbd"
    EndSection
    
    Section "Monitor"
    
        # HorizSync source: edid, VertRefresh source: edid
        Identifier     "Monitor0"
        VendorName     "Unknown"
        ModelName      "LG Electronics LG Ultra HD"
        HorizSync       30.0 - 135.0
        VertRefresh     56.0 - 61.0
        Option         "DPMS"
    EndSection
    
    Section "Monitor"
    
        # HorizSync source: unknown, VertRefresh source: unknown
        Identifier     "Monitor1"
        VendorName     "Unknown"
        ModelName      "HP ZR24w"
        HorizSync       24.0 - 80.0
        VertRefresh     59.0 - 61.0
        Option         "DPMS"
    EndSection
    
    Section "Monitor"
    
        # HorizSync source: unknown, VertRefresh source: unknown
        Identifier     "Monitor2"
        VendorName     "Unknown"
        ModelName      "HP ZR24w"
        HorizSync       24.0 - 80.0
        VertRefresh     59.0 - 61.0
        Option         "DPMS"
    EndSection
    
    Section "Device"
        Identifier     "Device0"
        Driver         "nvidia"
        VendorName     "NVIDIA Corporation"
        BoardName      "GeForce RTX 2070 SUPER"
        BusID          "PCI:10:0:0"
    EndSection
    
    Section "Device"
        Identifier     "Device1"
        Driver         "nvidia"
        VendorName     "NVIDIA Corporation"
        BoardName      "GeForce GTX 970"
        BusID          "PCI:11:0:0"
        Screen          0
    EndSection
    
    Section "Device"
        Identifier     "Device2"
        Driver         "nvidia"
        VendorName     "NVIDIA Corporation"
        BoardName      "GeForce GTX 970"
        BusID          "PCI:11:0:0"
        Screen          1
    EndSection
    
    Section "Screen"
        Identifier     "Screen0"
        Device         "Device0"
        Monitor        "Monitor0"
        DefaultDepth    24
        Option         "Stereo" "0"
        Option         "nvidiaXineramaInfoOrder" "DFP-3"
        Option         "metamodes" "nvidia-auto-select +0+0"
        Option         "SLI" "Off"
        Option         "MultiGPU" "Off"
        Option         "BaseMosaic" "off"
        SubSection     "Display"
            Depth       24
        EndSubSection
    EndSection
    
    Section "Screen"
        Identifier     "Screen1"
        Device         "Device1"
        Monitor        "Monitor1"
        DefaultDepth    24
        Option         "Stereo" "0"
        Option         "metamodes" "DVI-I-1: nvidia-auto-select +0+0 {rotation=left, AllowGSYNC=Off}"
        Option         "SLI" "Off"
        Option         "MultiGPU" "Off"
        Option         "BaseMosaic" "off"
        SubSection     "Display"
            Depth       24
        EndSubSection
    EndSection
    
    Section "Screen"
        Identifier     "Screen2"
        Device         "Device2"
        Monitor        "Monitor2"
        DefaultDepth    24
        Option         "Stereo" "0"
        Option         "nvidiaXineramaInfoOrder" "DFP-1"
        Option         "metamodes" "HDMI-0: nvidia-auto-select +0+0 {rotation=right, AllowGSYNC=Off}"
        Option         "SLI" "Off"
        Option         "MultiGPU" "Off"
        Option         "BaseMosaic" "off"
        SubSection     "Display"
            Depth       24
        EndSubSection
    EndSection

&#x200B;

ActIII03, 2020-06-06:
I'm having issues with configuration not persisting after reboot. 
I tried the following:
```
$sudo nvidia-setting
```
Then I configured my display setting and clicked "Save to X configuration File" after "Apply." When I reboot, changes do not persist.

I also tried:
```
$sudo nvidia-setting --load-config-only
```

I have my settings saved in <code>/etc/X11/xorg.conf.d/20-nvidia.conf</code>. Here are the permissions :
```
-rwxr-xr-x  1 root root 1698 Jun  5 16:49 20-nvidia.conf
```

20-nvidia-conf settings:
```
# nvidia-settings: X configuration file generated by nvidia-settings
# nvidia-settings:  version 440.64

Section "ServerLayout"
    Identifier     "Layout0"
    Screen      0  "Screen0" 0 0
    InputDevice    "Keyboard0" "CoreKeyboard"
    InputDevice    "Mouse0" "CorePointer"
    Option         "Xinerama" "0"
EndSection

Section "Files"
EndSection

Section "InputDevice"
    # generated from default
    Identifier     "Mouse0"
    Driver         "mouse"
    Option         "Protocol" "auto"
    Option         "Device" "/dev/psaux"
    Option         "Emulate3Buttons" "no"
    Option         "ZAxisMapping" "4 5"
EndSection

Section "InputDevice"
    # generated from default
    Identifier     "Keyboard0"
    Driver         "kbd"
EndSection

Section "Monitor"
    # HorizSync source: edid, VertRefresh source: edid
    Identifier     "Monitor0"
    VendorName     "Unknown"
    ModelName      "DELL U2415"
    HorizSync       30.0 - 83.0
    VertRefresh     49.0 - 61.0
    Option         "DPMS"
EndSection

Section "Device"
    Identifier     "Device0"
    Driver         "nvidia"
    VendorName     "NVIDIA Corporation"
    BoardName      "GeForce GTX 1060 6GB"
EndSection

Section "Screen"
    Identifier     "Screen0"
    Device         "Device0"
    Monitor        "Monitor0"
[... 52 lines cut ...]
/0/100/1b                     bridge         200 Series PCH PCI Express Root Port #17
/0/100/1c                     bridge         200 Series PCH PCI Express Root Port #1
/0/100/1c.2                   bridge         200 Series PCH PCI Express Root Port #3
/0/100/1c.2/0      wlp4s0     network        Wireless 3160
/0/100/1c.4                   bridge         200 Series PCH PCI Express Root Port #5
/0/100/1d                     bridge         200 Series PCH PCI Express Root Port #9
/0/100/1d/0                   storage        NVMe SSD Controller SM981/PM981/PM983
/0/100/1f                     bridge         200 Series PCH LPC Controller (Z270)
/0/100/1f.2                   memory         Memory controller
/0/100/1f.3                   multimedia     200 Series PCH HD Audio
/0/100/1f.4                   bus            200 Series/Z370 Chipset Family SMBus Controller
/0/100/1f.6        enp0s31f6  network        Ethernet Connection (2) I219-V
/0/1                          system         PnP device PNP0c02
/0/2                          communication  PnP device PNP0501
/0/3                          system         PnP device PNP0c02
/0/4                          system         PnP device PNP0c02
/0/5                          system         PnP device PNP0b00
/0/6                          generic        PnP device INT3f0d
/0/7                          system         PnP device PNP0c02
/0/8                          system         PnP device PNP0c02
/0/9                          system         PnP device PNP0c02
/0/a                          system         PnP device PNP0c02
```

Thanks for your help

WereDev, 2020-08-29:
@ActIII03 I'm not sure how to solve @OliverGee 's problem, but the settings not being saved for multiple monitors was a problem I had too.  Oddly, it didn't seem to be X11 or driver related, updating my linux kernel seemed to fix the problem along with disabling the HiDPI daemon.  Also as you're using an NVidia card, I wouldn't go above the 5.7.x kernels.  Kernel 5.8.x requires NVidia driver version 450.x which doesn't have a stable release yet; at least not on the Pop_OS store.

OliverGee, 2020-08-31:
Just an update;

I never resolved this issue. It seems the issue is an nvidia one, whereby monitors on a second GPU cannot be added to the same X session, resulting in them displaying but being unusable.

In the end I gave up and put all the monitors on the 2070 Super.