pop-os/nvidia-graphics-drivers #59: gdm gets stuck with nvidia-driver

Blackwidow-sudo, 2020-09-24:
I just wanted to install PopOS and everything was ok in the install process.

But after rebooting i got a blank screen.
I booted into a live-enviroment to see what boot.log says:
[  OK  ] Started Gnome Display Manager

Then i chroot into my partition and purged the nvidia-driver. After a reboot everything worked fine.
I did a full-upgrade and installed system76-driver-nvidia again.
But after that, i get a black screen again.

Is the nvidia driver broken?

jacobgkau, 2020-09-24:
Sounds like a possible duplicate of https://github.com/pop-os/pop/issues/1254 based on the screenshot in that issue (your additional detail is appreciated.)

The NVIDIA driver we're shipping is working on our products. What hardware are you using, including GPU and display?

Blackwidow-sudo, 2020-09-29:
> Sounds like a possible duplicate of [pop-os/pop#1254](https://github.com/pop-os/pop/issues/1254) based on the screenshot in that issue (your additional detail is appreciated.)
> 
> The NVIDIA driver we're shipping is working on our products. What hardware are you using, including GPU and display?
My Hardware:
- ASRock Z77 Pro3 Motherboard with 16GB DDR3
- Intel i5 3570K
- MSI GTX 1060 Gaming X 6GB
- Screen: Acer G246HL

Little nuance i forgot to say:
Its actually the second reboot after everything went wrong.
After the installation comes the first reboot where you get to create your user account.
I created my account and launched into the desktop. But when i reboot after that, i get a blank screen with the nvidia drivers.

Blackwidow-sudo, 2020-11-03:
Just a quick update:

I recently switched from MBR to GPT and installed PopOS again in UEFI-Mode.
This time it works. idk why though. its the same version on the same system.
Only thing changed is the different boot into it.