pop-os/nvidia-graphics-drivers #61: Hybrid mode: performance degrades when using only an external monitor

leviport, 2020-10-06:
If I am running oryp6 in hybrid mode, connect an external display, then close the lid, the external display is set as the primary as expected, but the performance of the desktop on the external display is basically unusable. The mouse cursor moves around at the usual speed, but everything else seems to be lagging behind by one or two seconds.

Nvidia mode still works as expected, so that can be used for the time being if someone wants to use external displays with the internal turned off.

leviport, 2020-10-07:
This is still present on 455.28. 

On oryp6, I'm also noticing that using two 1440p external displays makes the performance a bit choppy as well. It's still usable, but the slowness is noticeable. It seems that switching to Nvidia mode still gives the best performance with external displays.

leviport, 2020-10-09:
I've also noticed that applications that launch using Nvidia graphics, such as Steam games, do not suffer the same reduction in performance.

ArjunSingh99, 2020-10-31:
I have a similar issue, but on an HP. When running on Hybrid mode, the external display doesn't use the discrete GPU. I have to specifically switch to Nvidia graphics before using an external display so as to make it usable.

leviport, 2020-11-10:
This issue is still present on 455.38

miguelangelnieto, 2021-02-23:
Same problem here. I can only get good desktop experience in "NVIDIA Graphics" mode. In hybrid with external screen it lags and CPU consumes 30% all the time.

raf2k07, 2021-04-03:
This is still an issue on my Acer Nitro 5 AN515-54 with an Nvidia 1650 GPU.

leviport, 2021-08-02:
Update: this bug is still present on the 470.57.02 driver. The performance seems about the same. There is one small improvement, however. Now when I switch off the oryp6's internal display, I don't see the same ~2 seconds of rapid flashing on the external display that I was seeing before.

leviport, 2021-09-09:
I'm testing the new [470.63 update](https://github.com/pop-os/nvidia-graphics-drivers/pull/119) and it looks like it fixes hybrid mode on external displays.

sims1253, 2021-09-14:
I just updated and the bug persists. But it seems to only hit electron applications like slack, vs-code and the signal desctop client.

raf2k07, 2021-09-16:
I updated to 470.63 and hybrid mode is still terrible for external displays, even on non-electron applications. Everything on the desktop (barring the cursor) is choppy and generally runs at low fps. 

Laptop Specifications: 
Acer Nitro 5 AN515-54
nVidia 1650
i5-9300H
16GB RAM
1TB HDD

sazconaitbit, 2021-09-20:
this is a bug in the xorg

https://gitlab.freedesktop.org/xorg/xserver/-/issues/22
https://gitlab.freedesktop.org/xorg/xserver/-/issues/1028

people there make a patch to fix it .

maybe pop os devs can help us to see how to apply it

thomas-zimmerman, 2022-05-31:
I have a report from a System76 customer on a `kudu6` system that this may be fixed by the 515.43.04 Beta driver but possibly as far back as the 510.73.05 release when using the ".run" file directly from Nvidia. He has found that the same version as packaged using the either the System76 PPA and our packaging of the 510.73.05 driver or the Ubuntu based packaging of the driver has this performance loss when on the external display

(details in support case 69153 private to protect the innocent)

HadiModarres, 2022-06-08:
https://forums.developer.nvidia.com/t/high-cpu-usage-on-xorg-when-the-external-monitor-is-plugged-in/169173

I posted on this thread a year ago, but the issue hasn't been acknowledged, so it's unlikely there's a fix in the works

mmstick, 2022-06-08:
Also relevant https://forums.developer.nvidia.com/t/nvidia-x11-driver-busy-polls-kernel-on-clock-gettime-in-a-tight-loop/216722

mmstick, 2022-06-08:
Busy loop also happens when running OpenCL tasks.
Some fix it by preloading a library that overrides `clock_gettime`.
https://github.com/mbevand/silentarmy/issues/54
https://bitcointalk.org/index.php?topic=181328.0

mmstick, 2022-10-17:
I've heard that NVIDIA 525 fixed the busy polling issue which is significantly impacting GNOME performance in Mutter. That might improve this issue too.

spunkedy, 2022-11-28:
Just insatalled the 525, chrome is still laggy on screen drags, but seems to be running slightly better than the 515. 

Thanks for the info @mmstick

LucianoAngel, 2022-12-06:
I have the same issue, present lag on a external display (NVIDIA mode)

Have clean install of Pop OS 22.04, NVIDIA GTX 1650 (mobile) and NVIDIA Driver Version: 515.65.01

If i found any solution can be explain the solution on this Issue.

leviport, 2022-12-06:
525 was just released today, if you want to give it a try. Sounds like it doesn't completely fix it, but hopefully it helps.

carywreams, 2022-12-07:
a couple of questions I don't see in prior comments: 

* is there something in the 525 release notes that suggests it will help? (comments on this issue about 510, 515, and now 525 possibly addressing the issue).
* Is there an ETA for 525 appearing in the System76 PPAs for 2204 or 2210 ? 

Just hesitant to perform an out-of-band upgrade if  the OS isn't ready to include it.

Thanks.

leviport, 2022-12-07:
When I say it was released, I mean it's released on Pop. Also, Pop 22.10 doesn't exist.

carywreams, 2022-12-07:
no (good) luck here... after an update / upgrade and the pop shop install of 525. 
six-month-old kudu6

[image]

```
k6:~> nvidia-smi
Wed Dec  7 10:55:42 2022       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.60.11    Driver Version: 525.60.11    CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0  On |                  N/A |
| N/A   40C    P8    17W /  N/A |     54MiB /  6144MiB |     40%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      3921      G   /usr/lib/xorg/Xorg                 53MiB |
+-----------------------------------------------------------------------------+
```

radupopa2010, 2022-12-20:
This is still happening 

`nvidia-driver-525/jammy,now 525.60.11-1pop0~1670285463~22.04~b24c135 amd64 [installed,automatic]`

```
lsb_release -a
No LSB modules are available.
Distributor ID:	Pop
Description:	Pop!_OS 22.04 LTS
Release:	22.04
Codename:	jammy

lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
01:00.0 VGA compatible controller: NVIDIA Corporation GA104 [Geforce RTX 3070 Ti Laptop GPU] (rev a1)

```

ricmzn, 2023-01-18:
The new `525.78.01` driver has this release note:

woook, 2023-02-25:
I'm experiencing this in nvidia mode (not tried hybrid) with `525.85.05` and a fresh install of `Pop OS 22.04`.
Turning off the built in display cause huge lag on the external monitor (although not for the mouse cursor).

carywreams, 2023-02-25:
@woook ,

> I'm experiencing this in nvidia mode (not tried hybrid) with `525.85.05` and a fresh install of `Pop OS 22.04`. Turning off the built in display cause huge lag on the external monitor (although not for the mouse cursor).

I'm taking a look at mine and do not see the excessive CPU usage with nvidia config running only the external monitor. My nvidia-smi output:

```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.85.05    Driver Version: 525.85.05    CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0  On |                  N/A |
| N/A   38C    P8    15W / 115W |    252MiB /  6144MiB |     13%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      4010      G   /usr/lib/xorg/Xorg                168MiB |
|    0   N/A  N/A      4138      G   /usr/bin/gnome-shell               81MiB |
+-----------------------------------------------------------------------------+
```

woook, 2023-02-25:
I also don't see excessive CPU but there is a 5s lag between doing something and seeing it happen on screen. I guess I might have a different type of performance degradation.

```+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.85.05    Driver Version: 525.85.05    CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0  On |                  N/A |
| N/A   34C    P8     8W /  80W |     49MiB /  8192MiB |     12%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
```

mercifulboss, 2023-03-03:
> The new `525.78.01` driver has this release note:
> 
> > Fixed excess CPU usage in hybrid graphics configurations where an external display is connected to an NVIDIA discrete GPU and configured as a PRIME Display Offload sink (also known as "Reverse Prime").
> 
> But after updating to the new driver and rebooting, the issue is still present. I'm using a Dell laptop with a 3060 though, so I'm not sure if the fix is hardware dependent.
> 
> Also, the issue only happens on Xorg. Wayland runs normally when using only the external display in hybrid mode, with applications able to use either GPU (but has the usual Nvidia Wayland issues like no night light etc).

I have the same issue with a 2060. I am getting this issue on every driver after the 520 driver. Currently on the 525.89.02 driver and this issue persists.

jackleberryfin, 2023-03-09:
I've just bought a Gigabyte Aorus 5 SE4 i7 12700H with Nvidia RTX 3070. Running PopOs I am experiencing this problem also. Nvidia Driver 525.85.05

I took a dump of the nvidia-smi -a output during normal operation and while the lid is closed and noted the key differences in the form <good> -> <bad>

PCI/ GPU Link Info/ PCIe Generation/ Current 2 -> 1
PCI/ GPU Link Info/ PCIe Generation/ Device Current 2 -> 1
PCI/ Tx Throughput 147000 KB/s -> 0 KB/s
PCI/ Rx Throughput 147000 KB/s -> 0 KB/s
Performance State P5 -> P8

Nvidia lists the Performance States from P0 - P15 with P0 being max performance and P15 being lowest performance. 

Naturally the memory usage, utilization and power draw all dropped also.

I have also enabled persistence mode via the daemon to see if this was caused by the driver being unloaded while the system juggles displays. This did not make any difference. 

My external monitor is a 49" AOG AGON 120hz (5120x1440). The issue being discussed does not seem to be reproducible on the USB-C output of the laptop. However, the USB-C output doesn't support/recognize my native resolution or refresh rate making it pretty much useless. I am assuming that this works becase the USB-C is directly connected to the CPU which is doing the handover... however the mDP port is connected directly to the GPU and when the lid closes the PCIe lanes are forced into low-power mode or some such. My theory is that its probably a power management bug. 

Hopefully some of this information helps solve the issue.

loafylemon, 2023-04-15:
The problem still exists with the newest 525.105.17 driver. It only appears when the laptop's internal screen is off, and once it's turned on, the problem goes away. It's weird because the mouse pointer and animations aren't impacted by this.

```OS: Pop!_OS 22.04 LTS x86_64 
Host: ROG Strix G713QR_G713QR 1.0 
Kernel: 6.2.6-76060206-generic 
Uptime: 27 mins 
Packages: 2447 (dpkg), 73 (flatpak) 
Shell: bash 5.1.16 
Resolution: 1920x1080, 1920x1080 
DE: GNOME 42.5 
WM: Mutter 
WM Theme: Pop 
Theme: Pop-dark [GTK2/3] 
Icons: Pop [GTK2/3] 
Terminal: gnome-terminal 
CPU: AMD Ryzen 9 5900HX with Radeon Graphics (16) @ 3.300GHz 
GPU: NVIDIA GeForce RTX 3070 Mobile / Max-Q 
GPU: AMD ATI 06:00.0 Cezanne 
Memory: 4703MiB / 31502MiB

Sat Apr 15 20:37:33 2023       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.105.17   Driver Version: 525.105.17   CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0  On |                  N/A |
| N/A   45C    P3    21W / 130W |     46MiB /  8192MiB |     23%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1316      G   /usr/lib/xorg/Xorg                 45MiB |
+-----------------------------------------------------------------------------+```

carywreams, 2023-04-15:
This problem is like chasing a chicken. Yes, the XOrg burn is gone in hybrid mode (updated popos 2204; nvidia 525.105.17), but I'm seeing the "closed laptop screen"-induced sluggishness, too. Terminal window typing is **very slow**, while the mouse remains as responsive as usual. Additionally, re-opening the laptop so that the screen is active eliminates the sluggishness.

carywreams, 2023-04-15:
and a fun variant. I shut the lid, then rebooted from CLI, in order to boot with the lid closed, in hybrid mode, with the external monitor connected. It took a bit for the laptop to figure out the external monitor needed the login screen. Entering password was sluggish. But after entering and accepting my password, the laptop went to sleep (attached file, line #1386: PM Suspend). All of the ACPI errors that follow that line I usually see on my console resuming from a suspend. Maybe related, maybe not.  If _I knew_ it was definitely not related, I wouldn't bother with it here.

[dmesg.txt](https://github.com/pop-os/nvidia-graphics-drivers/files/11240876/dmesg.txt)

carywreams, 2023-04-17:
@thomas-zimmerman , perhaps.
is reverting from wayland a matter of re-subbing false/true ?

also - what is it you'd like me to look for? just sluggishness during login ?

thomas-zimmerman, 2023-04-17:
The session type when picked at the GDM screen is "sticky" but is easy to swap between with just a login/logout. I have found that a wayland session does not seem to have the very slow bug when on external display (and internal display disabled). This looks more like a bad interaction between the Nvidia driver and Xorg, not elsewhere in the stack. 

I've seen this both in hybrid and Nvidia mode for the Nvidia driver. It seems to happen most often on a HDMI connection to the external monitor.

carywreams, 2023-04-18:
@thomas-zimmerman ,

## Brief

...with laptop lid closed, ...

+   display manager is sluggish under HYBRID mode, but not NVIDIA mode
    +   user selection
    +   gear for X11 / Pop on Wayland selection
    +   password entry
+   terminal window typing **IS NOT** sluggish under either mode
+   HDMI monitor slow to receive login screen, both modes
+   HDMI monitor cycles through "no connection" and "power save" messages after reboots and logins, both modes
+   tiling under wayland is odd; very likely a different issue

## Verbose

### Initial Condition

+   X11
+   nvidia mode
+   second display config is separate, as opposed to "mirrored" screens
+   connected via HDMI cable
+   system config to not sleep while plugged into AC power, which is the case here

edited /etc/gdm3/custom.conf as suggested

### Switch: NVIDIA to HYBRID

``system76-power graphics hybrid``
closes lid, external monitor attached and active
``shutdown -r now``

+   HDMI monitor blank screen
+   HDMI monitor displays "no signal, check connection"
+   HDMI monitor displays "going to power save mode"
+   ultimately login window appears on HDMI monitor (laptop lid remains closed)
+   select user id to enter password
+   gear option appears, used to change to pop on wayland
+   entering password characters is sluggish, as before
+   password accepted, login starts
+   fans are heard (this is rare for my usage) spinning up; remain on for a bit
+   HDMI monitor blank screen
+   HDMI monitor displays "no signal, check connection"
+   HDMI monitor displays "going to power save mode"
+   after 60s, screen still blank
+   moves mouse to "awaken"
+   screen awakes after about 15 seconds - but no password prompt as if had been suspended
+   terminal window / top shows only gnome-shell  consuming considerable CPU%
+   typing in terminal window **IS NOT** sluggish
+   see notes on tiling behavior, below

### Switch: HYBRID to NVIDIA

``system76-power graphics nvidia``
``shutdown -r now``

+   HDMI monitor blank screen
+   moving mouse during this period made no difference
+   HDMI monitor displays "no signal, check connection"
+   HDMI monitor displays "going to power save mode"
+   ultimately login window appears on HDMI monitor (laptop lid remains closed)
+   select user id to enter password
+   gear option appears, used to confirm pop on wayland still selected
+   entering password characters **IS NOT** sluggish
+   password accepted, login starts
+   fans are not heard
+   HDMI monitor blank screen
+   HDMI monitor displays "no signal, check connection"
+   HDMI monitor displays "going to power save mode"
+   after 60s, screen still blank
+   moves mouse to "awaken"
+   screen awakes after about 15 seconds - but no password prompt as if had been suspended
+   typing in the terminal window **IS NOT** sluggish
+   see notes on tiling behavior, below

Thinking the sluggish entry of password characters may have been due to X11  | Wayland selection, I repeat the switch to hybrid mode. It was not. The sluggishness of the Display Manager followed HYBRID mode:

### Switch: NVIDIA to HYBRID

``system76-power graphics hybrid``
``shutdown -r now``

+   HDMI monitor blank screen
+   HDMI monitor displays "no signal, check connection"
+   HDMI monitor displays "going to power save mode"
+   ultimately login window appears on HDMI monitor (laptop lid remains closed)
+   select user id to enter password (switch from UID display to password input field is also sluggish)
+   use gear option; confirm pop on wayland still selected
+   entering password characters **IS AGAIN sluggish**
+   password accepted, login starts
+   fans are not heard
+   HDMI monitor blank screen
+   HDMI monitor displays "no signal, check connection"
+   HDMI monitor displays "going to power save mode"
+   after 60s, screen still blank
+   moves mouse to "awaken"
+   screen awakes after about 15 seconds - but no password prompt as if had been suspended
+   typing in the terminal window **IS NOT** sluggish
+   see notes on tiling behavior, below

## Odd Tiling Behavior

While likely a "behavior" related to wayland and pop os, rather than nvidia and X11,
I am recording the tiling issues here in the event th at both issues are somehow
interrelated.

Additional notes on screen tiling:

+   though 'tiling' option set, terminal window did not 100% tile the screen: took right-half of screen, top to bottom
+   tiling of terminal window possible with double-click on window chrome
+   closed terminal widow
+   opened google-chrome to check out tiling - also did not 100% tile the screen; occupied same space as terminal window
+   double clicking on window chrome did not have any effect.
+   ultimately, only three options were possible: left half, right half, top 3/4's (did not try for bottom 3/4's; didn't care)
+   Screen Grabs, cycling through opening and closing of laptop lid
    +   [start](https://user-images.githubusercontent.com/409739/232761600-be037439-8a00-4c02-8d8a-cb17b7163bb6.png) (three windows opened and positioned automatically)
    +   [open laptop lid](https://user-images.githubusercontent.com/409739/232761595-6bd0bc1c-e82e-468c-bfa3-e4012329e308.png), all window move to laptop screen:
    +   [close laptop lid](https://user-images.githubusercontent.com/409739/232761588-89278c45-dd89-42a4-a7e6-cf7d89440bc9.png), all windows move back to HDMI screen
+   Other odd window/tiling/menu behaviors were observed under Wayland, but leaving it here for now.

p0t4t0sandwich, 2023-08-29:
Fresh install today, came up with the same issue again.
Settings said they were using Nvidia graphics, I ran this regardless:
`system76-power graphics nvidia`
After a quick reboot all my electron-based apps running proper again

SneWs, 2023-10-09:
I'll add my observations as well, running a 3070 mobile chip with a 11th gen i9 in a ASUS ROG Zephyrus, using X11 everything becomes super laggy once I plug in my external monitor over Thunderbolt but as soon as I enabled and logged in with Wayland everything is running smooth. 

So here's hoping Wayland and NVIDIA becoming better and better together :)

mfbruno, 2024-01-19:
That's still a problem. I have an AN515-47-R9ES and the problem persists when I close the notebook lid and the external monitor is slow except for the mouse pointer.

MlopesD, 2024-01-30:
it's solved for me. I just selected "integrated graphics", restarted, this it didn't even recognise my 2 external monitors, then I selected again "nvidia graphics", restarted and boom, everything working very smothly.

Basically I changed and changed back the settings. they were originally on "nvidia graphics" and it was painfully laggy.

asus zenbook 15 pro 
Core™ i7-10870H
GeForce GTX 1650 Ti Mobile, driver 

Pop! OS 22.04 LTS
nvidia-driver 470.199.02

[image]

leviport, 2024-01-30:
@MlopesD  The bug is specifically about using Hybrid mode. NVIDIA mode has historically been fine with external displays on machines that have been affected by the bug. Glad to hear the NVIDIA mode workaround is still effective though.

MlopesD, 2024-01-30:
@leviport What I'm trying to explain is that they were originally on Nvidia graphics, but it wasn't working, it was very laggy.

I believe they appear as default on "Nvidia graphics", but it's a false statement, they are simply on another
setting. 

this was confusing at the beginning because, for everyone, that setting was working, but for me
that setting was the default one and was already selected, but not working at all.
So I thought of doing the change, and then do the change back, and it worked.

checkinindza, 2024-02-07:
Don't know if this will help in solving the issue, but I'll just write about my own situation.

Currently I have an HP Omen RTX 3060 laptop with 1440p 165Hz internal display. Without an external monitor I have no issues whatsoever. Connecting the external display introduces heavy lag, pretty much becomes unusable. But it only happens on the external monitor, move the app to the internal display and all issues are gone.

X11 + NVIDIA Only or Wayland + Hybrid Mode seems to solve most of the issues, at least when doing basic stuff. Can't test Wayland + NVIDIA Only, because for some reason I get stuck in login screen loop. 

But in games, doesn't matter which mode I choose, on external monitor even though fps stays at 60, it feels more like 25-30fps. And again, if I disconnect the external monitor and use only the internal, issues are gone. 

I'm on Pop_OS! 22.04 NVIDIA LTS with NVIDIA Proprietary Driver 545.29.06

Kidlike, 2024-02-28:
I am also affected by exactly the same situation as described by @checkinindza. I have a Legion 7 with NVidia 3070, and an AMD Ryzen 7 5800H. Nvidia drivers 545, kernel 6.7.5.

I think the OS doesn't matter... this issue is unique to AMD + NVidia + hybrid combo, and it seems to come from some driver (probably amdgpu).

I have this issue for 2 years now (basically ever since I got the laptop), and at this point I think I will just never buy NVidia again...

checkinindza, 2024-02-28:
> I am also affected by exactly the same situation as described by @checkinindza. I have a Legion 7 with NVidia 3070, and an AMD Ryzen 7 5800H. Nvidia drivers 545, kernel 6.7.5.
>
> I think the OS [...]

OS does matter. I hopped to EndeavourOS with KDE and the situation improved quite a lot. While doing basic usage I have no issues whatsoever and gaming improved quite a bit. It ain't perfect, that's for sure. But it's much better than it was on Pop_OS!.

d1mmu-dev, 2024-04-05:
Same situation here, Nvidia  + AMDGpu causes freezes of external display, on nvidia developers forum theres topic about bug in drivers:
https://forums.developer.nvidia.com/t/external-monitor-freezes-when-using-dedicated-gpu/265406/31

for me helped using only Nvidia GPU and set it to max performance in nvidia settings...

ahoneybun, 2024-05-14:
@Kidlike @checkinindza @d1mmu-dev can you confirm that it happens on the 550.67 driver (latest currently)?

d1mmu-dev, 2024-05-15:
> @Kidlike @checkinindza @d1mmu-dev can you confirm that it happens on the 550.67 driver (latest currently)?

after update, in hybrid graphics mode and using HDMI looks like there is no freezes at all
[image]

vkSegfault, 2024-05-27:
Three cents from me:
1. fresh install of Pop, AMD + Nvidia (550.67) and still huge lag on external monitor **but only on X11**, when turning on Wayland problem disappears
2. I verified Zorin 17.1 lately and there is no problem at all on X11 session (so this is Pop os specific) - but battery is draining really fast
3. No problem on Cosmic DE!

I know that current Gnome version is so outdated so I assume latest one + latest wayland works even better nowadays, there is also benefit of much better battery life on wayland (noticeable on old Gnome that Pop uses) and even better on Cosmic DE

closed 2026-01-07 by leviport