pop-os/nvidia-graphics-drivers #62: The current Nvidia driver 455.23 breaks some games with Proton

zelfir, 2020-10-06:
Running Pop_OS 20.04 from an optimus enabled laptop.
For example, The Elder Scrolls Online from Steam is now freezing on the connection screen since I upgraded from the 440.100 to the proposed 455.23.
I'm pretty sure it's the driver because using the latest stable from Nvidia (450.80.02) seems to fix the issue.
Is there a way to rollback to the 440.100 until you propose the 450.80.02 from your repo ?

mmstick, 2020-10-06:
Does `nvidia-smi` report 455.23?

zelfir, 2020-10-06:
Yep, 455.23.04 to be exact.

Redbatman89, 2020-10-07:
If you have timeshift then use that to roll back. 455.23.04 was giving me issues too with vulkan shaders taking so long to process.

jacobgkau, 2020-10-07:
455.28 has been released. @leviport found that it reduced Vulkan shader processing time. The NVIDIA changelog also indicates that some bugs with Vulkan applications (including a Steam Play title) were fixed.

@zelfir @Redbatman89 Can you check if you're still experiencing issues after upgrading to 455.28 and rebooting?

zelfir, 2020-10-07:
I can't update actually because http://ppa.launchpad.net/wireguard/wireguard/ubuntu Focal Release is missing the Release file :/

jacobgkau, 2020-10-07:
> I can't update actually because http://ppa.launchpad.net/wireguard/wireguard/ubuntu Focal Release is missing the Release file :/

That does not sound related to this issue, and that PPA is not a default repository included in Pop. Furthermore, it looks like that PPA has been decommissioned: https://launchpad.net/~wireguard 

You can remove it by editing your `/etc/sources.list` or `/etc/sources.list.d` files, or by using the GUI in Repoman, as described here: https://support.system76.com/articles/manage-repos-pop/

zelfir, 2020-10-07:
Indeed, it's not related to the issue but it was just cancelling the apt update command. Anyway I figured out and managed to update to 455.28. I rebooted and tested with TESO again but unfortunately the problem is still there :(

zelfir, 2020-10-08:
Well, it seems the issue was also linked to the shader cache used by the game.
Once Steam totally rebuild the shader cache with the new driver,  it solved the issue.

You can close this ticket :)

leviport, 2020-10-08:
Glad to hear it, thanks for letting us know!

closed 2020-10-08 by leviport

Redbatman89, 2020-10-08:
So then the new driver works fine then?

zelfir, 2020-10-08:
Yes