pop-os/nvidia-graphics-drivers #64: 455 breaks multi-monitor usage when laptop screen is turned off

mreinigjr, 2020-10-12:
Hello,

TL;DR
Can you please support nvidia-driver-450 from the official pop_os ppa?

I am on a Lenovo Thinkpad Extreme (gen 1...I think) with a GeForce GTX 1050 Ti Mobile.  I am running Pop!_OS 20.04 LTS.

After upgrading to 455 on 10/9 my computer would freeze once I disabled my laptop's screen so that I only use my connected monitors.  I do this by simply closing the lid.  After closing the lid, the connected monitors would lose their connection and the laptop screen would not be responsive.  Nothing I tried, including attempts to get to tty4, would revive my laptop. I have also confirmed that shutting off the laptop monitor from within the pop_os display settings would result in the same issue.  After digging into the syslogs, here is the last entry written.

```
Oct  9 16:26:44 mreinigjr /usr/lib/gdm3/gdm-x-session[2283]: (II) NVIDIA(0): Setting mode "DP-1.2: nvidia-auto-select @1920x1080 +1920+0 {AllowGSYNC=Off, ViewPortIn=1920x1080, ViewPortOut=1920x1080+0+0}"
Oct  9 16:26:47 mreinigjr kernel: [  584.685145] nvidia-modeset: WARNING: GPU:0: Lost display notification (0:0x00000000); continuing.
Oct  9 16:26:52 mreinigjr kernel: [  589.888657] nvidia-modeset: ERROR: GPU:0: Idling display engine timed out: 0x0000987d:0:0:442
```
I then proceed to downgrade nvidia to 450.  However, I quickly realized that the pop_os ppa for nvidia-driver-450 actually points to nvidia-driver-455, or at least that was what was being installed when I ran `sudo apt-get install nvidia-driver-450`.

I did find a solution, but it certainly is not ideal and I would prefer to get back to using the ppa.  My solution was to install the pop_os 455 drivers, then go to nvidia site and run the 450 installer.  It will fail to load the DKMS kernel module, but the install did work.  After several reboots, the change persists (I am seeing 450.80.02 in the x server settings) and the issue I had was resolved with the 450 drivers.

So long story short, would it be possible to support a build for the 450 drivers?  Also, if you have any other recommendations for me to fix the described issue and still use the 455 drivers, please let me know.  I did also try to downgrade to 440, but 455 kept getting installed again.  To reinstall the 440 drivers, I simply did a `sudo apt-get install nvidia-driver-440`, so if I remember correctly, I did not run a purge prior to installing 440, but I did prior to install 450 from the ppa, which again, just brought in the 455 drivers.  So not running the purge prior to attempting to downgrade to 440 may have been the reason.  I do see where 440 is still on the ppa, but 450 is not.

Just in case you are interested, here are my nvidia install logs.

```
nvidia-installer log file '/var/log/nvidia-installer.log'
creation time: Sun Oct 11 13:26:48 2020
installer version: 450.80.02

PATH: /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin

nvidia-installer command line:
    ./nvidia-installer

Using: nvidia-installer ncurses v6 user interface
-> Detected 12 CPUs online; setting concurrency level to 12.
WARNING: You do not appear to have an NVIDIA GPU supported by the 450.80.02 NVIDIA Linux graphics driver installed in this system.  For further details, please see the appendix SUPPORTED NV
IDIA GRAPHICS CHIPS in the README available on the Linux driver download page at www.nvidia.com.
-> Installing NVIDIA driver version 450.80.02.
-> An alternate method of installing the NVIDIA driver was detected. (This is usually a package provided by your distributor.) A driver installed via that method may integrate better with y
our system than a driver installed by nvidia-installer.

Please review the message provided by the maintainer of this alternate installation method and decide how to proceed:

The NVIDIA driver provided by Ubuntu can be installed by launching the "Software & Updates" application, and by selecting the NVIDIA driver from the "Additional Drivers" tab.

(Answer: Continue installation)
-> Would you like to register the kernel module sources with DKMS? This will allow DKMS to automatically build a new module, if you install a different kernel later. (Answer: Yes)
-> Install NVIDIA's 32-bit compatibility libraries? (Answer: Yes)
Looking for install checker script at ./libglvnd_install_checker/check-libglvnd-install.sh
   executing: '/bin/sh ./libglvnd_install_checker/check-libglvnd-install.sh'...
   Found libglvnd libraries: libGLESv2.so.2 libEGL.so.1 libGLX.so.0 libGL.so.1
   Found non-libglvnd libraries:
   Missing libraries: libGLESv1_CM.so.1 libOpenGL.so.0
   libglvnd appears to be installed.
Will not install libglvnd libraries.
-> Skipping GLVND file: "libOpenGL.so.0"
-> Skipping GLVND file: "libOpenGL.so"
-> Skipping GLVND file: "libGLESv1_CM.so.1.2.0"
-> Skipping GLVND file: "libGLESv1_CM.so.1"
-> Skipping GLVND file: "libGLESv1_CM.so"
-> Skipping GLVND file: "libGLESv2.so.2.1.0"
-> Skipping GLVND file: "libGLESv2.so.2"
-> Skipping GLVND file: "libGLESv2.so"
-> Skipping GLVND file: "libGLdispatch.so.0"
-> Skipping GLVND file: "libGLX.so.0"
-> Skipping GLVND file: "libGLX.so"
-> Skipping GLVND file: "libGL.so.1.7.0"
-> Skipping GLVND file: "libGL.so.1"
-> Skipping GLVND file: "libGL.so"
-> Skipping GLVND file: "libEGL.so.1.1.0"
-> Skipping GLVND file: "libEGL.so.1"
-> Skipping GLVND file: "libEGL.so"
-> Skipping GLVND file: "./32/libOpenGL.so.0"
-> Skipping GLVND file: "libOpenGL.so"
-> Skipping GLVND file: "./32/libGLdispatch.so.0"
-> Skipping GLVND file: "./32/libGLESv2.so.2.1.0"
-> Skipping GLVND file: "libGLESv2.so.2"
-> Skipping GLVND file: "libGLESv2.so"
-> Skipping GLVND file: "./32/libGLESv1_CM.so.1.2.0"
-> Skipping GLVND file: "libGLESv1_CM.so.1"
-> Skipping GLVND file: "libGLESv1_CM.so"
-> Skipping GLVND file: "./32/libGL.so.1.7.0"
-> Skipping GLVND file: "libGL.so.1"
-> Skipping GLVND file: "libGL.so"
-> Skipping GLVND file: "./32/libGLX.so.0"
-> Skipping GLVND file: "libGLX.so"
-> Skipping GLVND file: "./32/libEGL.so.1.1.0"
-> Skipping GLVND file: "libEGL.so.1"
-> Skipping GLVND file: "libEGL.so"
WARNING: Unable to determine the path to install the libglvnd EGL vendor library config files. Check that you have pkg-config and the libglvnd development libraries installed, or specify a path with --glvnd-egl-config-path.
Will install libEGL vendor library config file to /usr/share/glvnd/egl_vendor.d
-> Searching for conflicting files:
-> done.
-> Installing 'NVIDIA Accelerated Graphics Driver for Linux-x86_64' (450.80.02):
   executing: '/sbin/ldconfig'...
-> done.
-> Driver file installation is complete.
-> Installing DKMS kernel module:
-> done.
ERROR: Unable to load the 'nvidia-drm' kernel module.
ERROR: Installation has failed.  Please see the file '/var/log/nvidia-installer.log' for details.  You may find suggestions on fixing installation problems in the README available on the Linux driver download page at www.nvidia.com.
```

Thank you for the help and for the great work system76 is doing!

leviport, 2020-10-12:
Can you tell us which graphics mode you were in when you saw this issue? It sounds similar to an issue I encountered when using external displays in hybrid mode: https://github.com/pop-os/nvidia-graphics-drivers/issues/61. Does the issue disappear if you switch to Nvidia mode?

mreinigjr, 2020-10-13:
Hi @leviport, thanks for your input on this ticket and in #65.  I think the direction here is clear, so I am going to close this issue and manage the issue myself until the versions that popos is offering for nvidia coincide with what is compatible with my machine.  Thanks again!

closed 2020-10-13 by mreinigjr