pop-os/nvidia-graphics-drivers #65: PopOS ALWAYS installs the latest nvidia driver!

NinjaTurtle007, 2020-10-13:
nvidia-driver-455 was giving me issues so I did

sudo apt-get purge '*nvidia*'
sudo apt install nvidia-driver-440

Well to my surprise that installed nvidia-driver-455 again, as confirmed by the nvidia settings app.

i can see that it downloaded 450 and 440 as well but 455 is in use.

So how do we force it to install/use 440?

jacobgkau, 2020-10-13:
> So you might not want to push drivers (read somewhere this one is beta?)

Just to clarify, the version we are currently shipping (455.28) is not in beta. I believe you read that information in this other issue: https://github.com/pop-os/nvidia-graphics-drivers/issues/66, which was opened regarding version 455.23, which was a beta.

NinjaTurtle007, 2020-10-13:
Yes, my bad, sorry.

However, any easy way to revert back properly?
Or will the current latest driver be updated soon with a fix?

mreinigjr, 2020-10-13:
@NinjaTurtle007, your issue sounds a lot like mine #64.  In that issue I outline how I was able to successfully downgrade to 450.  Just be sure you have at least some nvidia driver installed before you do a reboot or shutdown.  Otherwise popos will not load and you will need to use chroot from a live cd or the recovery partition (if available) to fix the issue.  In a nutshell, the process was really simple, make sure you are in discrete mode, install popos 455 drivers, go to nvidia's website and download 450, run the installer.  Checkout my nvidia logs in #64 to see how I answered the installers questions.  Please note that the dkms module step will fail, but when you reboot, the nvidia application should state 450.  I am honestly not sure if you need to say yes to the dkms question, but I am just repeating to you the steps I followed.  **Again, if you do a reboot while you are in discrete mode and WITHOUT some version of the nvidia drivers installed, popos will not load.  If you don't know what I am talking about in regards to using chroot or the recover partition, then a complete reinstallation of popos might be needed.  So proceed with the downgrade at your own risk!**  If any of the maintainers of this repo would like to chime in, please do.

leviport, 2020-10-13:
> However, any easy way to revert back properly?

Not an easy way, no. You'd pretty much be purging it with apt, then installing (if you can find it) the version directly from Nvidia.

> Or will the current latest driver be updated soon with a fix?

Possibly, but it's hard to say with Nvidia drivers. Since it's proprietary, we're kinda at their mercy.

If you can better describe the actual issue you're having in a way that we can reproduce it, we might be able to give them a nudge. Right now this issue is extremely unclear, so editing it would make it much more helpful.

leviport, 2020-10-13:
>  In that issue I outline how I was able to successfully downgrade to 450.

It might be worth noting that the 450 driver was never released on Pop because of issues we found with it during testing. It may work for you, but I just wanted to make that clear.

mreinigjr, 2020-10-13:
Thanks @leviport, that is a great point.  I think one thing users of non-system76 hardware forget is that popos is really only heavily tested on system76 hardware and because it is primarily maintained for your hardware, using it on other hardware will likely work, but not guaranteed nor is it officially supported by system76 engineers.  Would you say that is accurate?

I would also like to add that nvidia has a tool to help you find the driver for your card:
https://www.nvidia.com/download/index.aspx?lang=en-us

NinjaTurtle007, 2020-10-13:
> @NinjaTurtle007, your issue sounds a lot like mine #64. In that issue I outline how I was able to successfully downgrade to 450. Just be sure you have at least some nvidia driver installed before [...]

It is indeed the same, I'm with the same laptop (x1 extreme gen 1), running via their pro dock 2 external ultrawide monitors. 
I didn't think someone else would have the same setup! :)

I don't close my laptop though, since I like to use it as a 3rd monitor. However when I undock it it always goes to black screen, (and often fans are up) and I also can't switch to terminal (ctrl+alt+F button) only hard restart fixes it.
Even that actually caused a problem today since I couldn't get after "decrypt successful" screen today after one of the hard restarts (full disk encryption here) 
Luckily switching in the bios from hybrid do discrete allowed me to boot and log-in.

thank you for your intstructions, finally something actionable to try.
I'm a little intimidated if I'll be able to manage but will try tomorrow. 

Do you think it will be safer to restart to the integrated card while playing with the drivers for the nvidia one though?

@leviport  why you guys have removed the option for people to "sudo apt-get install nvidia-driver-olderversion" and made it so it always installs the latest?
That would have made solving those issues so much easier.... :(

leviport, 2020-10-13:
> Thanks @leviport, that is a great point. I think one thing users (such as myself smile) of non-system76 hardware forget is that popos is really only heavily tested on system76 hardware and because it is primarily maintained for your hardware, using it on other hardware will likely work, but not guaranteed nor is it officially supported by system76 engineers. Would you say that is accurate?

Sounds pretty accurate, yeah. The bulk of testing is done on System76 hardware, but we do care quite a bit about usability across broad swaths of hardware. System76 hardware is already gives us pretty broad coverage, so that's usually sufficient to catch most bugs. Still, hardware is very diverse, and usually it's just specific models that cause trouble.

@NinjaTurtle007 We don't really remove anything, it's just that older versions become transitional to the latest version when a new version comes out. Nvidia drivers are tricky, and this is the best way to ensure that larger problems due to conflicting versions don't happen.

NinjaTurtle007, 2020-10-13:
Well pure ubuntu gives you the option to choose between couple of different drivers in their "additional drivers" tab that is removed in PopOS.
You could at least left the option for users to apt-get install at least one previous driver, for cases like this when the latest one is causing trouble.

NinjaTurtle007, 2020-10-14:
Hey what if I install nvidia-driver-435? Is it also transitional package to 455?

NinjaTurtle007, 2020-10-15:
> Hey what if I install nvidia-driver-435? Is it also transitional package to 455?

Please somebody let me know.

leviport, 2020-10-15:
I'm not sure, give it a try and see what happens. If it installs a version you don't want, just purge it before rebooting.

jacobgkau, 2020-10-15:
> > Hey what if I install nvidia-driver-435? Is it also transitional package to 455?
> 
> Please somebody let me know.

Aside from trying it yourself like Levi suggested, you can also see the answer to this question using the commands `apt policy nvidia-driver-435` and `apt depends nvidia-driver-435`. The candidate version for the 435 package is 440.100; 435 depends on 440, and 440 depends on 455, so yes, it is transitional to 455.

nvidia-driver-390 will install version 390, because 390 is a legacy driver version (for GPUs that NVIDIA has stopped supporting in their later drivers.) If your GPU is old enough to work with 390, then you could try that; according to [this list](https://www.nvidia.com/en-us/drivers/unix/legacy-gpu/), I do not think your GTX 1050 Ti will work with the legacy driver (assuming you have the same GPU as the person in #64.)

NinjaTurtle007, 2020-10-15:
@jacobgkau I was afraid of that!
So the only way is to install the driver from nvidia site (the .run file) ?
which is tricky to install since throws some errors during installation

kruegernet, 2020-10-18:
X1 Extreme Gen 2 with Lenovo TB3 dock is broken with 455, I can no longer use external monitor without weird subsections of the framebuffer corrupting all to hell (looks like bad pointer math or something of the sort).

Forcing users to a driver version is bad design. Not cool, guys.

NinjaTurtle007, 2020-10-18:
yes, i spent a week working on the laptop alone because of this. Reverting in ubuntu is as simple as clicking, while here you need to be advanced and there is risk you won't be able to boot if you mess up.
Really something popos should address.

z1g, 2020-10-31:
> yes, i spent a week working on the laptop alone because of this. Reverting in ubuntu is as simple as clicking, while here you need to be advanced and there is risk you won't be able to boot if you mess up.
> Really something popos should address.

FYI you can install the Gnome Software Center:

sudo apt install gnome-software

Then from it install Additional Drivers.  From Additional Drivers you can easily roll back to previous Nvidia drivers.

NinjaTurtle007, 2020-10-31:
> FYI you can install the Gnome Software Center:
>
> sudo apt install gnome-software
>
> Then from it install Additional Drivers. From Additional Drivers you can easily roll back to previous Nvidia drivers.

You sure that works on PopOS, have you tried?

z1g, 2020-10-31:
> You sure that works on PopOS, have you tried?

Yes.  On Pop!_OS 20.10 - Warcraft Classic would no longer work with the 455 driver and DXVK.  I searched for a week or more for a way to roll back to the 440 driver as that is what everyone said was the fix.  Finally found mention of the gnome software center and did that last nigh.  So far I am not having any issues, I am on the 440.95 driver and everything is working fine.

NinjaTurtle007, 2020-10-31:
Then you win the award for finding the easiest way to revert, you have no idea how much other stuff i tried.
Including downloading the driver from nvidia and trying to run it without GUI to no success.
I'll try this tomorrow.
in the meantime can i ask you for a screenshot from your nvidia settings window, to see the exact version you are running?

z1g, 2020-10-31:
> Then you win the award for finding the easiest way to revert, you have no idea how much other stuff i tried.
> Including downloading the driver from nvidia and trying to run it without GUI to no [...]

[image]

Redbatman89, 2020-11-04:
Did it work fro you guys????

apoger, 2020-11-04:
Yes it did. 440 running as a workaround.

Redbatman89, 2020-11-04:
> Yes it did. 440 running as a workaround.

Ah good, for some reason the driver they are pushing is part of the Linux Short Lived Branch and not Long Lived. Linux Short Lived is part of the temp/beta driver branch.

NinjaTurtle007, 2020-11-04:
> Yes it did. 440 running as a workaround.

Did you use the method explained here with gnome software? Still haven't tried it myself, looking for double confirmation it works that easy, because I've tried everything in more complicated ways and was unsuccesful.

apoger, 2020-11-04:
Yes, I used the gnome-software workaround.

pop-desktop will be uninstalled in the process. Just reinstall it after finishing the nvidia downgrade

Good luck.

Redbatman89, 2020-11-04:
> Yes, I used the gnome-software workaround.
> 
> pop-desktop will be uninstalled in the process. Just reinstall it after finishing the nvidia downgrade
> 
> Good luck.

Actually shouldn't POP OS include this in their software? I think it would be helpful. That way if Drivers start to give an issue its easy to roll back. Especially Since I don't want Ubuntu due to too much bloat.

NinjaTurtle007, 2020-11-05:
> Actually shouldn't POP OS include this in their software? I think it would be helpful. That way if Drivers start to give an issue its easy to roll back. Especially Since I don't want Ubuntu due to [...]

Yeah my thoughts exactly, they are making this unncessary hard when it's just 1 click solution in pure Ubuntu.
I offered that to the devs here, but they seem unfazed.

jdavidberger, 2021-01-08:
The workaround above didn't work for me. 

This has been my one persistent issue with pop-os; I've been fighting with nvidia drivers for the last year; it seems like anything past 435 causes random freezes on my machine and nvidia themselves hasn't been much of a help. The most recent driver doesn't finish booting. I'm in the middle of going through support channels for all of this but it'd be a big help if I could just revert back to 435 and not be broken while these problems are resolved.

mgcox2, 2021-01-27:
The worst part is then the new driver (460 in this case) doesn't build with your generic kernel for some reason (5.8.0-7630-generic) and you system is borked with nothing to go back to. Luckily I have Timeshift. But this takes time!

metadan, 2021-03-10:
X1Extreme gen2 here. Having loads of issue with 460 and finding it incredibly annoying I cant downgrade.

waltim, 2021-05-05:
I solved my problem following this sequence of commands.

https://www.reddit.com/r/pop_os/comments/n59ksz/dkms_errors_with_nvidiadkms460/gx13omu?utm_source=share&utm_medium=web2x&context=3

prittjam, 2021-07-01:
The 465.13 version disabled my USB-C external monitor. I can no longer use it. I am also trying to workaround and find it costs a lot of time. There should be an easier way to revert the Nvidia driver; they are known to introduce bugs in subsequent versions.

osbama, 2021-07-13:
I have this issue as well. I have a GTX1650, and after a routine system update. I was welcomed by a desktop at 640x480, with no straightforward way to roll back the forced update. This is costing me a lot of precious working time. NVIDIA is known for "blitz obsolescence" tactics. I strongly think Nvidia drivers must be treated separately from system updates, and left to users to decide which version to use.

leviport, 2021-07-13:
@osbama have you tried purging and reinstalling the driver yet? A 1650 is definitely supported by the current version, so I suspect there were troubles with how your system upgraded. 

To purge and reinstall the NVIDIA driver: 
```
sudo apt purge ~nnvidia
sudo apt clean
sudo apt install nvidia-driver-465
```

osbama, 2021-07-15:
@leviport, that was the first thing I tried before I started googling. Now, I rolled backed to a previous version of NVIDIA driver, switching to gnome-software. Nouveau also works.

Plausible deniability is the way of the modern Mobius, and I have seen this kind of behavior many times in my life from various manufactures. This kind of "driver bugs" will appear more frequently, as consumers are nudged towards "new" hardware. An update does not mean an upgrade, especially when the competition is this shallow (at least we had, Matrox, Via, etc. competing when I was younger). Hence I never update closed source drivers unless I am forced to, and I try to minimize software that forces me to update such drivers without giving me granular choice.  That is why I avoid Microsoft Windows for my work.

leviport, 2021-07-15:
If this issue has devolved into accusations of planned obsolescence, then I don't see any reason it needs to stay open any longer.

closed 2021-07-15 by leviport

cocadas03, 2021-11-11:
i have a question is: pop_os different another Linux distribution, if install driver nvidia. Linux distro use nouvea control version driver, and pop_os is ?????

waltim, 2021-11-11:
> i have a question is: pop_os different another Linux distribution, if install driver nvidia. Linux distro use nouvea control version driver, and pop_os is ?????

As far as I know (correct me if I'm talking nonsense) but Pop_OS usually uses NVIDIA's proprietary drivers.

If you have problems, you can switch Graphics drivers following this commands.
https://www.linuxbabe.com/desktop-linux/switch-intel-nvidia-graphics-card-ubuntu

 I think it probably requires a reboot. I hope I helped you with something.

cocadas03, 2021-11-12:
My question mea, I don't know pop OS use what control driver Nvidia,
different nouveau