pop-os/nvidia-graphics-drivers #69: Nvidia graphics mode causes Chromium Browsers to break after sleep and wake up

ArjunSingh99, 2020-10-28:
**Distribution (run `cat /etc/os-release`):**
 NAME="Pop!_OS"
VERSION="20.10"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 20.10"
VERSION_ID="20.10"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=groovy
UBUNTU_CODENAME=groovy
LOGO=distributor-logo-pop-os

 System Specifications:
OS: Pop!_OS 20.10 x86_64
Host: HP Pavilion 15 Notebook PC 0974120000405F100
Kernel: 5.8.0-7625-generic
CPU: Intel i5-5200U (4) @ 2.700GHz
GPU: NVIDIA GeForce 830M
GPU: Intel HD Graphics 5500
Memory: 4GB
Displays:
1366x768 (Laptop display)
1920x1080 (External display)
DE: GNOME
WM: Mutter

**Related Application and/or Package Version (run `apt policy $PACKAGE NAME`):**
brave-browser:
  Installed: (none)
  Candidate: 1.16.68
  Version table:
     1.16.68 500
        500 https://brave-browser-apt-release.s3.brave.com stable/main amd64 Packages
     1.15.76 500
        500 https://brave-browser-apt-release.s3.brave.com stable/main amd64 Packages
     1.15.75 500
        500 https://brave-browser-apt-release.s3.brave.com stable/main amd64 Packages
     1.15.72 500
        500 https://brave-browser-apt-release.s3.brave.com stable/main amd64 Packages
     1.14.84 500
        500 https://brave-browser-apt-release.s3.brave.com stable/main amd64 Packages
     1.14.81 500
        500 https://brave-browser-apt-release.s3.brave.com stable/main amd64 Packages
     1.13.86 500
        500 https://brave-browser-apt-release.s3.brave.com stable/main amd64 Packages
     1.12.112 -1
        100 /var/lib/dpkg/status

Note: This happens not only with Brave browser beta, but also with brave browser official, as well as Microsoft Edge Dev.

**Issue/Bug Description:**
When in Nvidia Graphics mode, and while using the aforementioned chromium based browser(s), with hardware acceleration turned on, it's noticed that after suspend and wake, graphical elements of the browser are broken, as seen in the screenshots. The browser becomes unusable, and the problem can only be solved by either restarting the browser, or turning hardware acceleration off.

This issue used to occur with firefox, however, it has been fixed, and even then, resizing the window or hovering the cursor over those broken graphics seemed to fix them.

**Steps to reproduce (if you know):**
Set graphics to Nvidia Graphics Mode.
Turn hardware acceleration on in the aforementioned chromium browser.
Suspend the device.
Wake the device.
Observe that many of the graphical elements are broken. Resizing/moving the window makes it worse, and it becomes unusable till the browser is restarted.

**Expected behavior:**
The browser should function normally without any issues, and should be usable without having to restart the browser without losing any potentially important information.

**Other Notes:**
This issue was prevalent even in Pop!_OS 20.04
This does not happen with the official Chromium browser available in the Pop shop.
This was a problem with firefox, that has since been fixed.

Brave browser:
[image]
[image]
[image]
Microsoft Edge:
[image]
[image]
[image]

jacobgkau, 2020-10-28:
Try running this command to add a kernel option for the NVIDIA driver:

```
sudo kernelstub -a nvidia.NVreg_PreserveVideoMemoryAllocations=1
```

Then reboot. Does that stop the visual corruption after suspend/resume?

---

System76 customers can reach out to [support](https://support.system76.com/) for technical assistance. For non-System76 hardware, you can seek community support on [Reddit](https://reddit.com/r/pop_os/) or [Mattermost](https://chat.pop-os.org).

ArjunSingh99, 2020-10-28:
> Try running this command to add a kernel option for the NVIDIA driver:
>
> ```
> sudo kernelstub -a nvidia.NVreg_PreserveVideoMemoryAllocations=1
> ```
>
> Then reboot. Does that stop the visual corruption [...]

Hello. Running that command didn't make a difference. Nothing changed, resizing the windows resulted in the same bug.

jonneymendoza, 2020-11-07:
Hi too have the exact same issue

ugemassolo, 2020-11-07:
Same here. It is less frecuent since I updated Chrome but it still happens sometimes.

manu-git-code, 2020-11-09:
Same for me on a desktop pc with an nvidia graphic card and popos. I had it on PopOs 20.04, and now also on PopOs 20.10. Everytime i resume from suspend, any Chrome window that i left open before suspending will have rendering issues. Shutting Chrome down and reopening it is enough to fix the issue, but it's quite annoying as I can't leave Chrome windows open and then suspend, as I know that once i resume from suspend I'll be forced to shut Chrome down to avoid rendering issues.

pnkfb9, 2020-11-14:
same here on Dell laptop with nVidia graphics card. Issue were here on 20.04 and still here on 20.10 version of PopOs.

andirsun, 2020-11-25:
I think is a nvidia driver in general, I am facing same bug on elementary os and nvidia mx250 in my laptop. Slack has de same effect after suspend.

GiantEnemyCrab, 2020-11-25:
I am not sure if others can reproduce this but I don't have this issue any more in nvidia driver 455.38, and it is GTX 1650, pop!_os 20.10. I used to get graphical glitch in Chrome after wake up from sleep, but I don't get this since upgrading to nvidia driver 455.38.

ArjunSingh99, 2020-11-25:
OP here, I'm not facing this issue on Nvidia 455.38. Can the other users who faced this issue confirm this?

leviport, 2020-11-25:
There was a recent update to Chromium that should have fixed the issue. If nobody else is seeing it happen on an up-to-date system, this should be safe to close. If I'm incorrect, just let me know and I'll reopen.

closed 2020-11-25 by leviport

jonneymendoza, 2020-11-25:
> There was a recent update to Chromium that should have fixed the issue. If nobody else is seeing it happen on an up-to-date system, this should be safe to close. If I'm incorrect, just let me know and I'll reopen.

I did update my google chrome manually by downloading it again from the google chrome website. and still had the same problem.

I will double check what gpu drivers i have. I have a GTX 1050

leviport, 2020-11-25:
This issue is about Chromium, but I was unaware that Chrome was having trouble. Does the Chrome version in the repos work any better? It can be installed with `sudo apt install google-chrome-stable`.

jonneymendoza, 2020-11-25:
> This issue is about Chromium, but I was unaware that Chrome was having trouble. Does the Chrome version in the repos work any better? It can be installed with `sudo apt install google-chrome-stable`.

O i thougth it was related to chrome because the original poster here told me to refer to this thread on the exact same issue i am having in chrome.

See here https://www.reddit.com/r/pop_os/comments/jnwaxw/pop_os_chrome_rendering_issues/

pnkfb9, 2020-11-25:
> This issue is about Chromium, but I was unaware that Chrome was having trouble. Does the Chrome version in the repos work any better? It can be installed with `sudo apt install google-chrome-stable`.

I can confirm the problem is found also in Chrome (I am using chrome on my Pop OS). Not tested with latest nVidia driver update, tho.

Selmi, 2020-11-25:
i see similar graphic problems after wakeup with spotify client and also in firefox after i enabled webrender compositing. with nvidia driver 455 and geforce 750

ArjunSingh99, 2020-11-26:
> This issue is about Chromium, but I was unaware that Chrome was having trouble. Does the Chrome version in the repos work any better? It can be installed with `sudo apt install google-chrome-stable`.

It isn't about Chromium. In fact, the Chromium browser itself didn't have a problem. I mentioned this in the other notes in my original post.
This issue pertains to Chromium based browsers. The browsers I was personally facing problems were on Brave Beta and MS Edge dev.
As of now, both seem to be fine. I haven't tried Google Chrome, though, so I don't know, but if others are having this problem, then maybe it still is an issue.

trusktr, 2021-04-01:
This is definitely a Chromium issue (and other Chromium-based or Chromium-forked apps inherit the defect).

In my Gnome desktop, Chromium glitches out after suspend, but all my other Gnome apps and Gnome Shell itself continue to operate buttery smooth without problem. Gnome apps never have this issue, while only my Chromium apps do, and only while using my Nvidia GPU (no issues with the integrated Intel GPU, except it is slower in general).

I think Chromium team may need to investigate what Gnome does with GPU handling after suspend, and adapt their code.

Here's a recent [Chromium bug report](https://bugs.chromium.org/p/chromium/issues/detail?id=1134981&q=linux%20glitches%20suspend&can=2). Please share any feedback there.

trusktr, 2021-04-01:
> In fact, the Chromium browser itself didn't have a problem. I mentioned this in the other notes in my original post.
> This issue pertains to Chromium based browsers

Maybe you got lucky? I noticed with changing versions of Chromium, the issue got better, or worse, so it could be that your Chromium was on a better version, while other browsers used a worse version of Chromium.