pop-os/nvidia-graphics-drivers #71: Oryp6 with 20.10: Can regularly induce a lockup when resuming from suspend

leviport, 2020-10-29:
While in Hybrid or Nvidia graphics modes, if I suspend the machine with the power adapter plugged in, then unplug it while it's sleeping, then wake it up, it pretty reliably freezes up completely, requiring a hard power-off by holding the power button. I can investigate further later and grab some logs.

leviport, 2020-10-30:
I got an Xid error on a successful resume, I'm wondering if it's related: 
```
Oct 30 09:12:21 pop-os kernel: NVRM: Xid (PCI:0000:01:00): 61, pid=2251, 0d20(3270) 00000000 00000000
```

leviport, 2020-10-30:
I'm starting to think this is a firmware bug. I did the same tests on gaze15 running proprietary firmware and EC and the lockups did not happen. Once I switched over to open firmware and EC, the lockups seem about as frequent as on oryp6.

jackpot51, 2021-01-22:
@leviport does 460.32.03 fix this?

leviport, 2021-01-22:
I just tried it, and 460.32 breaks suspend completely in Nvidia and Hybrid modes.

jackpot51, 2021-01-22:
Ok, so that isn't only a galp5 issue

leviport, 2021-01-25:
The fixed 460.32 version has fixed this issue.

closed 2021-01-25 by leviport