pop-os/nvidia-graphics-drivers #72: 455.28 Seems to somehow affect rendering on intel graphics mode.

Nashenas88, 2020-11-01:
uname -a: Linux pop-os 5.8.0-7625-generic #26~1603126178~20.10~210fe73-Ubuntu SMP Tue Oct 20 01:12:15 UTC  x86_64 x86_64 x86_64 GNU/Linux

I upgraded my linux drivers yesterday but did not restart. I've added the apt logs from the last few days, with a reboot each day.
<details>
<summary>Click to expand apt history</summary>
Start-Date: 2020-10-28  08:27:46
Commandline: /usr/bin/apt dist-upgrade
Requested-By: pfaria (1000)
Upgrade: distro-info-data:amd64 (0.44ubuntu1, 0.44ubuntu1.1), ca-certificates:amd64 (20200601, 20201027ubuntu0.20.10.1)
End-Date: 2020-10-28  08:27:50

Start-Date: 2020-10-29  20:16:04
Commandline: /usr/bin/apt dist-upgrade
Requested-By: pfaria (1000)
Upgrade: popsicle-gtk:amd64 (1.1.0~1600456165~20.10~9fa1f70, 1.2.0~1603918468~20.10~7c364df), popsicle:amd64 (1.1.0~1600456165~20.10~9fa1f70, 1.2.0~1603918468~20.10~7c364df), repoman:amd64 (2.0.0~1603294431~20.10~1555cb3, 2.0.0~1603825555~20.10~20bd07d), distro-info-data:amd64 (0.44ubuntu1.1, 0.44ubuntu1.3), pop-shell:amd64 (0.1.0~1603503416~20.10~05e531e, 1.0.0~1603994289~20.10~ecc9b38), tzdata:amd64 (2020b-1ubuntu1, 2020d-1ubuntu1), python3-repolib:amd64 (1.4.0~1603733612~20.10~f30cddb, 1.4.0~1603811627~20.10~edd989f)
End-Date: 2020-10-29  20:16:12

Start-Date: 2020-10-30  19:18:30
Commandline: /usr/bin/apt dist-upgrade
Requested-By: pfaria (1000)
Upgrade: libnvidia-compute-440:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-compute-440:i386 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-compute-455:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-compute-455:i386 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-encode-440:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-encode-440:i386 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-encode-455:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-encode-455:i386 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), nvidia-kernel-common-440:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), nvidia-kernel-common-455:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), xserver-xorg-video-nvidia-440:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-gl-440:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-gl-440:i386 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), xserver-xorg-video-nvidia-455:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-gl-455:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-gl-455:i386 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-fbc1-440:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-fbc1-440:i386 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-fbc1-455:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-fbc1-455:i386 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-decode-440:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-decode-440:i386 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-decode-455:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-decode-455:i386 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-cfg1-440:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-cfg1-455:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), nvidia-utils-440:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), nvidia-utils-455:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), nvidia-dkms-440:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), nvidia-dkms-455:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), nvidia-compute-utils-440:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), nvidia-compute-utils-455:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-ifr1-440:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-ifr1-440:i386 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-ifr1-455:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-ifr1-455:i386 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), nvidia-driver-440:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), nvidia-driver-455:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-common-440:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-common-455:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), libnvidia-extra-455:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), nvidia-kernel-source-440:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f), nvidia-kernel-source-455:amd64 (455.28-1pop0~1603223384~20.10~21eb2ec, 455.28-1pop0~1604073177~20.10~ef6315f)
End-Date: 2020-10-30  19:19:43
</details>

Only this morning I'm seeing random distortions when I have a lot of updates to the screen, e.g. calling `tree` from my home dir, or running an apt update. When I'm in Nvidia graphics mode the glitches disappear. I've been running on intel graphics for quite some time, and this is the first time I've seen some weird affect like this. 

The distortions look like small blocks of color being rendered to the wrong location, and it only seemed to affect the upper right triangle of the screen. I could never get a screen shot quick enough to see if it could be captured or not.

Nashenas88, 2020-11-01:
Now after using the nvidia driver for a bit, I'm see delayed rendering on webpages with the Nvidia Driver. The mouse will accurately change to reflect the contents (switch to text icon when over highlightable text, etc.), but the page itself will not update until some other content updates. An example of this for me happens on wanikani.com where I use the keyboard navigation to move between entries. Pressing Enter will update the content logically, but the page itself will not update the rendering until I move the mouse.

Nashenas88, 2020-11-01:
Accidentally closed while trying to upload a video of the above. The video won't seem to upload from my phone.

Nashenas88, 2020-11-07:
[image]

Here's a screen shot I was finally able to capture from a video.

crispy-cat, 2020-11-13:
Updating to version 455.38 appears to have fixed this for me,

closed 2024-12-04 by Nashenas88