pop-os/nvidia-graphics-drivers #75: So has anyone have trouble with new 455.38 Driver?

Redbatman89, 2020-11-13:
????

crispy-cat, 2020-11-13:
What troubles are you having? What hardware are you using?

dperkin6, 2020-11-16:
I just installed it today and now I get flickering on one of my three screens.

drr00t, 2020-11-17:
yes, i having problems do record videos and to use streaming platforms like streamyards, it is not detect my gpu. It is usually works thing with prevous nvidia driver.

Redbatman89, 2020-11-19:
So my issue is that Vulkan Shaders are still slow for processing for steam. 440.100 was still faster.

leviport, 2020-11-19:
> So my issue is that Vulkan Shaders are still slow for processing for steam. 440.100 was still faster.

In my experience, shader processing doesn't happen every time a game is launched. Only the first time it is launched, and sometimes after Nvidia, Steam, or the game itself is updated.

Since this issue doesn't seem to be a bug report or feature request, I am going to close it. Github is not really a great forum for general discussion about Pop, and I would instead direct people towards [Reddit](https://reddit.com/r/pop_os/),  [Mattermost](https://chat.pop-os.org), or [Pop Planet](https://pop-planet.info).

closed 2020-11-19 by leviport