pop-os/nvidia-graphics-drivers #77: Freezes of the Sublime Text

optozorax, 2020-12-01:
After upgrading to Pop!_OS 20.10 from the 20.04 version, my Sublime Text 3 starts freezing it's rendering. At some random point in time, Sublime just not render its contents for some seconds. AFAIK Sublime uses OpenGL to render, so I thought that updates of Nvidia drivers or update of system settings of rendering cause the problem.

madinya2, 2020-12-16:
I am having the same issue. Just upgraded by reinstallation from pop 20.04 to 20.10. Sublime 3 worked fine before the upgrade and now it is freezing for a few seconds at a time quite randomly. It happens rather frequently, perhaps once every few minutes of use.

bubth, 2020-12-20:
I also see the same problem in pop-os 20.10. It's quite annoying as it's happening very often. I don't have the same problem in Ubuntu 20.10.