pop-os/nvidia-graphics-drivers #82: Desktop craches /Relaunch when open OpenGL games

shagby, 2020-12-16:
When upgrading to pop os 20.10 and playing for example x-plane

1 Opening x-plane
2 Desktop relaunch and both my screens flashes just like when you log in to the system 

From x.org1.log

[    14.063] (--) NVIDIA(GPU-0): CRT-0: disconnected
[    14.063] (--) NVIDIA(GPU-0): CRT-0: 400.0 MHz maximum pixel clock
[    14.063] (--) NVIDIA(GPU-0): 
[    14.067] (--) NVIDIA(GPU-0): DFP-0: disconnected
[    14.067] (--) NVIDIA(GPU-0): DFP-0: Internal TMDS
[    14.067] (--) NVIDIA(GPU-0): DFP-0: 330.0 MHz maximum pixel clock
[    14.067] (--) NVIDIA(GPU-0): 
[    14.098] (--) NVIDIA(GPU-0): Acer H235H (DFP-1): connected
[    14.098] (--) NVIDIA(GPU-0): Acer H235H (DFP-1): Internal TMDS
[    14.098] (--) NVIDIA(GPU-0): Acer H235H (DFP-1): 600.0 MHz maximum pixel clock
[    14.098] (--) NVIDIA(GPU-0): 
[    14.098] (--) NVIDIA(GPU-0): DFP-2: disconnected
[    14.098] (--) NVIDIA(GPU-0): DFP-2: Internal DisplayPort
[    14.098] (--) NVIDIA(GPU-0): DFP-2: 960.0 MHz maximum pixel clock
[    14.098] (--) NVIDIA(GPU-0): 
[    14.098] (--) NVIDIA(GPU-0): DFP-3: disconnected
[    14.098] (--) NVIDIA(GPU-0): DFP-3: Internal TMDS
[    14.098] (--) NVIDIA(GPU-0): DFP-3: 165.0 MHz maximum pixel clock
[    14.098] (--) NVIDIA(GPU-0): 
[    14.098] (--) NVIDIA(GPU-0): Ancor Communications Inc VG248 (DFP-4): connected
[    14.098] (--) NVIDIA(GPU-0): Ancor Communications Inc VG248 (DFP-4): Internal DisplayPort
[    14.098] (--) NVIDIA(GPU-0): Ancor Communications Inc VG248 (DFP-4): 960.0 MHz maximum pixel clock
[    14.098] (--) NVIDIA(GPU-0): Ancor Communications Inc VG248 (DFP-4): supports NVIDIA 3D Vision stereo.
[    14.098] (--) NVIDIA(GPU-0): 
[    14.100] (--) NVIDIA(GPU-0): DFP-5: disconnected
[    14.100] (--) NVIDIA(GPU-0): DFP-5: Internal TMDS
[    14.100] (--) NVIDIA(GPU-0): DFP-5: 165.0 MHz maximum pixel clock
[    14.100] (--) NVIDIA(GPU-0): 
[    14.100] (--) NVIDIA(GPU-0): DFP-6: disconnected
[    14.100] (--) NVIDIA(GPU-0): DFP-6: Internal DisplayPort
[    14.100] (--) NVIDIA(GPU-0): DFP-6: 960.0 MHz maximum pixel clock
[    14.100] (--) NVIDIA(GPU-0): 
[    14.100] (--) NVIDIA(GPU-0): DFP-7: disconnected
[    14.100] (--) NVIDIA(GPU-0): DFP-7: Internal TMDS
[    14.100] (--) NVIDIA(GPU-0): DFP-7: 165.0 MHz maximum pixel clock
[    14.100] (--) NVIDIA(GPU-0): 
[    14.144] (II) NVIDIA(0): Setting mode "HDMI-0: nvidia-auto-select @1920x1080 +0+0 {ViewPortIn=1920x1080, ViewPortOut=1920x1080+0+0}, DP-2: 1920x1080_144 @1920x1080 +0+0 {ViewPortIn=1920x1080, ViewPortOut=1920x1080+0+0}"
[    14.303] (II) NVIDIA(0): Setting mode "HDMI-0: nvidia-auto-select @1920x1080 +1920+0 {ViewPortIn=1920x1080, ViewPortOut=1920x1080+0+0}, DP-2: 1920x1080_144 @1920x1080 +0+0 {ViewPortIn=1920x1080, ViewPortOut=1920x1080+0+0}"
[    15.507] (--) NVIDIA(GPU-0): CRT-0: disconnected
[    15.507] (--) NVIDIA(GPU-0): CRT-0: 400.0 MHz maximum pixel clock
[    15.507] (--) NVIDIA(GPU-0): 
[    15.511] (--) NVIDIA(GPU-0): DFP-0: disconnected
[    15.511] (--) NVIDIA(GPU-0): DFP-0: Internal TMDS
[    15.511] (--) NVIDIA(GPU-0): DFP-0: 330.0 MHz maximum pixel clock
[    15.511] (--) NVIDIA(GPU-0): 
[    15.541] (--) NVIDIA(GPU-0): Acer H235H (DFP-1): connected
[    15.541] (--) NVIDIA(GPU-0): Acer H235H (DFP-1): Internal TMDS
[    15.541] (--) NVIDIA(GPU-0): Acer H235H (DFP-1): 600.0 MHz maximum pixel clock
[    15.541] (--) NVIDIA(GPU-0): 
[    15.541] (--) NVIDIA(GPU-0): DFP-2: disconnected
[    15.541] (--) NVIDIA(GPU-0): DFP-2: Internal DisplayPort
[    15.541] (--) NVIDIA(GPU-0): DFP-2: 960.0 MHz maximum pixel clock
[    15.542] (--) NVIDIA(GPU-0): 
[    15.542] (--) NVIDIA(GPU-0): DFP-3: disconnected
[    15.542] (--) NVIDIA(GPU-0): DFP-3: Internal TMDS
[    15.542] (--) NVIDIA(GPU-0): DFP-3: 165.0 MHz maximum pixel clock
[    15.542] (--) NVIDIA(GPU-0): 
[    15.542] (--) NVIDIA(GPU-0): Ancor Communications Inc VG248 (DFP-4): connected
[    15.542] (--) NVIDIA(GPU-0): Ancor Communications Inc VG248 (DFP-4): Internal DisplayPort
[    15.542] (--) NVIDIA(GPU-0): Ancor Communications Inc VG248 (DFP-4): 960.0 MHz maximum pixel clock
[    15.542] (--) NVIDIA(GPU-0): Ancor Communications Inc VG248 (DFP-4): supports NVIDIA 3D Vision stereo.
[    15.542] (--) NVIDIA(GPU-0): 
[    15.543] (--) NVIDIA(GPU-0): DFP-5: disconnected
[... 166 lines cut ...]
[   281.417] (--) NVIDIA(GPU-0): DFP-2: disconnected
[   281.417] (--) NVIDIA(GPU-0): DFP-2: Internal DisplayPort
[   281.417] (--) NVIDIA(GPU-0): DFP-2: 960.0 MHz maximum pixel clock
[   281.417] (--) NVIDIA(GPU-0): 
[   281.418] (--) NVIDIA(GPU-0): DFP-3: disconnected
[   281.418] (--) NVIDIA(GPU-0): DFP-3: Internal TMDS
[   281.418] (--) NVIDIA(GPU-0): DFP-3: 165.0 MHz maximum pixel clock
[   281.418] (--) NVIDIA(GPU-0): 
[   281.418] (--) NVIDIA(GPU-0): Ancor Communications Inc VG248 (DFP-4): connected
[   281.418] (--) NVIDIA(GPU-0): Ancor Communications Inc VG248 (DFP-4): Internal DisplayPort
[   281.418] (--) NVIDIA(GPU-0): Ancor Communications Inc VG248 (DFP-4): 960.0 MHz maximum pixel clock
[   281.418] (--) NVIDIA(GPU-0): Ancor Communications Inc VG248 (DFP-4): supports NVIDIA 3D Vision stereo.
[   281.418] (--) NVIDIA(GPU-0): 
[   281.419] (--) NVIDIA(GPU-0): DFP-5: disconnected
[   281.419] (--) NVIDIA(GPU-0): DFP-5: Internal TMDS
[   281.419] (--) NVIDIA(GPU-0): DFP-5: 165.0 MHz maximum pixel clock
[   281.419] (--) NVIDIA(GPU-0): 
[   281.419] (--) NVIDIA(GPU-0): DFP-6: disconnected
[   281.419] (--) NVIDIA(GPU-0): DFP-6: Internal DisplayPort
[   281.419] (--) NVIDIA(GPU-0): DFP-6: 960.0 MHz maximum pixel clock
[   281.419] (--) NVIDIA(GPU-0): 
[   281.419] (--) NVIDIA(GPU-0): DFP-7: disconnected
[   281.419] (--) NVIDIA(GPU-0): DFP-7: Internal TMDS
[   281.419] (--) NVIDIA(GPU-0): DFP-7: 165.0 MHz maximum pixel clock
[   281.419] (--) NVIDIA(GPU-0):

shagby, 2020-12-16:
[dmes.log](https://github.com/pop-os/nvidia-graphics-drivers/files/5701192/dmes.log)
[gpu-manager.log](https://github.com/pop-os/nvidia-graphics-drivers/files/5701193/gpu-manager.log)
[xorg.0.log](https://github.com/pop-os/nvidia-graphics-drivers/files/5701194/xorg.0.log)
[xorg.1.log](https://github.com/pop-os/nvidia-graphics-drivers/files/5701195/xorg.1.log)

mmstick, 2020-12-16:
Needs gnome-shell logs from journalctl

shagby, 2020-12-16:
[systemctl log](https://github.com/pop-os/nvidia-graphics-drivers/files/5701840/systemctl.log)

shagby, 2020-12-16:
Here seems to be the problem

Dec 16 10:45:03 pop-os gnome-shell[2460]: Error in freeze/thaw accounting
Dec 16 10:45:12 pop-os systemd[1796]: org.gnome.Shell@x11.service: Main process exited, code=dumped, status=5/TRAP
Dec 16 10:45:12 pop-os gsd-power[2666]: Error setting property 'PowerSaveMode' on interface org.gnome.Mutter.DisplayConfig: GDBus.Error:org.freedesktop.DBus.Error.NoReply: Message recipient>
Dec 16 10:45:12 pop-os systemd[1796]: org.gnome.Shell@x11.service: Failed with result 'core-dump'.
Dec 16 10:45:12 pop-os systemd[1796]: org.gnome.Shell@x11.service: Scheduled restart job, restart counter is at 1.
Dec 16 10:45:12 pop-os systemd[1796]: Stopped GNOME Shell on X11.
Dec 16 10:45:12 pop-os systemd[1796]: Starting GNOME Shell on X11...

shagby, 2020-12-16:
This only happens when upgrading to 20.10, in 20.04 everything is working perfectly!
I guess it has something to do with Gnome 3.38 ?!

shagby, 2020-12-16:
When staring remmina, everything freezed and the desktop rebooted just like the game

[remmina.log](https://github.com/pop-os/nvidia-graphics-drivers/files/5701892/remmina.log)

TechnologyMan101, 2020-12-16:
I think it has to do with GNOME 3.38 and not with the Pop!_OS NVIDIA drivers. This same issue is on Ubuntu. I have not encountered the issue on rolling release distros like Manjaro, probably because those are already on GNOME 3.38.2 while Pop!_OS and Ubuntu are still on 3.38.1. And I'm also using Intel Integrated Graphics, and I don't have any NVIDIA graphics cards installed. GNOME 3.38.2 has seemed to fix many problems when I was using Manjaro on my secondary drive.

shagby, 2020-12-16:
But i have tried with Manjaro latest vers with gnome and it's the same issue there also.

mmstick, 2020-12-16:
3.38.2 version of GNOME Shell is currently stuck not building in proposed. It will be released soon the moment it's built and working

mmstick, 2020-12-16:
It's on its way out now

shagby, 2020-12-16:
Is it released to pop OS 20.10? Now ? 

Br Staffan 

Skickat från min iPhone

shagby, 2021-01-05:
Removed battery from motherboard and reinstalled my graphic card, the issues is now resolved!

closed 2021-01-05 by shagby