pop-os/nvidia-graphics-drivers #84: Nvidia graphics and fractional scaling in PopOS 20.04 LTS

shrvtv, 2021-01-01:
I simply can't use fractional scaling while using Nvidia Graphics battery mode. Setting 125%, 150%, 175% all leads to laggy 200% scaling. What do I need to do in order to fix this? Is it a driver problem? Everything works well on Hybrid Graphics battery mode, but I need the Nvidia one

Nvidia driver version on my HP Omen 15: 455.38

This bug has been posted on launchpad: https://bugs.launchpad.net/ubuntu/+source/gnome-control-center/+bug/1870736?comments=all

shrvtv, 2021-01-03:
Also, I am unable to start application with dedicated graphics card, while using fractional scaling. What should I do in order to fix this?

skyjebus, 2021-11-16:
Can confirm still an issue on Pop_OS 21.04 with my 3070.

kingfisher64, 2022-11-07:
It's an issue for me too with popOS 22.04 nVidia 3070. Installing the latest drivers 515.65.01 didn't work for me.