pop-os/nvidia-graphics-drivers #88: galp5 NVIDIA GPU crashes in Compute mode

jacobgkau, 2021-01-28:
On galp5 w/ 1650 or 1650 Ti, I get an NVIDIA GPU crash when I boot in Compute mode (the boot is successful, but the crash happens by the time I'm logged in and able to run dmesg): [nvidia-hang.txt](https://github.com/pop-os/nvidia-graphics-drivers/files/5890329/nvidia-hang.txt)

After this happens, I can't utilize the NVIDIA GPU (`nvidia-smi` just freezes), and the system doesn't shut down cleanly. This is happening with 460.32 and 460.39, 20.04 and 20.10, and stable firmware (`2020-12-15_79d6fa7`) and the `firmware-open` branch (`2021-01-19_d6d37c0`).

curiousercreative, 2022-06-08:
Is this still an issue?

leviport, 2022-06-09:
Seems to be working fine on a 1650Ti galp5

closed 2022-06-09 by leviport