pop-os/nvidia-graphics-drivers #89: Driver/library version mismatch

githubmo, 2021-02-03:
When trying to launch `nvidia-smi`, I get `Failed to initialize NVML: Driver/library version mismatch`

After removing the nvidia driver and reinstalling it with `sudo apt install system76-driver-nvidia`, I notice the following output:

```
DKMS: install completed.
Setting up system76-firmware-daemon (1.0.21~1609258571~20.10~6ff20f9) ...
Setting up libnvidia-encode-460:amd64 (460.39-1pop0~1611882295~20.10~5279175) ...
Setting up libnvidia-encode-460:i386 (460.39-1pop0~1611882295~20.10~5279175) ...
Setting up system76-driver (20.04.26~1611604726~20.10~4b9f4d0) ...
WARNING	invalid sys_vendor: 'Gigabyte Technology Co., Ltd.'
WARNING	invalid product_version: None
Setting up nvidia-driver-460 (460.39-1pop0~1611882295~20.10~5279175) ...
Processing triggers for dbus (1.12.20-1ubuntu1) ...
Processing triggers for desktop-file-utils (0.24-1ubuntu4) ...
Processing triggers for mime-support (3.64ubuntu1) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for gnome-menus (3.36.0-1ubuntu1) ...
Processing triggers for libglib2.0-0:amd64 (2.66.1-2) ...
Processing triggers for libglib2.0-0:i386 (2.66.1-2) ...
Processing triggers for libc-bin (2.32-0ubuntu3) ...
Processing triggers for man-db (2.9.3-2) ...
Setting up system76-driver-nvidia (20.04.26~1611604726~20.10~4b9f4d0) ...
INFO:Enable nvidia-375
Processing triggers for initramfs-tools (0.137ubuntu12) ...
update-initramfs: Generating /boot/initrd.img-5.10.12-xanmod1
cryptsetup: WARNING: Resume target cryptswap uses a key file
kernelstub.Config    : INFO     Looking for configuration...
kernelstub           : INFO     System information: 

    OS:..................Pop!_OS 20.04
    Root partition:....../dev/nvme0n1p8
    Root FS UUID:........e902e829-a4b8-4cc1-8a4d-ef7534481946
    ESP Path:............/boot/efi
    ESP Partition:......./dev/nvme0n1p5
    ESP Partition #:.....5
    NVRAM entry #:.......-1
    Boot Variable #:.....0000
    Kernel Boot Options:.quiet loglevel=0 systemd.show_status=false splash
    Kernel Image Path:.../boot/vmlinuz-5.10.12-xanmod1
    Initrd Image Path:.../boot/initrd.img-5.10.12-xanmod1
    Force-overwrite:.....False

kernelstub.Installer : INFO     Copying Kernel into ESP
kernelstub.Installer : INFO     Copying initrd.img into ESP
kernelstub.Installer : INFO     Setting up loader.conf configuration
kernelstub.Installer : INFO     Making entry file for Pop!_OS
kernelstub.Installer : INFO     Backing up old kernel
kernelstub.Installer : INFO     Making entry file for Pop!_OS
```

Notice the multiple references to `nvidia-375`.

Is there a way in which I can manually resolve this?

Redbatman89, 2021-02-03:
Is this recent? Can you use timeshift to rollback before this happened?

bflanagin, 2021-02-03:
So good news, bad news. Good news, it looks like the references to nvidia-375 is a bug (rather an oversight) and doesn't seem to to affect the machines I have, though I see the same message. Bad news, is that you still have an issue. 

What I'm concerned about is the lines:
```
WARNING	invalid sys_vendor: 'Gigabyte Technology Co., Ltd.'
WARNING	invalid product_version: None
```
Could you give us some info on your machine? What kind of nvidia card does it have?

jacobgkau, 2021-02-03:
`Failed to initialize NVML: Driver/library version mismatch` would typically show up if you run `nvidia-smi` after updating the driver, but before rebooting. Just to confirm, did you reboot after you updated the driver?

githubmo, 2021-02-03:
So I have the Gigabyte X570 I Pro Wifi (mini-itx) motherboard with a Ryzen 5950x and an EVGA XC3 Nvidia Geforce 3090.

I'm also running [xanmod](https://xanmod.org) kernel 5.10.12

I've rebooted multiple times but to know good.

Here are my installed packages:

```
❯ apt list --installed | grep nvidia

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libnvidia-cfg1-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 amd64 [installed,automatic]
libnvidia-common-455/groovy,groovy,now 460.39-1pop0~1611882295~20.10~5279175 all [installed]
libnvidia-common-460/groovy,groovy,now 460.39-1pop0~1611882295~20.10~5279175 all [installed,automatic]
libnvidia-compute-455/groovy,now 460.39-1pop0~1611882295~20.10~5279175 amd64 [installed]
libnvidia-compute-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 amd64 [installed]
libnvidia-compute-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 i386 [installed,automatic]
libnvidia-container-tools/groovy,now 1.3.0-1pop1~1601490873~20.10~cb62a8d amd64 [installed,automatic]
libnvidia-container1/groovy,now 1.3.0-1pop1~1601490873~20.10~cb62a8d amd64 [installed,automatic]
libnvidia-decode-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 amd64 [installed,automatic]
libnvidia-decode-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 i386 [installed,automatic]
libnvidia-encode-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 amd64 [installed,automatic]
libnvidia-encode-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 i386 [installed,automatic]
libnvidia-extra-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 amd64 [installed,automatic]
libnvidia-fbc1-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 amd64 [installed,automatic]
libnvidia-fbc1-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 i386 [installed,automatic]
libnvidia-gl-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 amd64 [installed,automatic]
libnvidia-gl-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 i386 [installed,automatic]
libnvidia-ifr1-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 amd64 [installed,automatic]
libnvidia-ifr1-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 i386 [installed,automatic]
nvidia-compute-utils-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 amd64 [installed,automatic]
nvidia-container-runtime/groovy,now 3.4.0-1pop1~1601325114~20.10~2880fc6 amd64 [installed]
nvidia-container-toolkit/groovy,now 1.3.0-1pop1~1601490793~20.10~270505a amd64 [installed]
nvidia-dkms-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 amd64 [installed,automatic]
nvidia-driver-455/groovy,now 460.39-1pop0~1611882295~20.10~5279175 amd64 [installed]
nvidia-driver-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 amd64 [installed,automatic]
nvidia-kernel-common-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 amd64 [installed,automatic]
nvidia-kernel-source-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 amd64 [installed,automatic]
nvidia-settings/groovy,now 440.82-0ubuntu1 amd64 [installed,automatic]
nvidia-utils-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 amd64 [installed,automatic]
xserver-xorg-video-nvidia-460/groovy,now 460.39-1pop0~1611882295~20.10~5279175 amd64 [installed,automatic]

❯ apt list --installed | grep cuda

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

system76-cuda-11.1/focal,now 0pop1 amd64 [installed]
system76-cuda-11.2/focal,groovy,now 0pop1 amd64 [installed]
system76-cuda/focal,groovy,now 0pop1 amd64 [installed]
```

Below is my neofetch. 

```
❯ neofetch
             /////////////                mo@pop-os 
         /////////////////////            --------- 
      ///////*767////////////////         OS: Pop!_OS 20.04 LTS x86_64 
    //////7676767676*//////////////       Host: X570 I AORUS PRO WIFI -CF 
   /////76767//7676767//////////////      Kernel: 5.10.12-xanmod1 
  /////767676///*76767///////////////     Uptime: 36 secs 
 ///////767676///76767.///7676*///////    Packages: 2931 (dpkg), 30 (flatpak), 13 (snap) 
/////////767676//76767///767676////////   Shell: zsh 5.8 
//////////76767676767////76767/////////   Resolution: 3840x2160 
///////////76767676//////7676//////////   DE: GNOME 
////////////,7676,///////767///////////   WM: Mutter 
/////////////*7676///////76////////////   WM Theme: Pop 
///////////////7676////////////////////   Theme: Flat-Remix-GTK-Blue-Dark-Solid [GTK2/3] 
 ///////////////7676///767////////////    Icons: Flat-Remix-Blue-Dark [GTK2/3] 
  //////////////////////'////////////     Terminal: tilix 
   //////.7676767676767676767,//////      CPU: AMD Ryzen 9 5950X (32) @ 3.400GHz 
    /////767676767676767676767/////       GPU: NVIDIA 0a:00.0 NVIDIA Corporation Device 2204 
      ///////////////////////////         Memory: 2428MiB / 64323MiB 
         /////////////////////
             /////////////                                        
                                                                  
```

mmstick, 2021-02-03:
As mentioned, your kernel isn't using the same driver as what's been installed. Since you're using a custom kernel, you'll need to ensure you've updated your initramfs.

githubmo, 2021-02-04:
I've tried quite a few way to update the intramfs but it did not work. Xanmod kernel 5.10.11 and prior seemed to have worked.

In the end, I have `sudo apt purge '*xanmod*'` and reverted to the stock kernel. It seems to be responding better with Zen 3 and works with the latest Nvidia drivers. I'll keep it that way and I'll close this ticket. Thanks for the help and its' a shame there was no simple and easy silver bullet to solve this issue with.

closed 2021-02-04 by githubmo