pop-os/nvidia-graphics-drivers #90: Please pull driver 460.39 its causing issues.

Redbatman89, 2021-02-03:
I had left this on reddit this morning but I'm going to post it here too. 

Yesterday when I had updated to Nvidia Driver 460.39 I began to notice some issues. One is that my suspend function got broken. When I was about to go to bed last night and put the computer into suspend it had locked up. The system was on but wouldn't go to suspend.

This morning when I had turned the computer on again after the hard shutdown from last night I had it on for 40 minutes till I had to go on a zoom meeting. When I had once again tried to put it into sleep mode it had once again locked up. The monitor went off but the PC was still on, so I had to hard shut down again. Immediately I had used time shift to roll back to driver 460.32.03.

Now Suspend/sleep mode is working again. Please pull this driver its causing issues.

bflanagin, 2021-02-03:
@Redbatman89 mind giving us some info on your computer? I'm running 460.39 and not seeing the issue.

Redbatman89, 2021-02-03:
> @Redbatman89 mind giving us some info on your computer? I'm running 460.39 and not seeing the issue.

If its my PC specs you want its a custom built one

MSI PC Mate Z97
NVIDIA ASUS GTX 1080
Intel I74790K 
32GB of RAM
2 SSDs Crucial Brand
1 2TB HDD WD BLACK

ArjunSingh99, 2021-02-06:
I'm having this exact same issue too. Lockscreen freezes and suspend issues. Happens only in Nvidia graphics mode.

My system specs:
CPU: i5 5200u
GPU: Nvidia 830M
RAM: 4GB
Pop os 20.04
Kernel: 5.8.0-7642-generic
Gnome 3.36

Redbatman89, 2021-02-08:
> I'm having this exact same issue too. Lockscreen freezes and suspend issues. Happens only in Nvidia graphics mode.
> 
> My system specs:
> CPU: i5 5200u
> GPU: Nvidia 830M
> RAM: 4GB
> Pop os 20.04
> Kernel: 5.8.0-7642-generic
> Gnome 3.36

Are you on a laptop? Mine is a desktop that is custom built.

ArjunSingh99, 2021-02-09:
> > I'm having this exact same issue too. Lockscreen freezes and suspend issues. Happens only in Nvidia graphics mode.
> > My system specs:
> > CPU: i5 5200u
> > GPU: Nvidia 830M
> > RAM: 4GB
> > Pop os 20.04
> > Kernel: 5.8.0-7642-generic
> > Gnome 3.36
> 
> Are you on a laptop? Mine is a desktop that is custom built.

Yes mine is a laptop.

sm-Fifteen, 2021-02-15:
Same here, though I also noticed this interesting tidbit in the kernel logs in journald once the system fails to resume:

```
PM: suspend exit
ACPI Error: Aborting method \_SB.PCI0.SAT0.SPT4._GTF due to previous error (AE_NOT_FOUND) (20200528/psparse-529)
ata5.00: configured for UDMA/133
RTW: wlxecade0742b85 cmd=57,2,0 process_time=1716
------------[ cut here ]------------
WARNING: CPU: 1 PID: 6912 at /var/lib/dkms/nvidia/460.39/build/nvidia/nv.c:3826 nv_restore_user_channels+0xce/0xe0 [nvidia]
Modules linked in: rfcomm xt_conntrack xt_MASQUERADE nf_conntrack_netlink xfrm_user xfrm_algo nft_counter xt_addrtype nft_compat nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables nfnetlink br_netfilter bridge stp llc snd_seq_dummy snd_hrtimer 8821cu(OE) cfg80211 aufs cmac algif_hash algif_skcipher af_alg overlay bnep nvidia_uvm(OE) nls_iso8859_1 nvidia_drm(POE) intel_rapl_msr nvidia_modeset(POE) mei_hdcp intel_rapl_common snd_hda_codec_realtek snd_hda_codec_generic x86_pkg_temp_thermal ledtrig_audio intel_powerclamp snd_hda_codec_hdmi snd_usb_audio kvm_intel snd_hda_intel snd_usbmidi_lib snd_intel_dspcfg nvidia(POE) snd_hda_codec kvm rapl snd_hda_core btusb intel_cstate btrtl snd_hwdep snd_seq_midi btbcm snd_seq_midi_event mc btintel snd_rawmidi efi_pstore bluetooth at24 snd_pcm snd_seq ecdh_generic joydev ecc input_leds hid_sony ff_memless snd_seq_device drm_kms_helper snd_timer mei_me snd cec soundcore mei rc_core fb_sys_fops mac_hid syscopyarea
 sysfillrect sysimgblt sch_fq_codel f71882fg coretemp parport_pc ppdev lp parport drm ip_tables x_tables autofs4 btrfs blake2b_generic dm_crypt raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear system76_io(OE) system76_acpi(OE) hid_logitech_hidpp hid_steam hid_logitech_dj uas usb_storage hid_generic usbhid hid gpio_ich crct10dif_pclmul crc32_pclmul ghash_clmulni_intel aesni_intel crypto_simd cryptd glue_helper ahci r8169 i2c_i801 lpc_ich i2c_smbus libahci xhci_pci realtek xhci_pci_renesas wmi video
CPU: 1 PID: 6912 Comm: nvidia-sleep.sh Tainted: P        W  OE     5.8.0-7642-generic #47~1612288990~20.10~b8113e7-Ubuntu
Hardware name: MSI MS-7758/Z77A-G43 (MS-7758), BIOS V2.13 03/07/2014
RIP: 0010:nv_restore_user_channels+0xce/0xe0 [nvidia]
Code: 24 d1 e6 be 01 00 00 00 4c 89 ef e8 dc a1 00 00 48 89 df e8 24 24 d1 e6 ba 02 00 00 00 4c 89 ee 4c 89 e7 e8 14 15 8c 00 eb 93 <0f> 0b eb c6 41 be 51 00 00 00 eb 9e 66 0f 1f 44 00 00 0f 1f 44 00
RSP: 0018:ffffb3b8c2c8bde8 EFLAGS: 00010206
RAX: 0000000000000003 RBX: ffff8a6029bfa800 RCX: ffffb3b8c2c8bd88
RDX: 0000000000000087 RSI: 0000000000000246 RDI: 0000000000000246
RBP: ffffb3b8c2c8be10 R08: 0000000000000000 R09: ffffffffc0bfd300
R10: ffff8a5fe33dac60 R11: 0000000000000001 R12: ffff8a6007d03000
R13: ffff8a6029bfa800 R14: 0000000000000003 R15: ffff8a6029bfacf8
FS:  00007fb78ddb2740(0000) GS:ffff8a603ec40000(0000) knlGS:0000000000000000
CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
CR2: 00000b7887be8408 CR3: 00000007f45aa004 CR4: 00000000001606e0
Call Trace:
 nv_set_system_power_state+0x228/0x3d0 [nvidia]
 nv_procfs_write_suspend+0xea/0x140 [nvidia]
 proc_reg_write+0x5a/0x90
 ? __sb_start_write+0x27/0x80
 vfs_write+0xc9/0x200
 ksys_write+0x67/0xe0
 __x64_sys_write+0x1a/0x20
 do_syscall_64+0x49/0xc0
 entry_SYSCALL_64_after_hwframe+0x44/0xa9
RIP: 0033:0x7fb78debdd57
Code: 0d 00 f7 d8 64 89 02 48 c7 c0 ff ff ff ff eb b7 0f 1f 00 f3 0f 1e fa 64 8b 04 25 18 00 00 00 85 c0 75 10 b8 01 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 51 c3 48 83 ec 28 48 89 54 24 18 48 89 74 24
RSP: 002b:00007ffd30758148 EFLAGS: 00000246 ORIG_RAX: 0000000000000001
RAX: ffffffffffffffda RBX: 0000000000000007 RCX: 00007fb78debdd57
RDX: 0000000000000007 RSI: 0000564b2ee804d0 RDI: 0000000000000001
RBP: 0000564b2ee804d0 R08: 000000000000000a R09: 0000564b2ee804d0
R10: 0000000000000077 R11: 0000000000000246 R12: 0000000000000007
R13: 00007fb78df996c0 R14: 00007fb78df9a4c0 R15: 00007fb78df998c0
---[ end trace e32939b49f3c06c0 ]---
------------[ cut here ]------------
WARNING: CPU: 1 PID: 6912 at /var/lib/dkms/nvidia/460.39/build/nvidia/nv.c:4021 nv_set_system_power_state+0x2c9/0x3d0 [nvidia]
Modules linked in: rfcomm xt_conntrack xt_MASQUERADE nf_conntrack_netlink xfrm_user xfrm_algo nft_counter xt_addrtype nft_compat nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables nfnetlink br_netfilter bridge stp llc snd_seq_dummy snd_hrtimer 8821cu(OE) cfg80211 aufs cmac algif_hash algif_skcipher af_alg overlay bnep nvidia_uvm(OE) nls_iso8859_1 nvidia_drm(POE) intel_rapl_msr nvidia_modeset(POE) mei_hdcp intel_rapl_common snd_hda_codec_realtek snd_hda_codec_generic x86_pkg_temp_thermal ledtrig_audio intel_powerclamp snd_hda_codec_hdmi snd_usb_audio kvm_intel snd_hda_intel snd_usbmidi_lib snd_intel_dspcfg nvidia(POE) snd_hda_codec kvm rapl snd_hda_core btusb intel_cstate btrtl snd_hwdep snd_seq_midi btbcm snd_seq_midi_event mc btintel snd_rawmidi efi_pstore bluetooth at24 snd_pcm snd_seq ecdh_generic joydev ecc input_leds hid_sony ff_memless snd_seq_device drm_kms_helper snd_timer mei_me snd cec soundcore mei rc_core fb_sys_fops mac_hid syscopyarea
 sysfillrect sysimgblt sch_fq_codel f71882fg coretemp parport_pc ppdev lp parport drm ip_tables x_tables autofs4 btrfs blake2b_generic dm_crypt raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear system76_io(OE) system76_acpi(OE) hid_lo [... cut ...]
This doesn't seem to happen every time, but looks like an interesting lead. I'm also noticing that the logs from before the issue started cropping up didn't mention `nvidia-suspend.service`, whereas now they do.

There's also [a related thread on the nvidia forum](https://forums.developer.nvidia.com/t/computer-not-suspending-when-using-460-27-04-beta-drivers-on-manjaro/165021) about a similar issue being related to the `PreserveVideoMemoryAllocations` [power management feature](https://download.nvidia.com/XFree86/Linux-x86_64/435.17/README/powermanagement.html), which is connected to nvidia's systemd power management systemd units.

sm-Fifteen, 2021-02-16:
So commenting out `options nvidia NVreg_PreserveVideoMemoryAllocations=1` in `/usr/lib/modprobe.d/nvidia-graphics-drivers.conf` before rebooting fixed the problem on my end. Oddly enough, creating a new file with `options nvidia NVreg_PreserveVideoMemoryAllocations=0` in `/etc/modeprobe.d` didn't override the value properly, you really have to comment out the option in /usr/lib.

sm-Fifteen, 2021-03-27:
> So commenting out `options nvidia NVreg_PreserveVideoMemoryAllocations=1` in `/usr/lib/modprobe.d/nvidia-graphics-drivers.conf` before rebooting fixed the problem on my end. Oddly enough, creating a [...]

Updates to the nvidia driver also seem to overwrite this, so you have to do this again after updating.

AlekzNet, 2021-05-15:
Same issue with 460.80 , GTX 960 and Ubuntu 21.04:

'''
May 15 21:26:02 zen kernel: [ 2055.091107] igb 0000:06:00.0 enp6s0: igb: enp6s0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
May 15 21:26:15 zen kernel: [ 2067.186249] ------------[ cut here ]------------
May 15 21:26:15 zen kernel: [ 2067.186251] WARNING: CPU: 5 PID: 11493 at /var/lib/dkms/nvidia/460.80/build/nvidia/nv.c:3826 nv_restore_user_channels+0xce/0xe0 [nvi
dia]
May 15 21:26:15 zen kernel: [ 2067.186424] Modules linked in: nfsv3 nfs fscache xt_CHECKSUM xt_MASQUERADE xt_conntrack ipt_REJECT nf_reject_ipv4 xt_tcpudp nft_comp
at nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_counter nf_tables nfnetlink bridge stp llc cmac algif_hash algif_skcipher af_alg bnep binfmt
_misc nls_iso8859_1 dm_multipath scsi_dh_rdac scsi_dh_emc scsi_dh_alua iwlmvm mac80211 libarc4 btusb btrtl iwlwifi btbcm btintel r8153_ecm cdc_ether bluetooth usbn
et r8152 ecdh_generic input_leds joydev mii ecc cfg80211 intel_rapl_msr intel_rapl_common snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio edac_mce_amd sn
d_hda_codec_hdmi snd_hda_intel kvm_amd snd_intel_dspcfg soundwire_intel soundwire_generic_allocation kvm soundwire_cadence snd_hda_codec snd_usb_audio snd_hda_core
 snd_usbmidi_lib rapl soundwire_bus mc efi_pstore wmi_bmof mxm_wmi snd_soc_core k10temp snd_compress ccp snd_pcm_dmaengine mac_hid nvidia_uvm(POE) sch_fq_codel snd
_emu10k1_synth snd_emux_synth snd_emu10k1 snd_util_mem
May 15 21:26:15 zen kernel: [ 2067.186453]  snd_ac97_codec snd_seq_midi_emul snd_hwdep snd_seq_virmidi ac97_bus snd_pcm snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq snd_seq_device snd_timer snd soundcore it87(OE) hwmon_vid msr parport_pc ppdev nfsd lp parport auth_rpcgss nfs_acl lockd grace sunrpc nfs_ssc ip_tables x_tables autofs4 btrfs blake2b_generic raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid0 multipath linear hid_plantronics hid_generic usbhid hid uas usb_storage nvidia_drm(POE) nvidia_modeset(POE) raid1 nvidia(POE) drm_kms_helper syscopyarea crct10dif_pclmul sysfillrect crc32_pclmul sysimgblt fb_sys_fops ghash_clmulni_intel cec aesni_intel ahci rc_core crypto_simd igb nvme cryptd r8169 xhci_pci glue_helper dca drm i2c_piix4 libahci nvme_core xhci_pci_renesas i2c_algo_bit realtek wmi
May 15 21:26:15 zen kernel: [ 2067.186483] CPU: 5 PID: 11493 Comm: nvidia-sleep.sh Tainted: P           OE     5.11.0-18-generic #19-Ubuntu
May 15 21:26:15 zen kernel: [ 2067.186485] Hardware name: Gigabyte Technology Co., Ltd. X570 AORUS MASTER/X570 AORUS MASTER, BIOS F30 09/07/2020
May 15 21:26:15 zen kernel: [ 2067.186486] RIP: 0010:nv_restore_user_channels+0xce/0xe0 [nvidia]
May 15 21:26:15 zen kernel: [ 2067.186649] Code: f6 0d da be 01 00 00 00 4c 89 ef e8 4c a2 00 00 48 89 df e8 64 f5 0d da ba 02 00 00 00 4c 89 ee 4c 89 e7 e8 44 73 8c 00 eb 93 <0f> 0b eb c6 41 be 51 00 00 00 eb 9e 66 0f 1f 44 00 00 0f 1f 44 00
May 15 21:26:15 zen kernel: [ 2067.186650] RSP: 0018:ffffbe16c4e3bde8 EFLAGS: 00010206
May 15 21:26:15 zen kernel: [ 2067.186652] RAX: 0000000000000003 RBX: ffff9433c0f90800 RCX: ffffbe16c4e3bd88
May 15 21:26:15 zen kernel: [ 2067.186652] RDX: 0000000000000087 RSI: 0000000000000246 RDI: 0000000000000246
May 15 21:26:15 zen kernel: [ 2067.186653] RBP: ffffbe16c4e3be10 R08: 0000000000000000 R09: 0000000000000000
May 15 21:26:15 zen kernel: [ 2067.186654] R10: 0000000000000001 R11: 0000000000000300 R12: ffff9434a3833000
May 15 21:26:15 zen kernel: [ 2067.186655] R13: ffff9433c0f90800 R14: 0000000000000003 R15: ffff9433c0f90cf8
May 15 21:26:15 zen kernel: [ 2067.186655] FS:  00007f7c2d2a3740(0000) GS:ffff9442beb40000(0000) knlGS:0000000000000000
May 15 21:26:15 zen kernel: [ 2067.186657] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
May 15 21:26:15 zen kernel: [ 2067.186657] CR2: 0000559557b2a7f0 CR3: 0000000110ff8000 CR4: 0000000000350ee0
May 15 21:26:15 zen kernel: [ 2067.186658] Call Trace:
May 15 21:26:15 zen kernel: [ 2067.186660]  nv_set_system_power_state+0x228/0x3d0 [nvidia]
May 15 21:26:15 zen kernel: [ 2067.186823]  nv_procfs_write_suspend+0xea/0x140 [nvidia]
May 15 21:26:15 zen kernel: [ 2067.186987]  proc_reg_write+0x5a/0x90
May 15 21:26:15 zen kernel: [ 2067.186989]  ? _cond_resched+0x1a/0x50
May 15 21:26:15 zen kernel: [ 2067.186991]  vfs_write+0xc6/0x270
May 15 21:26:15 zen kernel: [ 2067.186994]  ksys_write+0x67/0xe0
May 15 21:26:15 zen kernel: [ 2067.186995]  __x64_sys_write+0x1a/0x20
May 15 21:26:15 zen kernel: [ 2067.186996]  do_syscall_64+0x38/0x90
May 15 21:26:15 zen kernel: [ 2067.186997]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
May 15 21:26:15 zen kernel: [ 2067.186999] RIP: 0033:0x7f7c2d3adc27
May 15 21:26:15 zen kernel: [ 2067.187000] Code: 0d 00 f7 d8 64 89 02 48 c7 c0 ff ff ff ff eb b7 0f 1f 00 f3 0f 1e fa 64 8b 04 25 18 00 00 00 85 c0 75 10 b8 01 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 51 c3 48 83 ec 28 48 89 54 24 18 48 89 74 24
May 15 21:26:15 zen kernel: [ 2067.187001] RSP: 002b:00007fff06d3b6c8 EFLAGS: 00000246 ORIG_RAX: 0000000000000001
May 15 21:26:15 zen kernel: [ 2067.187002] RAX: ffffffffffffffda RBX: 0000000000000007 RCX: 00007f7c2d3adc27
May 15 21:26:15 zen kernel: [ 2067.187003] RDX: 0000000000000007 RSI: 000055ca3c99fbd0 RDI: 0000000000000001
May 15 21:26:15 zen kernel: [ 2067.187004] RBP: 000055ca3c99fbd0 R08: 000000000000000a R09: 000055ca3c99fbd0
May 15 21:26:15 zen kernel: [ 2067.187004] R10: 0000000000000077 R11: 0000000000000246 R12: 0000000000000007
May 15 21:26:15 zen kernel: [ 2067.187005] R13: 00007f7c2d4876c0 R14: 00007f7c2d4884a0 R15: 00007f7c2d4878a0
May 15 21:26:15 zen kernel: [ 2067.187007] ---[ end trace 617405aa79b78597 ]---
May 15 21:26:15 zen kernel: [ 2067.187019] ------------[ cut here ]------------
'''

'''
# modprobe --showconfig nvidia | grep "options nvidia"
#
# cat /usr/lib/modprobe.d/nvidia-graphics-drivers.conf
blacklist nouveau
blacklist lbm-nouveau
alias nouveau off
alias lbm-nouveau off
#
'''

tobz1000, 2021-07-09:
> So commenting out `options nvidia NVreg_PreserveVideoMemoryAllocations=1` in `/usr/lib/modprobe.d/nvidia-graphics-drivers.conf` before rebooting fixed the problem on my end. Oddly enough, creating a [...]

Just to add a data point, this seems to have fixed the issue for me too. Thank you @sm-Fifteen!

tobz1000, 2021-07-09:
> > So commenting out `options nvidia NVreg_PreserveVideoMemoryAllocations=1` in `/usr/lib/modprobe.d/nvidia-graphics-drivers.conf` before rebooting fixed the problem on my end. Oddly enough, creating a new file with `options nvidia NVreg_PreserveVideoMemoryAllocations=0` in `/etc/modeprobe.d` didn't override the value properly, you really have to comment out the option in /usr/lib.
> 
> Updates to the nvidia driver also seem to overwrite this, so you have to do this again after updating.
> 
> @ids1024: This is a consequence of [de515f6](https://github.com/pop-os/nvidia-graphics-drivers/commit/de515f69a4816b6e11633faba2d11b1f2a738c55), BTW. What was that commit meant to fix?

The commit message mentions fixing issues with WebRender. Now that I've disabled the setting, I believe I'm experiencing the issue it was meant to fix: Firefox does not render at all after waking from sleep, the window is just a white rectangle. However, triggering any redraw (e.g. Alt+Tab) gets it back to normal, so not a biggie (compared to being unable to wake from sleep!).