pop-os/nvidia-graphics-drivers #94: No games working with latest 465 drivers.

Vlek, 2021-05-04:
Saw update in Pop shop, attempted to install. Got errors stating that it could not. Updated drivers via commandline. Now Pop shop is showing no active updates.

Every time I attempt to play games that were previously working they fail. I have restarted multiple times as well.

Errors received when attempting to update this morning were as stated in this reddit post: https://www.reddit.com/r/pop_os/comments/n4t452/failed_to_update_operating_system_updates/

rmbleeker, 2021-05-04:
This bit me as well just now. Running updates on the commandline installed nvidia-driver-465, after which Steam fails to launch. The quickest solution I could find currently is to purge the nvidia 465 driver and install the nvidia 460 "server" driver.

On the commandline:
```
sudo apt-get purge .*nvidia.*465
sudo apt-get install nvidia-driver-460-server
```

If you prefer a GUI solution download "Software & Updates" from the Pop Shop and under "Additional Drivers" select nvidia-driver-460-server. Hit the "Apply Changes" button to install it. 

Reboot afterwards and you should be able to get back to gaming. YMMV.

bjr0421, 2021-05-04:
Having an issue related to this. I have 14 packages held back by apt, and cannot install the 465 drivers using the Pop shop. It seems like there are multiple people on the Pop!_OS subreddit having the same issue.

rmbleeker, 2021-05-04:
@bjr0421 consider yourself lucky. It seems the nvidia-465 driver can not be installed through the Pop Shop, only throught the command line. For now I would advice against installing it, since you won't be able to start Steam afterwards as well as many games.

Vlek, 2021-05-04:
@rmbleeker
> Reboot afterwards and you should be able to get back to gaming. YMMV.

As stated, this appears to be the preferred method for dealing with the issue as of right now per reddit posts I am seeing. I have tried the above method and I can confirm that it worked for me.

bflanagin, 2021-05-04:
Can you give us more information about your systems? We test the Nvidia driver for several weeks to make sure everything works correctly before we ship them.

bjr0421, 2021-05-04:
> Can you give us more information about your systems? We test the Nvidia driver for several weeks to make sure everything works correctly before we ship them.

I'm using a Dell XPS 15 9570, on Pop!_OS 20.10 with kernel 5.11. 
I just tried to update to 465 using the Pop shop again, and it worked. Nothing had changed or updated since my last comment, so unsure what changed. Also, this fixed the issue related to packages being held back.

bjr0421, 2021-05-04:
Update: Steam fails to launch using the NVIDIA GPU. Switched from Hybrid Graphics to NVIDIA Graphics and now Steam will not run. The log from the terminal is attached below.

`Running Steam on pop 20.10 64-bit
STEAM_RUNTIME is enabled automatically
Pins up-to-date!
Steam client's requirements are satisfied
WARNING: Using default/fallback debugger launch
/home/bjr0421/.steam/debian-installation/ubuntu12_32/steam -nominidumps -nobreakpad --help
[2021-05-04 16:38:44] Startup - updater built Apr 12 2021 18:51:36
SteamUpdateUI: An X Error occurred
X Error of failed request:  BadValue (integer parameter out of range for operation)
Major opcode of failed request:  152 (GLX)
Minor opcode of failed request:  3 (X_GLXCreateContext)
Value in failed request:  0x0
Serial number of failed request:  50
xerror_handler: X failed, continuing
Looks like steam didn't shutdown cleanly, scheduling immediate update check
[2021-05-04 16:38:44] Using the following download hosts for Public, Realm steamglobal
[2021-05-04 16:38:44] 1. https://cdn.cloudflare.steamstatic.com, /client/, Realm 'steamglobal', weight was 100, source = 'update_hosts_cached.vdf'
[2021-05-04 16:38:44] 2. https://cdn.akamai.steamstatic.com, /client/, Realm 'steamglobal', weight was 100, source = 'update_hosts_cached.vdf'
[2021-05-04 16:38:44] 3. http://media.steampowered.com, /client/, Realm 'steamglobal', weight was 1, source = 'baked in'
[2021-05-04 16:38:44] Checking for update on startup
[2021-05-04 16:38:44] Checking for available updates...
[2021-05-04 16:38:44] Downloading manifest: https://cdn.cloudflare.steamstatic.com/client/steam_client_ubuntu12
[2021-05-04 16:38:45] Download skipped: /client/steam_client_ubuntu12 version 1618256785, installed version 1618256785, existing pending version 0
[2021-05-04 16:38:45] Nothing to do
[2021-05-04 16:38:45] Verifying installation...
[2021-05-04 16:38:45] Performing checksum verification of executable files
[2021-05-04 16:38:45] Verification complete
`

Vlek, 2021-05-04:
@bflanagin
> Can you give us more information about your systems? We test the Nvidia driver for several weeks to make sure everything works correctly before we ship them.

OS: Pop!_OS 20.10 x86_64 
Kernel: 5.11.0-7614-generic 
Uptime: 1 hour, 5 mins 
Packages: 2410 (dpkg), 22 (flatpak) 
Shell: zsh 5.8 
Resolution: 1920x1080, 1920x1080, 1920x1080, 1920x1080 
DE: GNOME 3.38.3 
WM: Mutter 
WM Theme: Pop 
Theme: Pop-dark [GTK2/3] 
Icons: Pop [GTK2/3] 
Terminal: tilix 
CPU: AMD Ryzen 9 3900X (24) @ 3.800GHz 
GPU: NVIDIA GeForce GTX 1080 Ti 
Memory: 3466MiB / 64286MiB

bflanagin, 2021-05-04:
@bjr0421 Are you using the flatpak version of steam or the deb version? If you're using the flat pak version (and you might want to try this anyway) you need to run updates for them as well.

`flatpak update`
and for good messure
`sudo flatpak update`

bjr0421, 2021-05-04:
@bflanagin 
> @bjr0421 Are you using the flatpak version of steam or the deb version? If you're using the flat pak version (and you might want to try this anyway) you need to run updates for them as well.
> 
> `flatpak update`
> and for good messure
> `sudo flatpak update`

Using the deb version, installed from the Pop shop.

bflanagin, 2021-05-04:
I'm going to test a few more machines and get a few others on it. I've seen a few games on steam not start correctly, but others still do. Not sure what the correlation is. 

@bjr0421 as far as the steam issue is concerned, its possible that that earlier error mucked things up. From the command line you can issue these two commands. 

```
sudo apt-get purge nvidia*
sudo apt-get install nvidia-driver-465
```
Once done restart the computer.

rmbleeker, 2021-05-04:
> Can you give us more information about your systems? We test the Nvidia driver for several weeks to make sure everything works correctly before we ship them.

I don't think it's an issue with the driver itself, but rather with some 32-bit library dependencies missing for the new driver version. I reverted my system back to `nvidia-driver-460-server` as described in my post, now when I want to install the latest vesion this is what apt tells me it's going to do:
```
$ sudo apt install nvidia-driver-465
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  libnvidia-cfg1-465 libnvidia-common-465 libnvidia-compute-465 libnvidia-decode-465 libnvidia-encode-465 libnvidia-extra-465 libnvidia-fbc1-465
  libnvidia-gl-465 libnvidia-ifr1-465 nvidia-compute-utils-465 nvidia-dkms-465 nvidia-kernel-common-465 nvidia-kernel-source-465 nvidia-utils-465
  xserver-xorg-video-nvidia-465
Recommended packages:
  libnvidia-compute-465:i386 libnvidia-decode-465:i386 libnvidia-encode-465:i386 libnvidia-ifr1-465:i386 libnvidia-fbc1-465:i386 libnvidia-gl-465:i386
The following packages will be REMOVED:
  libnvidia-cfg1-460-server libnvidia-common-460-server libnvidia-compute-460-server libnvidia-compute-460-server:i386 libnvidia-decode-460-server
  libnvidia-decode-460-server:i386 libnvidia-encode-460-server libnvidia-encode-460-server:i386 libnvidia-extra-460-server libnvidia-fbc1-460-server
  libnvidia-fbc1-460-server:i386 libnvidia-gl-460-server libnvidia-gl-460-server:i386 libnvidia-ifr1-460-server libnvidia-ifr1-460-server:i386
  nvidia-compute-utils-460-server nvidia-dkms-460-server nvidia-driver-460-server nvidia-kernel-common-460-server nvidia-kernel-source-460-server
  nvidia-utils-460-server xserver-xorg-video-nvidia-460-server
The following NEW packages will be installed:
  libnvidia-cfg1-465 libnvidia-common-465 libnvidia-compute-465 libnvidia-decode-465 libnvidia-encode-465 libnvidia-extra-465 libnvidia-fbc1-465
  libnvidia-gl-465 libnvidia-ifr1-465 nvidia-compute-utils-465 nvidia-dkms-465 nvidia-driver-465 nvidia-kernel-common-465 nvidia-kernel-source-465
  nvidia-utils-465 xserver-xorg-video-nvidia-465
0 upgraded, 16 newly installed, 22 to remove and 0 not upgraded.
Need to get 0 B/170 MB of archives.
After this operation, 133 MB disk space will be freed.
```
As you can see, there is a difference between the packages that are going to be removed and those that are going to be installed and those differences are all 32-bit library packages related to the nvidia driver:
```
libnvidia-compute-460:i386
libnvidia-decode-460:i386
libnvidia-encode-460:i386
libnvidia-fbc1-460:i386
libnvidia-gl-460:i386
libnvidia-ifr1-460:i386
```
When trying to install those packages manually I get all kind of nasty conflicts that I don't have on my screen anymore unfortunately, but I'm sure you'll be able to reproduce the issue with this information.

rmbleeker, 2021-05-04:
I just did the upgrade to nvidia-driver-465 using the command above. Even when adding the `--install-recommends` and `--install-suggests` flags apt won't install the 32-bit libraries. When trying a manual installation of those packages apt complains there are no installation candidates for those packages:
```
$ sudo apt install libnvidia-compute-465:i386 libnvidia-decode-465:i386 libnvidia-encode-465:i386 libnvidia-ifr1-465:i386 libnvidia-fbc1-465:i386 libnvidia-gl-465:i386
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package libnvidia-compute-465:i386 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libnvidia-compute-465

Package libnvidia-decode-465:i386 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libnvidia-encode-465:i386 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libnvidia-ifr1-465:i386 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libnvidia-fbc1-465:i386 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libnvidia-gl-465:i386 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libnvidia-common-465

E: Package 'libnvidia-compute-465:i386' has no installation candidate
E: Package 'libnvidia-decode-465:i386' has no installation candidate
E: Package 'libnvidia-encode-465:i386' has no installation candidate
E: Package 'libnvidia-ifr1-465:i386' has no installation candidate
E: Package 'libnvidia-fbc1-465:i386' has no installation candidate
E: Package 'libnvidia-gl-465:i386' has no installation candidate
```

rmbleeker, 2021-05-04:
What strikes me as weird is that when trying to install the regular `nvidia-driver-460` package, not the "server" variant, apt will actually try to install the nvidia 465 driver instead of 460. So it might just be some weird dependency issue?
```
$ sudo apt install nvidia-driver-460
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  libnvidia-cfg1-465 libnvidia-common-465 libnvidia-compute-465 libnvidia-decode-465 libnvidia-encode-465 libnvidia-extra-465 libnvidia-fbc1-465
  libnvidia-gl-465 libnvidia-ifr1-465 nvidia-compute-utils-465 nvidia-dkms-465 nvidia-driver-465 nvidia-kernel-common-465 nvidia-kernel-source-465
  nvidia-utils-465 xserver-xorg-video-nvidia-465
Recommended packages:
  libnvidia-compute-465:i386 libnvidia-decode-465:i386 libnvidia-encode-465:i386 libnvidia-ifr1-465:i386 libnvidia-fbc1-465:i386 libnvidia-gl-465:i386
The following NEW packages will be installed:
  libnvidia-cfg1-465 libnvidia-common-465 libnvidia-compute-465 libnvidia-decode-465 libnvidia-encode-465 libnvidia-extra-465 libnvidia-fbc1-465
  libnvidia-gl-465 libnvidia-ifr1-465 nvidia-compute-utils-465 nvidia-dkms-465 nvidia-driver-460 nvidia-driver-465 nvidia-kernel-common-465
  nvidia-kernel-source-465 nvidia-utils-465 xserver-xorg-video-nvidia-465
0 upgraded, 17 newly installed, 0 to remove and 0 not upgraded.
Need to get 5.536 B/170 MB of archives.
After this operation, 498 MB of additional disk space will be used.
```

bjr0421, 2021-05-04:
@bflanagin 
Followed through and removed the NVIDIA drivers, then reinstalled them. Still having the same issue after rebooting unfortunately.

cybik, 2021-05-05:
Adding my report to the pool, after the update to 465 I can't launch Steam at all.

cybik, 2021-05-05:
I believe I somewhat diagnosed the issue: somehow, their CI has ignored i386 packages for 20.10, likely removing any i386 "side-packages" we would have had cleaning up the 460 series when installing 465. I will attempt hard-installing the i386 packages by hand and circle back.

cybik, 2021-05-05:
While I don't recommend it seeing as you'll be breaking your apt stuffs, force-installing the i386 packages from 21.xx onti 20.xx lets me launch Steam.

Again, I don't recommend it, but right now it's the one way.

mmstick, 2021-05-05:
Launchpad is arbitrarily not building i386 packages for focal or groovy. Unsure why at the moment. Our CI is packaging the i386 packages just fine for focal and groovy.

mmstick, 2021-05-05:
Might be this: https://discourse.ubuntu.com/t/i386-packages-in-launchpad-ppas/15367/4

> Launchpad only creates i386 PPA builds for packages that have i386 support in Ubuntu itself. There isn’t currently a way for a PPA owner to override this.

cybik, 2021-05-05:
@mmstick but then why is 21.04 getting the i386es

mmstick, 2021-05-05:
That I don't know. If you inspect the `.dsc`'s, they do in fact request for i386 builds, but Launchpad won't even attempt to start a i386 package build.

sigg3, 2021-05-05:
I have the same issue, also missing the i386 dependencies, attaching term.log:

`Log started: 2021-05-05  09:37:18
(Reading database ... 820547 files and directories currently installed.)
Preparing to unpack .../pop-gtk-theme_5.3.3~1619646713~20.10~5d60d02_all.deb ...
unlink: cannot unlink '/usr/share/themes/Pop/gtk-3.0': Is a directory
unlink: cannot unlink '/usr/share/themes/Pop-dark/gtk-3.0': Is a directory
Unpacking pop-gtk-theme (5.3.3~1619646713~20.10~5d60d02) over (5.3.3~1617290584~20.10~19f019b) ...
Setting up pop-gtk-theme (5.3.3~1619646713~20.10~5d60d02) ...
(Reading database ... 820547 files and directories currently installed.)
Preparing to unpack .../pop-gnome-shell-theme_5.3.3~1619646713~20.10~5d60d02_all.deb ...
Unpacking pop-gnome-shell-theme (5.3.3~1619646713~20.10~5d60d02) over (5.3.3~1617290584~20.10~19f019b) ...
Setting up pop-gnome-shell-theme (5.3.3~1619646713~20.10~5d60d02) ...
(Reading database ... 820547 files and directories currently installed.)
Preparing to unpack .../pop-shell_1.1.0~1619668471~20.10~a3fa53e_all.deb ...
Unpacking pop-shell (1.1.0~1619668471~20.10~a3fa53e) over (1.1.0~1619554528~20.10~42e8ac0) ...
Setting up pop-shell (1.1.0~1619668471~20.10~a3fa53e) ...
(Reading database ... 820548 files and directories currently installed.)
Preparing to unpack .../pop-sound-theme_5.3.3~1619646713~20.10~5d60d02_all.deb ...
Unpacking pop-sound-theme (5.3.3~1619646713~20.10~5d60d02) over (5.3.3~1617290584~20.10~19f019b) ...
Setting up pop-sound-theme (5.3.3~1619646713~20.10~5d60d02) ...
(Reading database ... 820548 files and directories currently installed.)
Preparing to unpack .../system76-power_1.1.16~1620080479~20.10~b74bcb7_amd64.deb ...
Unpacking system76-power (1.1.16~1620080479~20.10~b74bcb7) over (1.1.16~1618853881~20.10~29304ec) ...
Setting up system76-power (1.1.16~1620080479~20.10~b74bcb7) ...
(Reading database ... 820548 files and directories currently installed.)
Preparing to unpack .../00-libnvidia-cfg1-455_460.73.01-0ubuntu0.20.10.1_amd64.deb ...
Unpacking libnvidia-cfg1-455:amd64 (460.73.01-0ubuntu0.20.10.1) over (460.67-1pop0~1616430777~20.10~71e1ad1) ...
Preparing to unpack .../01-nvidia-compute-utils-455_460.73.01-0ubuntu0.20.10.1_amd64.deb ...
Unpacking nvidia-compute-utils-455:amd64 (460.73.01-0ubuntu0.20.10.1) over (460.67-1pop0~1616430777~20.10~71e1ad1) ...
Preparing to unpack .../02-nvidia-kernel-common-460_460.73.01-0ubuntu0.20.10.1_amd64.deb ...
Unpacking nvidia-kernel-common-460 (460.73.01-0ubuntu0.20.10.1) over (460.67-1pop0~1616430777~20.10~71e1ad1) ...
Preparing to unpack .../03-nvidia-dkms-460_460.73.01-0ubuntu0.20.10.1_amd64.deb ...
Removing all DKMS Modules
Done.
Unpacking nvidia-dkms-460 (460.73.01-0ubuntu0.20.10.1) over (460.67-1pop0~1616430777~20.10~71e1ad1) ...
Preparing to unpack .../04-nvidia-dkms-455_460.73.01-0ubuntu0.20.10.1_amd64.deb ...
Unpacking nvidia-dkms-455 (460.73.01-0ubuntu0.20.10.1) over (460.67-1pop0~1616430777~20.10~71e1ad1) ...
Preparing to unpack .../05-nvidia-driver-455_460.73.01-0ubuntu0.20.10.1_amd64.deb ...
Unpacking nvidia-driver-455 (460.73.01-0ubuntu0.20.10.1) over (460.67-1pop0~1616430777~20.10~71e1ad1) ...
Preparing to unpack .../06-nvidia-kernel-common-455_460.73.01-0ubuntu0.20.10.1_amd64.deb ...
Unpacking nvidia-kernel-common-455:amd64 (460.73.01-0ubuntu0.20.10.1) over (460.67-1pop0~1616430777~20.10~71e1ad1) ...
Preparing to unpack .../07-nvidia-kernel-source-455_460.73.01-0ubuntu0.20.10.1_amd64.deb ...
Unpacking nvidia-kernel-source-455 (460.73.01-0ubuntu0.20.10.1) over (460.67-1pop0~1616430777~20.10~71e1ad1) ...
Preparing to unpack .../08-nvidia-utils-455_460.73.01-0ubuntu0.20.10.1_amd64.deb ...
Unpacking nvidia-utils-455:amd64 (460.73.01-0ubuntu0.20.10.1) over (460.67-1pop0~1616430777~20.10~71e1ad1) ...
Preparing to unpack .../09-openvpn_2.4.9-3ubuntu1.1_amd64.deb ...
Unpacking openvpn (2.4.9-3ubuntu1.1) over (2.4.9-3ubuntu1) ...
Preparing to unpack .../10-pop-shell-plugin-system76-power_1.1.0~1619668471~20.10~a3fa53e_all.deb ...
Unpacking pop-shell-plugin-system76-power (1.1.0~1619668471~20.10~a3fa53e) over (1.1.0~1619554528~20.10~42e8ac0) ...
Preparing to unpack .../11-xserver-xorg-video-nvidia-455_460.73.01-0ubuntu0.20.10.1_amd64.deb ...
Unpacking xserver-xorg-video-nvidia-455:amd64 (460.73.01-0ubuntu0.20.10.1) over (460.67-1pop0~1616430777~20.10~71e1ad1) ...
Setting up openvpn (2.4.9-3ubuntu1.1) ...
Setting up xserver-xorg-video-nvidia-455:amd64 (460.73.01-0ubuntu0.20.10.1) ...
Setting up nvidia-kernel-common-460 (460.73.01-0ubuntu0.20.10.1) ...
update-initramfs: deferring update (trigger activated)
Setting up nvidia-kernel-source-455 (460.73.01-0ubuntu0.20.10.1) ...
Setting up nvidia-driver-455 (460.73.01-0ubuntu0.20.10.1) ...
Setting up libnvidia-cfg1-455:amd64 (460.73.01-0ubuntu0.20.10.1) ...
Setting up nvidia-utils-455:amd64 (460.73.01-0ubuntu0.20.10.1) ...
Setting up nvidia-compute-utils-455:amd64 (460.73.01-0ubuntu0.20.10.1) ...
Setting up nvidia-dkms-460 (460.73.01-0ubuntu0.20.10.1) ...
update-initramfs: deferring update (trigger activated)
INFO:Enable nvidia
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/lenovo_thinkpad
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/dell_latitude
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/put_your_quirks_here
Loading new nvidia-460.73.01 DKMS files...
Building for 5.11.0-7614-generic
Building for architecture x86_64
[... 481 lines cut ...]
    ESP Path:............/boot/efi
    ESP Partition:......./dev/nvme0n1p1
[... cut ...]
kernelstub.Installer : INFO     Copying initrd.img into ESP
kernelstub.Installer : INFO     Setting up loader.conf configuration
kernelstub.Installer : INFO     Making entry file for Pop!_OS
kernelstub.Installer : INFO     Backing up old kernel
kernelstub.Installer : INFO     Making entry file for Pop!_OS
Log ended: 2021-05-05  10:08:05` 

I need my computer today, so I found this workaround (YMMV):

`sudo apt-get purge nvidia*
sudo apt-get install nvidia-driver-450
sudo reboot`

This is suboptimal, but gets me through the day :P

trevorjamesmartin, 2021-05-05:
```
sudo apt update
sudo apt-get install nvidia-driver-460-server
```

fixes the issue on my system

cybik, 2021-05-05:
@mmstick i see that the answer for now is to yank the 20.xx packages until a resolution can be reached. still doesn't fix it for those of us that are on the newer packages though.

luizfelipevbll, 2021-05-05:
I had the same issue, I cant wait for a new release so just executed 
```sh
sudo apt-get purge nvidia*
sudo apt-get install nvidia-driver-460

```
This solved my problem ponctually

Redbatman89, 2021-05-05:
I had to use timeshift again. This update caused steam unable launch for me

cybik, 2021-05-05:
Circling back on my previous postings: again not recommended, but another countermeasure that would allow you to use 465 with 20.xx, is to _**manually**_ install the packages for **_18.04_** into 20.xx via commands.

Obviously I will not spell it out (if you know, you know), but installing all but the headless packages has resulted in my package configuration not having broken packages, and I have all the required packages on amd64 and the requisite alternate libraries forcibly installed via the i386 counterparts.

YMMV and all that, I am not responsible if you break your system with my advice and `n=1` experience, yadda yadda yadda.

hovo1990, 2021-05-05:
as @luizfelipevbll  mentioned, the command fixed the problem 

sudo apt-get purge nvidia*
sudo apt-get install nvidia-driver-460 

but for steam launching again, had to install 
sudo apt-get install libnvidia-gl-460:i386

(Here you can find more info: https://forums.developer.nvidia.com/t/cannot-start-steam-after-updating-to-driver-465/176850)

henriqueffc, 2021-05-05:
https://www.reddit.com/r/pop_os/comments/n5j8c0/nvidia_driver_4652402_issues/

Hello Pop!_OS NVIDIA users! We released NVIDIA driver 465.24.02 yesterday but unfortunately failed to detect that the 32-bit binaries had not been published in the Launchpad PPA. This causes upgrade issues and issues launching 32-bit binaries, which includes a number of games. We deleted these packages, so that the NVIDIA driver will be downgraded to 460.67 which should still be functional.

We have discovered the cause of this to be the i386 whitelist in place on Launchpad. Our staging repository was able to build the i386 package, so our testing was successful. When releasing on Launchpad, because nvidia-graphics-drivers-465 is not in the whitelist for 20.04 and 20.10, it did not build 32-bit binaries for those distributions. It did, however, build them for 18.04 and 21.04.

This issue has been reported to Ubuntu at the following links:

- https://discourse.ubuntu.com/t/i386-packages-in-launchpad-ppas/15367/6

- https://discourse.ubuntu.com/t/community-process-for-32-bit-compatibility/12598/107

We are working to update 20.04 and 20.10 to NVIDIA driver 460.73.01, which includes many of the improvements in 465.24.02, but will presumably build the 32-bit binaries correctly.

Thank you for your patience!

henriqueffc, 2021-05-06:
> The following text was published on reddit:
> 
> https://www.reddit.com/r/pop_os/comments/n5j8c0/nvidia_driver_4652402_issues/
> 
> Hello Pop!_OS NVIDIA users! We released NVIDIA driver 465.24.02 yesterday but unfortunately failed to detect that the 32-bit binaries had not been published in the Launchpad PPA. This causes upgrade issues and issues launching 32-bit binaries, which includes a number of games. We deleted these packages, so that the NVIDIA driver will be downgraded to 460.67 which should still be functional.
> 
> We have discovered the cause of this to be the i386 whitelist in place on Launchpad. Our staging repository was able to build the i386 package, so our testing was successful. When releasing on Launchpad, because nvidia-graphics-drivers-465 is not in the whitelist for 20.04 and 20.10, it did not build 32-bit binaries for those distributions. It did, however, build them for 18.04 and 21.04.
> 
> This issue has been reported to Ubuntu at the following links:
> 
>     * https://discourse.ubuntu.com/t/i386-packages-in-launchpad-ppas/15367/6
> 
>     * https://discourse.ubuntu.com/t/community-process-for-32-bit-compatibility/12598/107
> 
> 
> We are working to update 20.04 and 20.10 to NVIDIA driver 460.73.01, which includes many of the improvements in 465.24.02, but will presumably build the 32-bit binaries correctly.
> 
> Thank you for your patience!

I made the last update made available to the system. Worked well. I use the system in hybrid mode. Pop! _OS 20.04 LTS.

[](https://imgur.com/VzDySrL)

They did what they announced on Reddit.

sigg3, 2021-05-06:
The issue seems to be resolved now.

leviport, 2021-05-06:
This looks to be resolved by https://github.com/pop-os/nvidia-graphics-drivers/pull/96 and https://github.com/pop-os/nvidia-graphics-drivers/pull/97. 

If anyone's Nvidia drivers are still in a strange state, try purging and reinstalling the driver:
```
sudo apt purge ~nnvidia
sudo apt install nvidia-driver-460
```

If that doesn't work, please report here so we can reopen the issue.

closed 2021-05-06 by leviport

schuelermine, 2021-05-13:
In case you're getting errors with the listed commands, your shell may be expanding `~` to your home directory (`zsh` and `fish` do this, for  instance).

Try  
`sudo apt purge "~nnvidia"` or  
`sudo apt purge \~nnvidia`
instead of the first command.

Jnewbon, 2021-08-03:
After been having nvidia driver problems for a while which came to a head when i was forced to OS upgrade, only way i was able to do so was the purge the nvidia drivers from my machine, let the OS upgrade to 21.04 and attempt to reinstall.

The errors I'm getting form the 470 driver are almost identical to some of the errors in issues marked as duplicate of this (#98)

```
Loading new nvidia-470.57.02 DKMS files...
Building for 5.11.0-7620-generic
Building for architecture x86_64
Building initial module for 5.11.0-7620-generic
ERROR (dkms apport): kernel package linux-headers-5.11.0-7620-generic is not supported
Error! Bad return status for module build on kernel: 5.11.0-7620-generic (x86_64)
Consult /var/lib/dkms/nvidia/470.57.02/build/make.log for more information.
dpkg: error processing package nvidia-dkms-470 (--configure):
 installed nvidia-dkms-470 package post-installation script subprocess returned error exit status 10
dpkg: dependency problems prevent configuration of nvidia-driver-470:
 nvidia-driver-470 depends on nvidia-dkms-470 (>= 470.57.02); however:
  Package nvidia-dkms-470 is not configured yet.
```

The only driver i am able to successfully install is the nvidia-driver-390