pop-os/nvidia-graphics-drivers #95: Unable to run OS update on pop shop (nvidia driver version conflict)

opusmagna, 2021-05-05:
Since 2 days ago, I am unable to run the system updates in pop shop where the failure details are as follows;

```
The following packages have unmet dependencies:
  nvidia-driver-460: Depends: nvidia-kernel-source-460 (= 460.67-1pop0~1616430777~20.10~71e1ad1) but 460.73.01-0ubuntu0.20.10.1 is to be installed
                     Depends: libnvidia-extra-460 (= 460.67-1pop0~1616430777~20.10~71e1ad1) but 460.73.01-0ubuntu0.20.10.1 is to be installed
                     Depends: nvidia-compute-utils-460 (= 460.67-1pop0~1616430777~20.10~71e1ad1) but 460.73.01-0ubuntu0.20.10.1 is to be installed
                     Depends: nvidia-utils-460 (= 460.67-1pop0~1616430777~20.10~71e1ad1) but 460.73.01-0ubuntu0.20.10.1 is to be installed
                     Depends: xserver-xorg-video-nvidia-460 (= 460.67-1pop0~1616430777~20.10~71e1ad1) but 460.73.01-0ubuntu0.20.10.1 is to be installed
                     Depends: libnvidia-cfg1-460 (= 460.67-1pop0~1616430777~20.10~71e1ad1) but 460.73.01-0ubuntu0.20.10.1 is to be installed
```

Something wrong with the update release maybe? Please note that no manual nvidia driver installation was performed on the machine. 

Additionally, there is nvidia-driver-460 at version 460.67 awaiting to be updated. That also fails with the following error/info output;
```
The following packages have unmet dependencies:
  nvidia-driver-460: Depends: libnvidia-gl-460 (= 460.73.01-0ubuntu0.20.10.1) but 460.67-1pop0~1616430777~20.10~71e1ad1 is to be installed
                     Depends: nvidia-kernel-source-460 (= 460.73.01-0ubuntu0.20.10.1) but 460.67-1pop0~1616430777~20.10~71e1ad1 is to be installed
                     Depends: libnvidia-compute-460 (= 460.73.01-0ubuntu0.20.10.1) but 460.67-1pop0~1616430777~20.10~71e1ad1 is to be installed
                     Depends: libnvidia-extra-460 (= 460.73.01-0ubuntu0.20.10.1) but 460.67-1pop0~1616430777~20.10~71e1ad1 is to be installed
                     Depends: nvidia-compute-utils-460 (= 460.73.01-0ubuntu0.20.10.1) but 460.67-1pop0~1616430777~20.10~71e1ad1 is to be installed
                     Depends: libnvidia-decode-460 (= 460.73.01-0ubuntu0.20.10.1) but 460.67-1pop0~1616430777~20.10~71e1ad1 is to be installed
                     Depends: libnvidia-encode-460 (= 460.73.01-0ubuntu0.20.10.1) but 460.67-1pop0~1616430777~20.10~71e1ad1 is to be installed
                     Depends: nvidia-utils-460 (= 460.73.01-0ubuntu0.20.10.1) but 460.67-1pop0~1616430777~20.10~71e1ad1 is to be installed
                     Depends: xserver-xorg-video-nvidia-460 (= 460.73.01-0ubuntu0.20.10.1) but 460.67-1pop0~1616430777~20.10~71e1ad1 is to be installed
                     Depends: libnvidia-cfg1-460 (= 460.73.01-0ubuntu0.20.10.1) but 460.67-1pop0~1616430777~20.10~71e1ad1 is to be installed
                     Depends: libnvidia-ifr1-460 (= 460.73.01-0ubuntu0.20.10.1) but 460.67-1pop0~1616430777~20.10~71e1ad1 is to be installed
                     Depends: libnvidia-fbc1-460 (= 460.73.01-0ubuntu0.20.10.1) but 460.67-1pop0~1616430777~20.10~71e1ad1 is to be installed
                     Recommends: nvidia-prime (>= 0.8)
                     Recommends: libnvidia-compute-460:i386 (= 460.73.01-0ubuntu0.20.10.1)
                     Recommends: libnvidia-decode-460:i386 (= 460.73.01-0ubuntu0.20.10.1)
                     Recommends: libnvidia-encode-460:i386 (= 460.73.01-0ubuntu0.20.10.1)
                     Recommends: libnvidia-ifr1-460:i386 (= 460.73.01-0ubuntu0.20.10.1)
                     Recommends: libnvidia-fbc1-460:i386 (= 460.73.01-0ubuntu0.20.10.1)
                     Recommends: libnvidia-gl-460:i386 (= 460.73.01-0ubuntu0.20.10.1)
```

I do not think this causes any problems with the way the GPU works, however I am currently unable to receive the OS updates as mentioned. 
Is there any workaround to this? Thanks in advance.

bflanagin, 2021-05-05:
duplicate of  https://github.com/pop-os/nvidia-graphics-drivers/issues/94. Please follow this ticket for updates

closed 2021-05-05 by bflanagin

bflanagin, 2021-05-05:
Duplicate of  https://github.com/pop-os/nvidia-graphics-drivers/issues/94. Please follow this ticket for updates