pop-os/nvidia-graphics-drivers #102: Graphical issues after suspend (Gnome + Nvidia)

QEDeD, 2021-05-13:
**Distribution (run `cat /etc/os-release`):**
 20.10
 
 **Related Application and/or Package Version:**
Nvidia driver version: 460.73.01

**Issue/Bug Description:**
After waking from suspend, the desktop is basically unusable and requires a reboot due to graphical glitching.
I will try to make a gif of the issue next time it appears.

**Steps to reproduce (if you know):**

- Suspend
- Wake up 
- The issue is there

**Expected behavior:**
Suspend shouldn't break anything.

QEDeD, 2021-05-13:
Sure thing :) 

CPU: AMD Ryzen 5 5600X 
GPU: GeForce GTX 1070 Ti
Mobo: Gigabyte X570 AORUS ELITE
RAM: 4 x 8 GB DDR4 @ 3600 MHz (F4-3600C16-8GTZKW)
Storage Space:  Samsung SSD 970 EVO Plus 1TB

AlekzNet, 2021-05-15:
Same thing. Posted here: https://github.com/pop-os/nvidia-graphics-drivers/issues/90#issuecomment-841732284

bflanagin, 2021-07-23:
Sorry for the delay in updating this issue. We weren't able to reproduce the issue on our lab units, and have since dropped support for 20.10. If you are still experiencing the issue on 21.04 and the latest Nvidia driver feel free to create a new issue.

closed 2021-07-23 by bflanagin