pop-os/nvidia-graphics-drivers #112: Feature request: way to install the previous driver version

leviport, 2021-07-15:
There is regularly [unrest](https://github.com/pop-os/nvidia-graphics-drivers/issues/65) when new NVIDIA driver versions are released. If users had a way to roll back from troublesome upgrades, I think NVIDIA drivers would be a lot less frustrating for many users.

prittjam, 2021-07-16:
This is a much desired feature and would make PopOS more stable for nvidia users. Unfortunately, I've experienced several problems during nvidia upgrades, and I think many others have, too. There is just too much hardware variety out there. If the user has an ability to revert to a known working driver, then it would save a lot of time trying strange kludges to stop PopOS from installing the latest nvidia driver.

SamSpiri, 2021-08-19:
I can't see teams video in calls anymore. This is caused by new nvidia drivers. I want to rollback to 455 or 460 but to do this I need to hit my system hard with vanilla drivers. I've got bad filling about doing this and I have a risk to spend a few days in terminal until I get it working. Not nice place with darkness and fire everywhere. I've been there a few times in my life unfortunately.

trancilo, 2021-09-15:
Experiencing issues with nvidia 470 on my Lenovo p17 Gen1
Not been able to downgrade.
Solution was a reinstall with an old Popos image "pop-os_20.04_amd64_nvidia_24.iso"
And hold all nvidia-460 packages

`sudo apt-mark hold  libnvidia-cfg1-460 libnvidia-common-460 libnvidia-compute-460 libnvidia-decode-460 libnvidia-encode-460 libnvidia-extra-460 libnvidia-fbc1-460 libnvidia-gl-460 libnvidia-ifr1-460 libxnvctrl0 nvidia-compute-utils-460 nvidia-dkms-460 nvidia-driver-460 nvidia-kernel-common-460 nvidia-kernel-source-460 nvidia-settings nvidia-utils-460 xserver-xorg-video-nvidia-460`

Not the most beautiful solution but at least I don't have to hold back all package updates
Not great, not terrible

frbaroni, 2021-10-01:
My setup just crashed with flickering issues, +1 for this issue.

Daasin, 2022-01-11:
Was there any update on this?

trancilo, 2022-01-24:
Tried to update my popos20.04 to nvidia-470 again. Thought maybe my [issue](https://github.com/pop-os/pop/issues/1854#issuecomment-919959399) would be fixed. Its not.
But I did find a solution to downgrade to the previous driver again.
`apt install software-properties-gtk`
`software-properties-gtk`
In this UI I could select nvidia driver 460. After pressing Apply I got some errors. But closing and re-opening software-properties-gtk showed my it was successful. 
After reboot I was indeed on 460 again.

leviport, 2022-03-03:
With the 510 update, the 470 driver has also been made available for users who need it.

closed 2022-03-03 by leviport