pop-os/nvidia-graphics-drivers #115: Use last drivers and activate wayland

johnnync13, 2021-07-23:
[image]

jacobgkau, 2021-07-23:
Can you please elaborate on the issue? What is the expected behavior and what is the behavior you are seeing instead? Is this a feature request to enable Wayland by default?

johnnync13, 2021-07-23:
The iso comes by default with nvidia 460 but in version 470 is when nvidia has solved many problems with wayland. It would be convenient that the iso comes with the version I indicate and give the option to activate wayland in the installation. @jacobgkau

jacobgkau, 2021-07-23:
> The iso comes by default with nvidia 460 but in version 470 is when nvidia has solved many problems with wayland. It would be convenient that the iso comes with the version I indicate

NVIDIA driver version 470.57.02 has been released this morning, and new iso images are being built right now (although they will need to be tested and may not be released until next week.) You can already update an existing system with `sudo apt update`/`sudo apt upgrade` or via the Pop!_Shop.

> and give the option to activate wayland in the installation. @jacobgkau

Wayland is not officially supported in Pop!_OS 20.04 or 21.04, although it is on the roadmap for future versions. You can comment out the line `WaylandEnable=false` in `/etc/gdm3/custom.conf` to allow it as an option in GDM.

johnnync13, 2021-07-23:
could you send me the iso?

jacobgkau, 2021-07-23:
> could you send me the iso?

No, the iso is still being built and is not released yet. Once it is released, it will appear at [pop.system76.com](https://pop.system76.com/). (The current iso version is 6 for 21.04 and 30 for 20.04, so once the version number increments higher than that, you'll know they've been updated.)

closed 2021-07-23 by johnnync13