pop-os/nvidia-graphics-drivers #118: Nvidia drivers fail to load

plakkiesza, 2021-08-20:
[nvidia-bug-report.log.gz](https://github.com/pop-os/nvidia-graphics-drivers/files/7023637/nvidia-bug-report.log.gz)
I Select the Nvidia graphics in the power settings menu then restart(Ive also tried this with the cli.). After restart it shows that its there and selected. However it is definitely not loaded  I get from Nvidia-smi. 

NVIDIA-SMI has failed because it couldn't communicate with the NVIDIA driver. Make sure that the latest NVIDIA driver is installed and running.

I get things like this in syslog, The NVIDIA GPU 0000:01:00.0 (PCI ID: 10de:2191) installed in this system has fallen off the bus and is not responding to commands.

The system works perfectly in windows, I also had it running on LInux mint 19.3.

I have tried reinstalling drivers in so many ways. Im getting so frustrated with the system.

I really like the look and feel of POP OS. I'm very keen to buy a pc that comes with it. Just need to save up a bit.

Can anyone help me with this problem. I have spent days going through google and log files to no avail. The main reason I tried pop os was because I saw an article praising it for working with NVIDIA. Id really like to carry on using it if possible.

Thanks in advance for any help.

jacobgkau, 2021-08-20:
What hardware are you using? What laptop model, motherboard, and GPU?

plakkiesza, 2021-08-20:
Apologies, I thought it was in the attached file, oops.

I have a Dell G3 3590, This has a NVIDIA GTX 1660 Ti graphics card. Im using Pop os 20 LTS version.

Slackadays, 2021-10-30:
I had a very similar issue with my HP Spectre which has a GTX 1650 Ti instead. The fix was to just try to install either version of PopOS from September or newer because of some recent kernel or package change that fixed compatibility, or there was a BIOS update from Windows.