pop-os/nvidia-graphics-drivers #133: Text & Icon corruption after resume from suspend on Nvidia

Rick1029, 2022-02-05:
After resuming from suspend on Nvidia graphics, about 60% of the time it results in major corruption of text and icons requiring a shell restart to fix. This seems to be true independent of any other factor. This has only ever occurred to me on Nvidia graphics. Only gnome and gnome extensions is affected - all apps I've seen work normally. See example photos below.

Pop 21.10
Gnome 40.4.0 X11 (have not tried Wayland)
Cosmic version 1
nvidia-driver-495:
  Installed: 495.46-0ubuntu0.21.10.1
  Candidate: 495.46-0ubuntu0.21.10.1

Login screen after resume.
[image]

Desktop in workspace overview.
[image]

Photo of arcmenu. The Cosmic launchers have the same corruption but I can't screenshot them as they disappear when I click to take the screenshot.
[image]

Portion of the status bar.
[image]

Some elements in the calendar are fine.
[image]

peterHoburg, 2022-03-18:
I own a Kudu6, and I am having this issue. 58582 is my ticket, and I thought I could chime in. 

When running `cat /sys/power/mem_sleep` I get `[s2idle]`. Yes, that is the entire output. 

I ended up following the intel guide [found here](https://01.org/blogs/qwang59/2018/how-achieve-s0ix-states-linux) to confirm that the kudu6 IS `S0ix`. 

I would be happy to disable the `NVreg_EnableS0ixPowerManagement` flag and rebuild the boot image. Unfortunately, I don't know how to rebuild the image, so some instruction would be really helpful, or just point me to the makefile/docs.

cwsystem76, 2022-03-18:
@peterHoburg I've responded in the ticket conversation.

For anyone else reading this issue, rebuilding the boot image is just a matter of running `sudo update-initramfs -c -k all` after making the manual changes to the config files, and then rebooting.

cwsystem76, 2022-03-21:
@peterHoburg Thank you for sharing that information. We'll continue to work on this via the service ticket as well.

cwsystem76, 2022-03-23:
New Nvidia 510 series driver is out: https://www.nvidia.com/download/driverResults.aspx/187162/en-us

From the v510.60.02 release notes:

> Fixed a regression that could cause OpenGL applications to hang or render incorrectly after suspend/resume cycles or VT-switches 

This is not yet in the packaging system so any immediate testing will require using the Nvidia installer, which we don't recommend to end users.

cwsystem76, 2022-03-23:
Initial testing on an oryp6 with the unpackaged Nvidia drivers has v510.60.02 performing identically to the previous revision v510.54. The same fix to `system76-power.conf` works as well. These new drivers may have a fix for S0iX systems but we'll need to wait until we have proper packaging to test safely.

cwsystem76, 2022-03-31:
Additional testing with packaged v510.60.02 drivers from a staging repo (https://github.com/pop-os/nvidia-graphics-drivers/tree/nvidia-510.60.02) show exactly the same problem on on _oryp6_. Going to look into having someone test on an S0iX system using the same staging repo.

cwsystem76, 2022-04-05:
Additional testing shows that the GNOME corruption still occurs, but only after a significantly longer suspend time (maybe 30m, possibly less). Back to the drawing board...

cwsystem76, 2022-04-14:
Initial testing shows that the corruption after suspend+resume still occurs with the Pop!OS v22.04 beta and driver 510.60.02. Continuing testing.

SUPERCILEX, 2022-04-29:
FYI I'm on an XPS unencrypted and the issue still occurs.

thor314, 2022-06-03:
Running a System76 Oryx pro w/ PopOS in nvidia mode also produces this issue on resume from suspend.

cwsystem76, 2022-06-04:
Initial testing suggests that the v515 Nvidia drivers _may_ resolve this issue, at least on an S3 suspend mode laptop like an _oryp6_.

We do not recommend downloading the drivers directly from the Nvidia site, as they are not packaged specifically for the OS, will be harder to uninstall if something goes wrong, and may have other unintended effects. We have a testing repository with pre-release packaging, and folks who want to test this for themselves can do so with the following instructions. Please only do so _at your own risk_ on non-critical systems, as these drivers have not been extensively tested.

Run these commands in the terminal, one at a time:
```
cd ~/Downloads
git clone https://github.com/pop-os/pop
sudo ./pop/scripts/apt add nvidia-515.48.07
sudo apt update
sudo apt purge ~nnvidia
sudo apt install nvidia-driver-515
```
If the system is a hybrid graphics laptop, run the following as well:
```
system76-power graphics nvidia
```
Either way, then reboot the system. 

Eventually, the testing repo will be deleted, which will result in `apt` and/or the Pop!Shop complaining that it can't find it, which may prevent system updates from going through. The repository can be most easily removed by opening up the Pop!Shop, clicking on the gear in the upper right to open the "Repoman" tool, and then removing the "Pop Development Branch nvidia-515.48.07" repository. There are additional details about "Repoman" [here, along with screenshots](https://support.system76.com/articles/manage-repos-pop).

shkm, 2022-06-14:
> Initial testing suggests that the v515 Nvidia drivers _may_ resolve this issue, at least on an S3 suspend mode laptop like an _oryp6_.
>
> We do not recommend downloading the drivers directly from the [...]

Thanks for this, @cwsystem76. Just wanted to report that there's no change for me on my (custom) desktop and Wayland. Seems to work for some, as reported on the [Nvidia forums](https://forums.developer.nvidia.com/t/corrupted-graphics-upon-resume-gnome-41-x-org-495-44-driver/194565/19)

cwsystem76, 2022-06-20:
The `nvidia-515.48.07` staging repo was removed over the weekend, possibly because it's going to be going live this week. If you added it and are now getting errors, you can remove the repo via the Pop!Shop as described previously.

cwsystem76, 2022-06-20:
The package `nvidia-driver-515` should be available in the main repos now.

That said, this problem is only partially resolved, it appears: switching users will still cause the problem to come back, although it's a bit different. Some GNOME apps (noteably the Settings app and GNOME terminal) will be all black until GNOME is reset with the Alt-F2 shortcut. Switching TTYs does not _appear_ to trigger the problem.

rstanuwijaya, 2022-07-29:
Any simple way to update from `nvidia-510` driver to `nvidia-515` driver as it is already available officially?

leviport, 2022-07-29:
>Any simple way to update from nvidia-510 driver to nvidia-515 driver as it is already available officially?

Pop-shop > installed. You'll see an install button for it.

Apacelus, 2023-03-05:
So I had the same issue as in the initial comment, on a non-system76 system:
[image]
Driver version: 525.85.05

I **added** `options nvidia NVreg_PreserveVideoMemoryAllocations=1` to `/etc/modprobe.d/system76-power.conf`, rebuilt my initramfs, rebooted and now the system wakes up from sleep without any issues.

System is clean, I did a refresh about 2 months ago.

tavinus, 2023-03-08:
The fix above from @Apacelus worked for me, even though I am on X11 (instead of Wayland).  
  
[image]  

Just had to
```
sudo nano /etc/modprobe.d/system76-power.conf
```
Add this line to the end of the file and save/exit (Ctrl + X)
```
options nvidia NVreg_PreserveVideoMemoryAllocations=1
```
Then rebuild initramfs with
```
sudo update-initramfs -c -k all
```
Then reboot and it should be fixed.  
  
The only problem is that this mod will probably disappear in future updates (or will it not?).
  
No more image corruption after waking up.  
Thanks heaps!  
This was getting me crazy.

serkanerip, 2024-05-28:
I'm having the same issue see the screenshots. I'm going to try @Apacelus's fix but wanted to share that the issue still remains. 

OS: Pop!_OS 22.04 LTS
GNOME: 42.9
Windowing System: X11
Nvidia Driver: 550.67

[image]
[image]

panyann, 2024-07-09:
@tavinus @Apacelus  Thanks, it worked!
Hopefully Nvidia could fix that or add this option to be enabled by default..? I saw that this error is known for at least 4 years already...
https://bugs.launchpad.net/ubuntu/+source/mutter/+bug/1876632

wengjincheng, 2024-11-14:
> @tavinus @Apacelus Thanks, the workaround worked on my Ubuntu! Hopefully Nvidia could fix that or add this option to be enabled by default..? I saw that this error is known for at least 4 years already... https://bugs.launchpad.net/ubuntu/+source/mutter/+bug/1876632

Hi bro, the ubuntu system, I can modify which file? because ubuntu do not have system76-power file, sorry for your time

wengjincheng, 2024-11-14:
> @wengjincheng Sorry, I'm not using Ubuntu any more. Try finding where NVreg_PreserveVideoMemoryAllocations should be used. Maybe you should create the file, I don't remember.

Thanks for your reply, did you use pop_os at last?

Skladnayazebra, 2025-03-23:
@wagnerck thank you that [huge comment](https://github.com/pop-os/nvidia-graphics-drivers/issues/133#issuecomment-1071018868)! I was able to solve the problem by enabling S0ix on supported system. It took only adding these options to `/etc/modprobe.d/system76-power.conf` and rebooting the system
```
options nvidia NVreg_EnableS0ixPowerManagement=1
options nvidia NVreg_S0ixPowerManagementVideoMemoryThreshold=1024
```
The post in more detail:
https://www.reddit.com/r/pop_os/comments/1ji92av/til_how_to_fix_the_problem_with_messed_up_ui/

wagnerck, 2025-03-23:
Oh hey, you're welcome. I haven't been at System76 since late 2022 (this account was named "@cwsystem76" at the time so it was more obvious that I was part of their tech support team), but I'm glad that the work I was doing at the time was helpful for you.

Fe3000, 2025-08-24:
> The fix above from [@Apacelus](https://github.com/Apacelus) worked for me, even though I am on X11 (instead of Wayland). Driver version: 525.85.05
> 
> Just had to
> 
> ```
> sudo nano /etc/modprobe.d/system76-power.conf
> ```
> 
> Add this line to the end of the file and save/exit (Ctrl + X)
> 
> ```
> options nvidia NVreg_PreserveVideoMemoryAllocations=1
> ```
> 
> Then rebuild initramfs with
> 
> ```
> sudo update-initramfs -c -k all
> ```
> 
> Then reboot and it should be fixed.

Tried this solution but sleep-mode/suspended-mode stopped working. The computer keeps being awake all the time instead.
Has anyone encountered this too?

Driver Version: 570.172.08

OS Name: Pop! OS 22.04 LTS
OS Type: 64-bitars
GNOME Version: 42.9
Windowing System: X11