pop-os/nvidia-graphics-drivers #135: Nvidia degraded performance after updating to 510.54

angelaraya, 2022-02-23:
System: Pop OS 21.10, kernel 5.15.23-76051523-generic, X.org.
Hardware: Nvidia GTX 1070, Ryzen 7 2700X.

After updating to the latest drivers (via `apt dist-upgrade`) the performace for graphic applications has been degreaded. Most games do not start or run at 5~6FPS. The output of `nvidia-smi` is this:
```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 510.54       Driver Version: 510.54       CUDA Version: 11.6     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:07:00.0  On |                  N/A |
|  0%   61C    P0    41W / 151W |   2246MiB /  8192MiB |     14%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      2602      G   /usr/lib/xorg/Xorg               1452MiB |
|    0   N/A  N/A      3353      G   /usr/bin/gnome-shell              474MiB |
|    0   N/A  N/A      6583      G   x-terminal-emulator                10MiB |
|    0   N/A  N/A      7210      G   ...100260292180194236,131072      121MiB |
|    0   N/A  N/A      7891      G   x-terminal-emulator                10MiB |
|    0   N/A  N/A      8580      G   x-terminal-emulator                10MiB |
|    0   N/A  N/A      9169      G   /usr/bin/nvidia-settings            0MiB |
+-----------------------------------------------------------------------------+
```

If there is anything else I can provide to help diagnose this please let me know.

Apieceofbluesky, 2022-02-23:
Hey, i have quite similar system (1070 ti), and no problems. My output looks very similar to yours:

`+-----------------------------------------------------------------------------+
| NVIDIA-SMI 510.54       Driver Version: 510.54       CUDA Version: 11.6     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0  On |                  N/A |
|  0%   53C    P0    46W / 180W |   2377MiB /  8192MiB |      5%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1979      G   /usr/lib/xorg/Xorg                947MiB |
|    0   N/A  N/A      2487      G   /usr/bin/gnome-shell              561MiB |
|    0   N/A  N/A      3559      G   ...oud-3.4.1-x86_64.AppImage        5MiB |
|    0   N/A  N/A      4589      G   /usr/bin/nvidia-settings            0MiB |
|    0   N/A  N/A      5134      G   gnome-control-center               63MiB |
|    0   N/A  N/A      7917      G   ...llation/ubuntu12_32/steam      134MiB |
|    0   N/A  N/A      7927      G   ./steamwebhelper                   63MiB |
|    0   N/A  N/A      7988      G   ...ef_log.txt --shared-files      155MiB |
|    0   N/A  N/A     25372      G   ...763447011884279745,131072      292MiB |
|    0   N/A  N/A     26497      G   .../debug.log --shared-files       83MiB |
+-----------------------------------------------------------------------------+
`

leviport, 2022-10-17:
Is this still a problem? It seems to me that it could be a duplicate of https://github.com/pop-os/nvidia-graphics-drivers/issues/61

Plus, there are now two other versions of the Nvidia driver available in Pop Shop (470 and 515) that might behave better than 510 on certain hardware.