pop-os/nvidia-graphics-drivers #140: Broken Nvidia 510.54 Driver Update from 2022-02-23 Breaks/Fails Nvidia GTX 760 and Ignores GPU

ganakee, 2022-02-24:


jacobgkau, 2022-02-24:
The 470 -> 510 transitional packaging was removed with https://github.com/pop-os/nvidia-graphics-drivers/pull/136 (the 470 drivers will be available indefinitely under https://github.com/pop-os/nvidia-graphics-drivers-470 now), and work is ongoing in https://github.com/pop-os/shop/pull/330 to allow easier installation/selection of the older drivers.

ganakee, 2022-02-24:
Thank you @jacobgkau . Glad to see work on the issue. It is not clear, however, from the work proceeding whether you can specifically keep a working driver. It looks that way, but in my case, a mere `apt update` created the problems--quite unexpectedly. Unfortunately, I was quickly trying to recover the system at the time rather than neat troubleshooting.

jgreenbow, 2022-02-24:
# Cleaner Mitigation 

## TL;DR
Get to a terminal. If you don't have any graphics, like I did, you can do any of:
a. booting from live USB
b. holding space at boot and selecting Pop!_OS (oldkern) or (old-conf)
c. [booting to recovery](https://support.system76.com/articles/pop-recovery/)

```bash
sudo add-apt-repository -y "deb [arch=amd64,i386] http://apt.pop-os.org/staging/master $(lsb_release -cs) main"
sudo bash -c 'cat > /etc/apt/preferences.d/nvidia-470-override' << EOF
Package: *nvidia*470*
Pin: release o=pop-os-staging-master
Pin-Priority: 1002
EOF
sudo apt purge -y ~nnvidia
sudo apt update && sudo apt install -y nvidia-driver-470
sudo reboot
```

### To Undo
Revert to follow regular Pop_OS releases (once issue is fixed)
```bash
sudo rm archive_uri-http_apt_pop-os_org_staging_master-impish.list 
# or remove corresponding entry from "Extra Sources" list in Pop!_Shop
sudo rm /etc/apt/preferences.d/nvidia-470-override
sudo apt update && sudo apt upgrade
```

## Explanation

- previous drivers (470.86) are still available on Pop_OS staging apt repo
- add staging repo to apt repo list (see: [https://apt.pop-os.org/](https://apt.pop-os.org/))
- add apt preference to install all nvidia-driver-470 related packages from staging at a higher priority than release (1002 vs 1001)
- remove all nvidia packages (`purge` might be overkill, `remove` may be sufficient)
- install the `nvidia-driver-470` package

## Step-by-step

0.
1. add staging repo
`sudo add-apt-repository "deb [arch=amd64,i386] http://apt.pop-os.org/staging/master $(lsb_release -cs) main"`
**or**
a. create new list in `/etc/apt/sources.list.d/` (e.g. `pop-os-staging.list`)
`sudo nano /etc/apt/sources.list.d/pop-os-staging.list`
b. copy the following
```
## X-Repolib-Name: Pop_OS Staging
deb [arch=amd64,i386] http://apt.pop-os.org/staging/master impish main
# deb-src [arch=amd64,i386] http://apt.pop-os.org/staging/master impish main
```

2. add package repo rule in `/etc/apt/preferences.d/` (e.g. `nvidia-470-override`)
`sudo nano /etc/apt/preferences.d/nvidia-470-override`
copy the following
```
Package: *nvidia*470*
Pin: release o=pop-os-staging-master
Pin-Priority: 1002
```
you can confirm that this worked by checking what the install candidate for an example package
`sudo apt-cache policy libnvidia-gl-470` 

3. remove nvidia packages
`sudo apt remove ~nnvidia`
if next steps fail, you can try
`sudo apt purge ~nnvidia`

4. update apt
`sudo apt update`
you can see all nvidia driver packages and their versions with
`apt list |grep -P '^(?!linux).*nvidia\S*470'`

5. install working driver
`sudo apt install nvidia-driver-470`

6. reboot to see if it works

### My situation
My GTX 980 also not working with nvidia-driver-510, totally broken video on update from Pop!_Shop. I was able to boot from bootd "oldconf" and sort things out.

leviport, 2022-02-24:
These updates were released from staging and should be available soon.

0x18katyan, 2022-02-25:
@jgreenbow Followed your instructions and it worked. I had to chroot from the recovery partition and follow your steps up to 5 but the dpkg couldn't configure the later installed 470 packages so I booted using the old kern.conf and reconfigured them. I hadn't fixed a broken Nvidia drivers installation in a long time after switching to Pop_OS! and I don't miss it a bit. Hope the fixes come out soon.

seocat, 2022-02-25:
Hello everyone,
I also have this problem and would like to ask if an official bug fix is to be expected in the next few days or if everyone has to fiddle around with Timeshift and Console?!

kcoderhtml, 2022-02-25:
I have a similar problem with PopShop trying to install Nvidia upgrades but failing because of unmet dependencies.

Error
`The following packages have unmet dependencies:
  libnvidia-fbc1-470: Conflicts: libnvidia-fbc1
                      Conflicts: libnvidia-fbc1
  libnvidia-fbc1-470: Conflicts: libnvidia-fbc1
                      Conflicts: libnvidia-fbc1
  libnvidia-fbc1-510: Conflicts: libnvidia-fbc1
                      Conflicts: libnvidia-fbc1
  libnvidia-fbc1-510: Conflicts: libnvidia-fbc1
                      Conflicts: libnvidia-fbc1
  libnvidia-gl-470: Conflicts: libnvidia-gl
                    Conflicts: libnvidia-gl
  libnvidia-gl-470: Conflicts: libnvidia-gl
                    Conflicts: libnvidia-gl
  libnvidia-gl-510: Conflicts: libnvidia-gl
                    Conflicts: libnvidia-gl
  libnvidia-gl-510: Conflicts: libnvidia-gl
                    Conflicts: libnvidia-gl
  libnvidia-extra-470: Conflicts: libnvidia-extra
  libnvidia-extra-510: Conflicts: libnvidia-extra
  nvidia-compute-utils-470: Conflicts: nvidia-compute-utils
                            Conflicts: nvidia-persistenced
  nvidia-compute-utils-510: Conflicts: nvidia-compute-utils
                            Conflicts: nvidia-persistenced
  nvidia-dkms-470: Conflicts: nvidia-dkms-kernel
  nvidia-dkms-510: Conflicts: nvidia-dkms-kernel
  libnvidia-encode-470: Conflicts: libnvidia-encode
                        Conflicts: libnvidia-encode
  libnvidia-encode-470: Conflicts: libnvidia-encode
                        Conflicts: libnvidia-encode
  libnvidia-encode-510: Conflicts: libnvidia-encode
                        Conflicts: libnvidia-encode
  libnvidia-encode-510: Conflicts: libnvidia-encode
                        Conflicts: libnvidia-encode
  nvidia-utils-470: Conflicts: nvidia-smi
                    Conflicts: nvidia-utils
  nvidia-utils-510: Conflicts: nvidia-smi
                    Conflicts: nvidia-utils
  xserver-xorg-video-nvidia-470: Conflicts: nvidia-driver-binary
                                 Conflicts: xorg-driver-binary
  xserver-xorg-video-nvidia-510: Conflicts: nvidia-driver-binary
                                 Conflicts: xorg-driver-binary
  libnvidia-decode-470: Conflicts: libnvidia-decode
                        Conflicts: libnvidia-decode
  libnvidia-decode-470: Conflicts: libnvidia-decode
                        Conflicts: libnvidia-decode
  libnvidia-decode-510: Conflicts: libnvidia-decode
                        Conflicts: libnvidia-decode
  libnvidia-decode-510: Conflicts: libnvidia-decode
                        Conflicts: libnvidia-decode
  nvidia-kernel-common-470: Conflicts: nvidia-kernel-common
  nvidia-kernel-common-510: Conflicts: nvidia-kernel-common
  libnvidia-cfg1-470: Conflicts: libnvidia-cfg1-any
  libnvidia-cfg1-510: Conflicts: libnvidia-cfg1-any
  nvidia-kernel-source-470: Conflicts: nvidia-kernel-source
  nvidia-kernel-source-510: Conflicts: nvidia-kernel-source
  libnvidia-compute-470: Conflicts: libcuda-10.0-1
                         Conflicts: libcuda-10.1-1
                         Conflicts: libcuda-10.2-1
                         Conflicts: libcuda-11.0-1
                         Conflicts: libcuda-11.1-1
                         Conflicts: libcuda-11.2-1
                         Conflicts: libcuda-5.0-1
                         Conflicts: libcuda-5.5-1
                         Conflicts: libcuda-6.0-1
                         Conflicts: libcuda-6.5-1
                         Conflicts: libcuda-7.0-1
                         Conflicts: libcuda-7.5-1
[... 184 lines cut ...]
 • libnvidia-fbc1-470 NVIDIA OpenGL-based Framebuffer Capture runtime library Version:
   470.86-1pop0~1637635465~21.10~47710c4
 • libnvidia-fbc1-470 NVIDIA OpenGL-based Framebuffer Capture runtime library Version:
   470.86-1pop0~1637635465~21.10~47710c4
 • libnvidia-gl-470 NVIDIA OpenGL/GLX/EGL/GLES GLVND libraries and Vulkan ICD Version:
   470.86-1pop0~1637635465~21.10~47710c4
 • libnvidia-gl-470 NVIDIA OpenGL/GLX/EGL/GLES GLVND libraries and Vulkan ICD Version:
   470.86-1pop0~1637635465~21.10~47710c4
 • libnvidia-ifr1-470 NVIDIA OpenGL-based Inband Frame Readback runtime library Version:
   470.86-1pop0~1637635465~21.10~47710c4
 • libnvidia-ifr1-470 NVIDIA OpenGL-based Inband Frame Readback runtime library Version:
   470.86-1pop0~1637635465~21.10~47710c4
 • nvidia-compute-utils-470 NVIDIA compute utilities Version:
   470.86-1pop0~1637635465~21.10~47710c4
 • nvidia-dkms-470 NVIDIA DKMS package Version: 470.86-1pop0~1637635465~21.10~47710c4
 • nvidia-driver-460 Transitional package for nvidia-driver-470 Version:
   470.86-1pop0~1637635465~21.10~47710c4
 • nvidia-driver-470 NVIDIA driver metapackage Version: 470.86-1pop0~1637635465~21.10~47710c4
 • nvidia-kernel-common-470 Shared files used with the kernel module Version:
   470.86-1pop0~1637635465~21.10~47710c4
 • nvidia-kernel-source-470 NVIDIA kernel source package Version:
   470.86-1pop0~1637635465~21.10~47710c4
 • nvidia-utils-470 NVIDIA driver support binaries Version: 470.86-1pop0~1637635465~21.10~47710c4
 • xserver-xorg-video-nvidia-470 NVIDIA binary Xorg driver Version:
   470.86-1pop0~1637635465~21.10~47710c4`

FilBot3, 2022-02-25:
I'm having other issues. For some reason, the 510 Nvidia driver is downloaded, but not installing.

```bash
❯ sudo apt upgrade --fix-broken
[sudo] password for filbot: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  libnvidia-cfg1-510 libnvidia-compute-510 libnvidia-compute-510:i386 libnvidia-decode-510 libnvidia-decode-510:i386 libnvidia-encode-510 libnvidia-encode-510:i386 libnvidia-extra-510 libnvidia-fbc1-510 libnvidia-fbc1-510:i386
  libnvidia-gl-510 libnvidia-gl-510:i386 nvidia-compute-utils-510 nvidia-driver-510 nvidia-kernel-common-510 nvidia-kernel-source-510 nvidia-utils-510 xserver-xorg-video-nvidia-510
0 upgraded, 0 newly installed, 0 to remove and 18 not upgraded.
```

Pop Shop is giving me some other errors too. Expand details for more.

<details>

```bash
The following packages have unmet dependencies:
  libnvidia-common-470: Conflicts: libnvidia-common
  libnvidia-common-510: Conflicts: libnvidia-common
  libnvidia-fbc1-470: Conflicts: libnvidia-fbc1
                      Conflicts: libnvidia-fbc1
  libnvidia-fbc1-470: Conflicts: libnvidia-fbc1
                      Conflicts: libnvidia-fbc1
  libnvidia-fbc1-510: Conflicts: libnvidia-fbc1
                      Conflicts: libnvidia-fbc1
  libnvidia-fbc1-510: Conflicts: libnvidia-fbc1
                      Conflicts: libnvidia-fbc1
  libnvidia-gl-470: Conflicts: libnvidia-gl
                    Conflicts: libnvidia-gl
  libnvidia-gl-470: Conflicts: libnvidia-gl
                    Conflicts: libnvidia-gl
  libnvidia-gl-510: Conflicts: libnvidia-gl
                    Conflicts: libnvidia-gl
  libnvidia-gl-510: Conflicts: libnvidia-gl
                    Conflicts: libnvidia-gl
  libnvidia-extra-470: Conflicts: libnvidia-extra
  libnvidia-extra-510: Conflicts: libnvidia-extra
  nvidia-compute-utils-470: Conflicts: nvidia-compute-utils
                            Conflicts: nvidia-persistenced
  nvidia-compute-utils-510: Conflicts: nvidia-compute-utils
                            Conflicts: nvidia-persistenced
  nvidia-dkms-470: Conflicts: nvidia-dkms-kernel
  nvidia-dkms-510: Conflicts: nvidia-dkms-kernel
  libnvidia-encode-470: Conflicts: libnvidia-encode
                        Conflicts: libnvidia-encode
  libnvidia-encode-470: Conflicts: libnvidia-encode
                        Conflicts: libnvidia-encode
  libnvidia-encode-510: Conflicts: libnvidia-encode
                        Conflicts: libnvidia-encode
  libnvidia-encode-510: Conflicts: libnvidia-encode
                        Conflicts: libnvidia-encode
  nvidia-utils-470: Conflicts: nvidia-smi
                    Conflicts: nvidia-utils
  nvidia-utils-510: Conflicts: nvidia-smi
                    Conflicts: nvidia-utils
  xserver-xorg-video-nvidia-470: Conflicts: nvidia-driver-binary
                                 Conflicts: xorg-driver-binary
  xserver-xorg-video-nvidia-510: Conflicts: nvidia-driver-binary
                                 Conflicts: xorg-driver-binary
  libnvidia-decode-470: Conflicts: libnvidia-decode
                        Conflicts: libnvidia-decode
  libnvidia-decode-470: Conflicts: libnvidia-decode
                        Conflicts: libnvidia-decode
  libnvidia-decode-510: Conflicts: libnvidia-decode
                        Conflicts: libnvidia-decode
  libnvidia-decode-510: Conflicts: libnvidia-decode
[... 129 lines cut ...]
                         Conflicts: libcuda-5.5-1
                         Conflicts: libcuda-5.5-1
                         Conflicts: libcuda-6.0-1
                         Conflicts: libcuda-6.0-1
                         Conflicts: libcuda-6.5-1
                         Conflicts: libcuda-6.5-1
                         Conflicts: libcuda-7.0-1
                         Conflicts: libcuda-7.0-1
                         Conflicts: libcuda-7.5-1
                         Conflicts: libcuda-7.5-1
                         Conflicts: libcuda-8.0-1
                         Conflicts: libcuda-8.0-1
                         Conflicts: libcuda-9.0-1
                         Conflicts: libcuda-9.0-1
                         Conflicts: libcuda-9.1-1
                         Conflicts: libcuda-9.1-1
                         Conflicts: libcuda-9.2-1
                         Conflicts: libcuda-9.2-1
                         Conflicts: libnvidia-compute
                         Conflicts: libnvidia-compute
                         Conflicts: nvidia-opencl-icd
                         Conflicts: nvidia-opencl-icd
```

</details>

leviport, 2022-02-25:
It looks like the 470 transitional sometimes isn't getting removed properly. This can be fixed with:
```
sudo apt remove '*nvidia*470*'
sudo apt update && sudo apt upgrade
```

@seocat the updates have been released and the 470 driver is now available to install.

Akpoby359, 2022-02-25:
Thank you so much.

kcoderhtml, 2022-02-25:
@leviport  Thank you so much! It worked for me very well.

ganakee, 2022-02-25:
@leviport Thank you. Does the mention of the updates being released mean that we can UNHOLD the packages and safely do updates via apt or POP SHOP? Thanks for the work. I just want to clarify. What I mean here is, can we UNHOLD (assuming the workaround above) and safely do standard updates and upgrades? (sorry for being so precise, but on my system, when broken, this is a real pain to fix and what to avoid that again. :-) )

leviport, 2022-02-25:
@ganakee I didn't read that whole procedure closely, but yes that should be fine.

seocat, 2022-02-25:
@leviport Thanks, will check this at home 👍🏻

FilBot3, 2022-02-25:
> It looks like the 470 transitional sometimes isn't getting removed properly. This can be fixed with:
> 
> ```
> sudo apt remove '*nvidia*470*'
> sudo apt update && sudo apt upgrade
> ```
> 
> @seocat the updates have been released and the 470 driver is now available to install.

This did the thing. Thank you!

ganakee, 2022-02-26:
Thank you. I am not an expert on Nvidia settings. However, there may be an issue with merely running 

```
sudo apt remove '*nvidia*470*'
sudo apt update && sudo apt upgrade
```
This removes, for me, all NVidia modules (as it should according to the command). But there is no subsequent re-install of the now missing 470 drivers.

1. I ran a manual backup via Timeshift.
2. Then I marked all packages temporarily placed on HOLD to UNHOLD. (See above)
3. Then I ran sudo apt remove '*nvidia*470*'
sudo apt update && sudo apt upgrade
4. I rebooted.
5. For some odd reason, the displays were recognized, although reversed, but NIGHTLIGHT is set to on. However, at this point, running nvidia-settings or nvidia-smi fails because the sudo apt remove '*nvidia*470*' sudo apt update && sudo apt upgrade apparently removes all Nvidia modules and drivers.
6. For my system (NVidia GTX760), I also needed to run sudo apt install nvidia-driver-470 to restore the NVidia driver and card settings.
7. Then I rebooted again.
8. Again, on boot the NIGHTLIGHT came on, but now the NVidia 470 drivers appear to be used--nvidia-settings and nvidia-smi report the NVidia Driver and card.
9. pop-os:~$ nvidia-smi
Sat Feb 26 15:23:41 2022       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 470.86       Driver Version: 470.86       CUDA Version: 11.4     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0 N/A |                  N/A |
| 40%   41C    P8    N/A /  N/A |    362MiB /  1991MiB |     N/A      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|  No running processes found                                                 |
+-----------------------------------------------------------------------------+
Thanks for the suggestions and working on this issue.

andirsun, 2022-02-27:
```
❯ sudo apt upgrade
[sudo] password for andirsun: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  libnvidia-cfg1-510 libnvidia-common-510 libnvidia-compute-510
  libnvidia-compute-510:i386 libnvidia-decode-510 libnvidia-decode-510:i386
  libnvidia-encode-510 libnvidia-encode-510:i386 libnvidia-extra-510
  libnvidia-fbc1-510 libnvidia-fbc1-510:i386 libnvidia-gl-510
  libnvidia-gl-510:i386 nvidia-compute-utils-510 nvidia-driver-510
  nvidia-kernel-source-510 nvidia-utils-510 xserver-xorg-video-nvidia-510
0 upgraded, 0 newly installed, 0 to remove and 18 not upgraded.
``` 
I have last 510 driver. 
[image]

seocat, 2022-02-28:
Hello,

for me it worked:

1. sudo apt remove '*nvidia*470*'
2. sudo apt update && sudo apt upgrade
3. Reboot
4. "Nvidia 470" driver re-install via Pop-Shop
5. Reboot
6. All fine like before

Thank you all for helping!

leviport, 2022-03-03:
With the 470 driver made available once more, I think this issue can be closed.

closed 2022-03-03 by leviport

Frikster, 2022-03-10:
> for me it worked:
> 
> 1. `sudo apt remove '*nvidia*470*'`
> 2. `sudo apt update && sudo apt upgrade`
> 3. Reboot
> 4. "Nvidia 470" driver re-install via Pop-Shop
> 5. Reboot
> 6. All fine like before
> 
> Thank you all for helping!

This worked for me while just doing step 1+2 did not work