pop-os/nvidia-graphics-drivers #142: 390 drivers don't complie with 5.6.11 kernel in 21.10

vajonam, 2022-03-08:
Looks like some patches are missing. 

https://launchpad.net/ubuntu/+source/nvidia-graphics-drivers-390. This breaks on the update of the kernel to 5.16.11. 

however downloading the latest binary package and manually install works, so looks like the current binary has already been patched for 5.16.11.

AllanRPetersen, 2022-03-13:
I have the same problem. Can't install nvidia-driver-390. I always get an error and then when I reboot the screen is blank.

@vajonam how did you get it to work with manual install?

vajonam, 2022-03-13:
> I have the same problem. Can't install nvidia-driver-390. I always get an error and then when I reboot the screen is blank.
> 
> 
> 
> @vajonam how did you get it to work with manual install? Is it because Pop OS tries to install 390.144 instead of 390.147?

Uninstall any nvidia-drivers-390 via apt

Download latest 390 driver from nvidia and Chmod and run it. It should compile and install the driver

AllanRPetersen, 2022-03-13:
Do I need to turn off the nouveau driver as well?

I was hoping the NVIDIA Pop OS image would have these drivers but I think it only has never drivers.

vajonam, 2022-03-13:
Yes. Add the nouveau drivers to the mod probe black list and then run a update-initrd 

The 390 drivers used to work prior to this latest kernel. Nvidia has updated the 390 driver to compile against 5.6.11 pop hasn't updated the driver package to include those changes.

AllanRPetersen, 2022-03-13:
I am really struggling to figure out how to do this. I tried installing the 418-server driver as well. No luck yet 😪 

Would this work (using 390.147 driver):
https://forum.zorin.com/t/how-to-install-the-latest-nvidia-driver-on-zorin-os-16-pop-os-21-04/10109

vajonam, 2022-03-13:
Yes. Those steps will work.

AllanRPetersen, 2022-03-13:
That doesn't work. Once I get to step 8 and reboot I just get a blank screen. It is like there is no drivers present so it just displays nothing. I think I will cut my losses and use the Nvidia 470 driver even though it doesn't support my GT650M gpu 🙄

vajonam, 2022-03-13:
Probably best if your aren't familiar with messing with nvidia drivers.

AllanRPetersen, 2022-03-18:
Is this the fix here:
https://www.mail-archive.com/kernel-packages@lists.launchpad.net/msg472784.html

vajonam, 2022-03-18:
> Is this the fix here: https://www.mail-archive.com/kernel-packages@lists.launchpad.net/msg472784.html

Yes this is the same info from the original bug. That is where I think the fix is (I think)

avocadojj, 2022-03-27:
hello same error in my pc. i try those step. but none of those step work. what should i do ? tia

AllanRPetersen, 2022-03-28:
@avocadojj  I can't help you sadly. I wasn't able to get it to work. So now I am using the nvidia 470 driver even though nvidia says it doesn't support my GPU.

vajonam, 2022-03-28:
- Remove all nvidia driver packages from apt, 
- download the latest 340 driver from nvidia, 
- include nouveau drivers in modprobe blacklist and `update-initramfs -u -k all`
- install 340 drivers by executing the installer package.

AllanRPetersen, 2022-03-29:
Why intall nvidia 340 and not nvidia 390?

avocadojj, 2022-03-29:
@AllanRPetersen Thanks for your info, @vajonam i'll try later when my pop got nvidia problem again.

in the end, i reinstall pop and do update. But i never install driver 390 from pop shop. i think that make my driver don't work

AllanRPetersen, 2022-03-29:
@avocadojj  This should work:
https://www.youtube.com/watch?app=desktop&v=I-txHcFgHrw

I think my problem was that I didn't patch the nvidia 390.144 driver so it couldn't compile with the 5.16 kernel.
I should have done this first:

https://www.youtube.com/watch?v=9m972Op8BWU

vajonam, 2022-03-29:
> I think my problem was that I didn't patch the nvidia 390.144 driver so it couldn't compile with the 5.16 kernel.

The point of downloading the latest diver binaries from nvidia is they are already patched.

roadkell, 2022-03-30:
I've successfully installed the 390.147 driver from NVIDIA website using this guide:
[https://www.if-not-true-then-false.com/2021/debian-ubuntu-linux-mint-nvidia-guide/](https://www.if-not-true-then-false.com/2021/debian-ubuntu-linux-mint-nvidia-guide/)
My system: Lenovo ThinkPad T430, NVIDIA NVS 5400M, Pop!_OS 21.10 (5.16.11-76051611-generic)

AllanRPetersen, 2022-04-01:
> I've successfully installed the 390.147 driver from NVIDIA website using this guide: https://www.if-not-true-then-false.com/2021/debian-ubuntu-linux-mint-nvidia-guide/ My system: Lenovo ThinkPad T430, NVIDIA NVS 5400M, Pop!_OS 21.10 (5.16.11-76051611-generic)

Once I get to part 2.5 I get this error:

The following packages have unmet dependencies:
 pop-desktop : Depends: brltty but it is not going to be installed
               Recommends: io.elementary.sideload but it is not installable
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.

Not sure what to do....

jackpot51, 2022-04-01:
We have not taken on maintenance of the 390 driver, but given this issue, it seems like we should so that we can ensure it works alongside the latest kernel.

rods-paiva, 2022-04-06:
> Once I get to part 2.5 I get this error:
>
> The following packages have unmet dependencies: pop-desktop : Depends: brltty but it is not going to be installed Recommends: io.elementary.sideload but it [...]

This error was due to the dracut package. The correct one is: sudo apt install dracut-core. After this I've successfully installed the 390.147 driver

AllanRPetersen, 2022-04-06:
Thanks! That solved the dracut problem. The next error I get is when I try to update grub2.

And when trying to install the Linux it says noveau isn't black listed. So somehow I must have messed that part up as well.

rods-paiva, 2022-04-06:
> Thanks! That solved the dracut problem. The next error I get is when I try to update grub2.
> 
> And when trying to install the Linux it says noveau isn't black listed. So somehow I must have messed that part up as well.

what is the error when updating grub2?

rods-paiva, 2022-04-06:
> Thanks! That solved the dracut problem. The next error I get is when I try to update grub2.
> 
> And when trying to install the Linux it says noveau isn't black listed. So somehow I must have messed that part up as well.

What's the error when updating grub2 ?

roadkell, 2022-04-06:
@AllanRPetersen I've skipped installing `dracut`, as Pop_OS uses `initramfs-tools` instead.
Also, step 2.6 from the guide is probably unnecessary, as `.run` installer seems to be able to disable `nouveau` automatically.

AllanRPetersen, 2022-04-06:
If I remember correctly I can't update grub2 because it isn't there. Then it suggests installing grub2.

If I don't do part 2.6 then the nvidia installers says that the nouveau driver can be disabled by the installer. But after doing this the installer still won't install.

Could it be because I am trying to do this on an old macbook pro 2012?

roadkell, 2022-04-08:
Forgot to mention: I've launched the `.run` installer with `--no-cc-version-check` option, otherwise the compiler mismatch prevents further installation:
```
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Ubuntu 11.2.0-7ubuntu2) 11.2.0
  You are using:           cc (Ubuntu 11.2.0-7ubuntu2) 11.2.0
```
Also, I've skipped running `nvidia-xconfig` in the end: it is safer to edit `xorg.conf` manually later.

AllanRPetersen, 2022-04-08:
Thanks for all the help! I managed to get it to work somehow :)

flyfishMT, 2022-06-17:
> We have not taken on maintenance of the 390 driver, but given this issue, it seems like we should so that we can ensure it works alongside the latest kernel.

This sounds great, is this planned work? I'm running 22.04 with a 5.17 kernel and have an older graphics card and can't get 390 to install.

04ELY, 2022-06-29:
390.151 highlight says:
Fixed a driver installation failure on Linux kernel 5.17 release candidates, where the NVIDIA kernel module failed to build with error "implicit declaration of function 'PDE'".