pop-os/nvidia-graphics-drivers #146: Not able to install `system76-driver-nvidia` on Ubuntu 22.04

thomas-zimmerman, 2022-04-22:
There are currently dependency issues with installing the 'system76-driver-nvidia` package on Ubuntu 22.04.

If you follow the [steps in the support article](https://support.system76.com/articles/install-ubuntu#apt-preferences-file) and create the apt preference file first, you will not be able to install the Nvidia driver at all with the System76-dev/stable PPA enabled. There is the additional problem that you need to install *either* the `nvidia-driver-510` or the `nvidia-driver-470` driver to get the `system76-driver-nvidia` package to install. 

In testing in a VM, you can take these steps to likely get a Nvidia driver:

* Install Ubuntu 22.04
* Install Nvidia driver : `sudo apt install nvidia-driver-510`
* Add System76 PPA : `sudo apt-add-repository -y ppa:system76-dev/stable`
* Install System76 Packaged Nvidia driver: `sudo apt install system76-driver-nvidia`

This will install the Ubuntu packaged Nvidia driver (510.60.02) but updates will fail with dependency errors. 
[image]

Installing the `system76-driver` or creating the `apt preference` file first will just get dependency errors when attempting to install Nvidia drivers:
* Install Ubuntu 22.04
* Add System76 PPA : `sudo apt-add-repository -y ppa:system76-dev/stable
* Install System76 driver : `sudo apt install system76-driver
* Install Nvidia driver: `sudo apt install system76-driver-nvidia`
Dependency errors:
[image]

thomas-zimmerman, 2022-04-22:
Testing on a `oryp5`, using the Ubuntu packaged `nvidia-driver-510` works. Getting further updates to continue working:

* Install Ubuntu
* Install Nvidia driver (or check the box for third party drivers): `sudo apt install nvidia-driver-510`
* Add the System76 PPA: `sudo apt-add-repository -y ppa:system76-dev/stable`
* Install the `system76-driver` : `sudo apt install system76-driver`
* Remove the apt preference file: `sudo mv /etc/apt/preferences.d/system76-apt-preferences /etc/apt/system76-apt-preferences.backup`

This allows support for firmware and kernel support from System76 but does not break the dependency tree for apt. Once this is fixed you'll either want to move the apt preference file back in place or reinstall the `system76-driver` package to get it back: `sudo apt reinstall system76-driver`

Thoughts on removing the apt preference file before we get the dependency tree fixed for our builds of the Nvidia drivers?

leviport, 2022-04-26:
This should be fixed now, if anyone wants to help confirm.

thomas-zimmerman, 2022-04-26:
I'm still not able to install `nvidia-driver-510` while the apt preference file is in place. We'll also need to install *either* the 510 or the 470 driver when attempting to install the `system76-driver-nvidia` for the first time:
[image]

jackpot51, 2022-04-26:
Working on it now

jackpot51, 2022-04-29:
Fixed a few days ago

closed 2022-04-29 by jackpot51