pop-os/nvidia-graphics-drivers #156: Error processing package, kernel 5.17.5 is not supported for nviidia-dkms

4a454646, 2022-06-14:
I recently upgraded to 22.04 from 21.10 and the nvidia drivers have been completely bricked. I am unable to switch to NVIDIA graphics, and most importantly the package manager is locked up by the broken packages.

Upon trying to install anything, the following message pops up:
```
Setting up nvidia-dkms-510 (510.73.05-1pop0~1653592955~22.04~4ec3405) ...
update-initramfs: deferring update (trigger activated)
INFO:Enable nvidia
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/system76-nvidia-quirks-oryp3-ess
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/dell_latitude
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/lenovo_thinkpad
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/system76-nvidia-quirks-serw10
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/system76-nvidia-quirks-oryp3
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/system76-nvidia-quirks-oryp2-ess
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/system76-nvidia-quirks-oryp3-b
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/put_your_quirks_here
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/system76-nvidia-quirks-bonw11
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/system76-nvidia-quirks-thelio-massive-b1
DEBUG:Parsing /usr/share/ubuntu-drivers-common/quirks/system76-nvidia-quirks-oryp2
Removing old nvidia-510.73.05 DKMS files...
Deleting module nvidia-510.73.05 completely from the DKMS tree.
Loading new nvidia-510.73.05 DKMS files...
Building for 5.17.5-76051705-generic
Building for architecture x86_64
Building initial module for 5.17.5-76051705-generic
ERROR (dkms apport): kernel package linux-headers-5.17.5-76051705-generic is not supported
Error! Bad return status for module build on kernel: 5.17.5-76051705-generic (x86_64)
Consult /var/lib/dkms/nvidia/510.73.05/build/make.log for more information.
dpkg: error processing package nvidia-dkms-510 (--configure):
 installed nvidia-dkms-510 package post-installation script subprocess returned error exit status 10
Setting up system76-dkms (1.0.14~1643391291~22.04~78ede46) ...
Removing old system76-1.0.14~1643391291~22.04~78ede46 DKMS files...
Deleting module system76-1.0.14~1643391291~22.04~78ede46 completely from the DKMS tree.
Loading new system76-1.0.14~1643391291~22.04~78ede46 DKMS files...
Building for 5.17.5-76051705-generic
Building initial module for 5.17.5-76051705-generic
ERROR (dkms apport): kernel package linux-headers-5.17.5-76051705-generic is not supported
Error! Bad return status for module build on kernel: 5.17.5-76051705-generic (x86_64)
Consult /var/lib/dkms/system76/1.0.14~1643391291~22.04~78ede46/build/make.log for more information.
dpkg: error processing package system76-dkms (--configure):
 installed system76-dkms package post-installation script subprocess returned error exit status 10
Setting up system76-io-dkms (1.0.1~1616424744~22.04~57c15ff) ...
Removing old system76-io-1.0.1~1616424744~22.04~57c15ff DKMS files...
Deleting module system76-io-1.0.1~1616424744~22.04~57c15ff completely from the DKMS tree.
Loading new system76-io-1.0.1~1616424744~22.04~57c15ff DKMS files...
Building for 5.17.5-76051705-generic
Building initial module for 5.17.5-76051705-generic
ERROR (dkms apport): kernel package linux-headers-5.17.5-76051705-generic is not supported
Error! Bad return status for module build on kernel: 5.17.5-76051705-generic (x86_64)
Consult /var/lib/dkms/system76-io/1.0.1~1616424744~22.04~57c15ff/build/make.log for more information.
dpkg: error processing package system76-io-dkms (--configure):
 installed system76-io-dkms package post-installation script subprocess returned error exit status 10
dpkg: dependency problems prevent configuration of system76-driver:
 system76-driver depends on system76-dkms; however:
  Package system76-dkms is not configured yet.
 system76-driver depends on system76-io-dkms; however:
  Package system76-io-dkms is not configured yet.

dpkg: error processing package system76-driver (--configure):
 dependency problems - leaving unconfigured
Setting up system76-acpi-dkms (1.0.2~1640122755~22.04~04145b8) ...
No apport report written because MaxReports is reached already
                                                              Removing old system76_acpi-1.0.2~1640122755~22.04~04145b8 DKMS files...
Deleting module system76_acpi-1.0.2~1640122755~22.04~04145b8 completely from the DKMS tree.
Loading new system76_acpi-1.0.2~1640122755~22.04~04145b8 DKMS files...
Building for 5.17.5-76051705-generic
Building initial module for 5.17.5-76051705-generic
ERROR (dkms apport): kernel package linux-headers-5.17.5-76051705-generic is not supported
Error! Bad return status for module build on kernel: 5.17.5-76051705-generic (x86_64)
Consult /var/lib/dkms/system76_acpi/1.0.2~1640122755~22.04~04145b8/build/make.log for more information.
dpkg: error processing package system76-acpi-dkms (--configure):
[... 61 lines cut ...]
 system76-io-dkms
 system76-driver
 system76-acpi-dkms
 nvidia-driver-510
 system76-driver-nvidia
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
This happens no matter the package, no matter the situation, and only disappears upon removing the faulty packages.
I have tried reinstalling the driver, running the usual 
```sudo apt clean
sudo apt update -m
sudo dpkg --configure -a
sudo apt install -f
sudo apt full-upgrade
sudo apt autoremove --purge```,
installing a lower driver level ([470 seemed to work for this user here](https://github.com/pop-os/pop/issues/2367), but not for me), however none of this helped.

I, having experienced issues with upgrading in the past, backed up all my apt installs from the previous version via `apt list --installed` and tried installing the previous version's packages back on, but no luck.

Notably, the current kernel (`5.17.5-76051705-generic`) did not change with the upgrade; it stayed constant between 21.10 and 22.04, which adds to my confusion, since the error message is stating that the 5.17.5xx kernel is not supported—yet I had been using 5.17.5xx with 21.10 for over half a year without too much issue. 
I tried booting into an older kernel, 5.16.xx, however this brings me to a TTY terminal and prompts me to run `journcalctl -xb`, which brings up hundreds of cryptic lines, I can retrieve them if needed.

Some additional info: longtime Pop user, currently running a Lenovo Legion with Ryzen 5800h and NVIDIA RTX 3060. Have had issues when upgrading before, I hoped that it would be better now and wanted to stick to an LTS (22.04) but unfortunately things didn't go as planned. 

I can provide any additional information if requested.

4a454646, 2022-06-14:
Note that I am struggling with the solution presented in [nvidia470#6](https://github.com/pop-os/nvidia-graphics-drivers-470/pull/6).
I am running the script as an sh, however trying to use `add <repo>` gives me the error `gpg: can't open '/home/scripts/.iso.asc': No such file or directory`. I feel like I'm missing something pretty obvious for this solution but am still extremely confused as to how I'm supposed to use the script.

4a454646, 2022-06-14:
Lost about 12 hours to this with still no progress. I got the script working by commenting out the line which added a GPG key, and the repos seem to have been added.
I used the script to add the relevant staging repositories, but none seemed to work. I've since tried switching to an older kernel (5.15xx), switching to a newer kernel (5.18.2), purging older kernels, and updating the repositories all the while, but nothing seems to work.

In the present state, the apt script has added two repos from staging, `linux-5.18.2` and `nvidia-515.48.07`. I am running kernel 5.18.2 and am trying to install the nvidia graphics of version 515 and 510, neither of which is yielding results. Here is the output:
```
jeff@pop-os:~$ sudo apt install nvidia-driver-515
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libnvidia-cfg1-515 libnvidia-common-515 libnvidia-compute-515 libnvidia-compute-515:i386 libnvidia-decode-515 libnvidia-decode-515:i386 libnvidia-egl-wayland1 libnvidia-encode-515
  libnvidia-encode-515:i386 libnvidia-extra-515 libnvidia-fbc1-515 libnvidia-fbc1-515:i386 libnvidia-gl-515 libnvidia-gl-515:i386 nvidia-compute-utils-515 nvidia-dkms-515
  nvidia-kernel-common-515 nvidia-kernel-source-515 nvidia-settings nvidia-utils-515 xserver-xorg-video-nvidia-515
The following NEW packages will be installed:
  libnvidia-cfg1-515 libnvidia-common-515 libnvidia-compute-515 libnvidia-compute-515:i386 libnvidia-decode-515 libnvidia-decode-515:i386 libnvidia-egl-wayland1 libnvidia-encode-515
  libnvidia-encode-515:i386 libnvidia-extra-515 libnvidia-fbc1-515 libnvidia-fbc1-515:i386 libnvidia-gl-515 libnvidia-gl-515:i386 nvidia-compute-utils-515 nvidia-dkms-515 nvidia-driver-515
  nvidia-kernel-common-515 nvidia-kernel-source-515 nvidia-settings nvidia-utils-515 xserver-xorg-video-nvidia-515
0 upgraded, 22 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/353 MB of archives.
After this operation, 894 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Selecting previously unselected package libnvidia-cfg1-515:amd64.
(Reading database ... 337907 files and directories currently installed.)
Preparing to unpack .../00-libnvidia-cfg1-515_515.48.07-1pop0~1654263250~22.04~dd969cd_amd64.deb ...
Unpacking libnvidia-cfg1-515:amd64 (515.48.07-1pop0~1654263250~22.04~dd969cd) ...
Selecting previously unselected package libnvidia-common-515.
Preparing to unpack .../01-libnvidia-common-515_515.48.07-1pop0~1654263250~22.04~dd969cd_all.deb ...
Unpacking libnvidia-common-515 (515.48.07-1pop0~1654263250~22.04~dd969cd) ...
Selecting previously unselected package libnvidia-compute-515:i386.
Preparing to unpack .../02-libnvidia-compute-515_515.48.07-1pop0~1654263250~22.04~dd969cd_i386.deb ...
Unpacking libnvidia-compute-515:i386 (515.48.07-1pop0~1654263250~22.04~dd969cd) ...
Selecting previously unselected package libnvidia-compute-515:amd64.
Preparing to unpack .../03-libnvidia-compute-515_515.48.07-1pop0~1654263250~22.04~dd969cd_amd64.deb ...
Unpacking libnvidia-compute-515:amd64 (515.48.07-1pop0~1654263250~22.04~dd969cd) ...
Selecting previously unselected package libnvidia-decode-515:i386.
Preparing to unpack .../04-libnvidia-decode-515_515.48.07-1pop0~1654263250~22.04~dd969cd_i386.deb ...
Unpacking libnvidia-decode-515:i386 (515.48.07-1pop0~1654263250~22.04~dd969cd) ...
Selecting previously unselected package libnvidia-decode-515:amd64.
Preparing to unpack .../05-libnvidia-decode-515_515.48.07-1pop0~1654263250~22.04~dd969cd_amd64.deb ...
Unpacking libnvidia-decode-515:amd64 (515.48.07-1pop0~1654263250~22.04~dd969cd) ...
Selecting previously unselected package libnvidia-egl-wayland1:amd64.
Preparing to unpack .../06-libnvidia-egl-wayland1_1%3a1.1.9-1.1_amd64.deb ...
Unpacking libnvidia-egl-wayland1:amd64 (1:1.1.9-1.1) ...
Selecting previously unselected package libnvidia-encode-515:amd64.
Preparing to unpack .../07-libnvidia-encode-515_515.48.07-1pop0~1654263250~22.04~dd969cd_amd64.deb ...
Unpacking libnvidia-encode-515:amd64 (515.48.07-1pop0~1654263250~22.04~dd969cd) ...
Selecting previously unselected package libnvidia-encode-515:i386.
Preparing to unpack .../08-libnvidia-encode-515_515.48.07-1pop0~1654263250~22.04~dd969cd_i386.deb ...
Unpacking libnvidia-encode-515:i386 (515.48.07-1pop0~1654263250~22.04~dd969cd) ...
Selecting previously unselected package libnvidia-extra-515:amd64.
Preparing to unpack .../09-libnvidia-extra-515_515.48.07-1pop0~1654263250~22.04~dd969cd_amd64.deb ...
Unpacking libnvidia-extra-515:amd64 (515.48.07-1pop0~1654263250~22.04~dd969cd) ...
Selecting previously unselected package libnvidia-fbc1-515:i386.
Preparing to unpack .../10-libnvidia-fbc1-515_515.48.07-1pop0~1654263250~22.04~dd969cd_i386.deb ...
Unpacking libnvidia-fbc1-515:i386 (515.48.07-1pop0~1654263250~22.04~dd969cd) ...
Selecting previously unselected package libnvidia-fbc1-515:amd64.
Preparing to unpack .../11-libnvidia-fbc1-515_515.48.07-1pop0~1654263250~22.04~dd969cd_amd64.deb ...
Unpacking libnvidia-fbc1-515:amd64 (515.48.07-1pop0~1654263250~22.04~dd969cd) ...
Selecting previously unselected package libnvidia-gl-515:i386.
Preparing to unpack .../12-libnvidia-gl-515_515.48.07-1pop0~1654263250~22.04~dd969cd_i386.deb ...
dpkg-query: no packages found matching libnvidia-gl-510
Unpacking libnvidia-gl-515:i386 (515.48.07-1pop0~1654263250~22.04~dd969cd) ...
[... cut ...]
make[2]: *** [scripts/Makefile.build:288: /var/lib/dkms/nvidia/515.48.07/build/nvidia/nv-acpi.o] Error 1
make[1]: *** [Makefile:1840: /var/lib/dkms/nvidia/515.48.07/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.18.2-76051802-generic'
make: *** [Makefile:82: modules] Error 2
```

Definitely going crazy, will get back to this tomorrow morning. Am considering doing a clean reinstall of Pop (like I did when the last update failed), but I don't want to go through the hassle of setting up my entire workspace again. Please help. Thanks.

leviport, 2022-06-14:
I've not heard other reports of this, so it may have been a one-off. It may have helped to purge and reinstall Nvidia drivers with these commands:
```
sudo apt purge ~nnvidia
sudo apt autoremove
sudo apt clean
sudo apt install nvidia-driver-510
```
and then rebooting. 

Since this sounds like a one-off (`apt` can get itself into trouble like this from time to time, since it's non-deterministic), I'm going to close this issue. If others encounter the same thing, we can reopen it.

4a454646, 2022-06-14:
I definitely tried purging every nvidia driver off my system + cleaning and autoremoving at least a dozen times with no luck, hopefully anyone else that encounters this is able to follow my steps and fix it.

klaribot, 2022-06-30:
I've encountered this issue twice now, with two separate new installations of Pop!_OS 22.04, upgrading to EITHER nvidia-driver-510 or nvidia-driver-515. My main workstation has been completely out of commission for the past month because of this issue. My AMD Ryzen 5 3600X has no integrated graphics to fall back to.

I love Pop, and I don't want to switch to another distro for a *very* long time just because of one little bug with Nvidia that isn't getting solved.

4a454646, 2022-06-30:
I'm assuming that you tried the solution that worked for me, in a comment above.
Unfortunately if that didn't work, you're just going to have to follow the rabbithole of error messages and hopefully its solvable. In my case, it directed me to disable Nvidia drivers first, then switch cpp versions, then download from the proper distribution site, and only then did it work.

klaribot, 2022-06-30:
I couldn't complete your solution entirely, as the links to the 515 staging repository and the latest tarxz are both broken. (404)

4a454646, 2022-06-30:
Not sure why they did that, @leviport why have the 515 drivers been taken off the [staging page](https://apt-origin.pop-os.org/staging/)?

Regardless @klaribot luckily I kept the download around and just [uploaded it to Google Drive](https://drive.google.com/file/d/1AbNqNRxNjU3TeubK9IsJUkbJU-DjLW2f/view?usp=sharing), hopefully this works for you.

leviport, 2022-06-30:
Because 515 was released.

klaribot, 2022-06-30:
....Actually, I think I might've found my problem.

I've been installing the 21.10 ISO (since that's what was on my ventoy usb at the time), and then upgrading to 22.04, and then updated Nvidia, and it breaks.

However, installing the 22.04 ISO directly, seems to have the issue solved.
In that case, the problem must lie in between the in-place upgrade from 21.10 to 22.04.

Regardless, nvidia-driver-515 seems to be installed (and is already working) by default on the 22.04 ISO. I will try to spin this problem up in a QEMU VM and see if this solution still works, regardless.

linusyoung, 2022-08-05:
Got similar error when upgrading.

already removed all nvidia packages but can't fix system76-dkms and system76-dkms.

`Loading new system76-io-1.0.2~1655490480~22.04~0217576 DKMS files...
Building for 5.18.10-76051810-generic
Building initial module for 5.18.10-76051810-generic
ERROR (dkms apport): kernel package linux-headers-5.18.10-76051810-generic is not supported
Error! Bad return status for module build on kernel: 5.18.10-76051810-generic (x86_64)
Consult /var/lib/dkms/system76-io/1.0.2~1655490480~22.04~0217576/build/make.log for more information.
dpkg: error processing package system76-io-dkms (--configure):
 installed system76-io-dkms package post-installation script subprocess returned error exit status 10
Processing triggers for initramfs-tools (0.140ubuntu6.1) ...
update-initramfs: Generating /boot/initrd.img-5.18.10-76051810-generic
kernelstub.Config    : INFO     Looking for configuration...
kernelstub           : INFO     System information: 

    OS:..................Pop!_OS 22.04
    Root partition:....../dev/nvme0n1p3
    Root FS UUID:........abd645c9-8556-4933-b76b-8e8eb7fa818a
    ESP Path:............/boot/efi
    ESP Partition:......./dev/nvme0n1p2
    ESP Partition #:.....2
    NVRAM entry #:.......-1
    Boot Variable #:.....0000
    Kernel Boot Options:.quiet loglevel=0 systemd.show_status=false splash
    Kernel Image Path:.../boot/vmlinuz-5.18.10-76051810-generic
    Initrd Image Path:.../boot/initrd.img-5.18.10-76051810-generic
    Force-overwrite:.....False

kernelstub.Installer : INFO     Copying Kernel into ESP
kernelstub.Installer : INFO     Copying initrd.img into ESP
kernelstub.Installer : INFO     Setting up loader.conf configuration
kernelstub.Installer : INFO     Making entry file for Pop!_OS
kernelstub.Installer : INFO     Backing up old kernel
kernelstub.Installer : INFO     No old kernel found, skipping
Errors were encountered while processing:
 system76-dkms
 system76-io-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
`

`  GNU nano 5.6.1                                        /var/lib/dkms/system76-io/1.0.2~1655490480~22.04~0217576/build/make.log                                                  
DKMS make.log for system76-io-1.0.2~1655490480~22.04~0217576 for kernel 5.18.10-76051810-generic (x86_64)
Fri 05 Aug 2022 21:50:15 AEST
make: Entering directory '/usr/src/linux-headers-5.18.10-76051810-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0
  You are using:           gcc (Ubuntu 11.2.0-7ubuntu2) 11.2.0
  CC [M]  /var/lib/dkms/system76-io/1.0.2~1655490480~22.04~0217576/build/system76-io.o
gcc: error: unrecognized command-line option '-mharden-sls=all'
make[1]: *** [scripts/Makefile.build:288: /var/lib/dkms/system76-io/1.0.2~1655490480~22.04~0217576/build/system76-io.o] Error 1
make: *** [Makefile:1844: /var/lib/dkms/system76-io/1.0.2~1655490480~22.04~0217576/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.18.10-76051810-generic'
`