pop-os/nvidia-graphics-drivers #167: Video-Tubes (youtube, freetube,  odysse...) not working

imlinstress, 2022-10-20:
I use latest Pop-os on Desktop with latest Nvidia drivers and on my notebook without Nvidia.

While on the notebook videos run smoothly, i can't watch them on Desktop PC. It wasn't always like this only few weeks a go!

Any suggestions pls?

leviport, 2022-10-20:
Might be worth purging and reinstalling nvidia drivers:
```
sudo apt purge ~nnvidia
sudo apt autoremove && sudo apt clean
sudo apt install nvidia-driver-515
```

And then reboot if there are no errors. If there are errors, share them here.

imlinstress, 2022-10-20:
Unfortunately the problem still persists. An OS refresh failed also to solve the issue. I'm afraid, i'm gonna have to reinstall the whole thing, since some other programs now are throwing errors too.
Thanks anyway for willing to help!

closed 2022-10-20 by imlinstress