pop-os/nvidia-graphics-drivers #169: Issue with GUI after computer wakes from sleep

superbazzooka, 2022-11-07:


superbazzooka, 2022-11-07:
> I'm not seeing any screenshots here. Are you seeing text corruption like what is shown in this issue #133?

Hello, yes it seems that the issue is very similar, although the graphical bug seems to be slightly different. Also the bug doesn't happen 60% of the time but almost never, Steam so far seems to be the only application where this problem happens always and consistently.

n3m0-22, 2022-11-07:
I would check the NVIDIA driver version you are using. You can do this by running the command `nvidia-smi`. If the driver version is not `515` you can update it by running the following:

```
sudo apt purge ~nnvida
sudo apt autoremove && sudo apt clean
sudo apt install nvidia-driver-515
```

If that does not fix it you can try running the same commands substituting `470` for `515` to see if the older driver helps.

Finally you can also try changing the graphics mode. If it is not already change it from NVIDIA to Hybrid.

raneameya, 2022-11-22:
I am also facing the same issue. My drivers are the latest `515` (not the open one). Logging out and logging back in solves it

closed 2022-12-02 by superbazzooka