pop-os/nvidia-graphics-drivers #172: Strange Game Timing Issue then crash in Miles Morales with 525

tfinnegan937, 2022-11-29:
When playing Spider-Man Miles Morales with Nvidia driver 525 installed, the game runs at full speed, but the animation in the background of the main menu will start and stop randomly (Independent of the actual game speed). After a minute or two of this, the game crashes altogether. This does not occur on 515

mmstick, 2022-11-29:
Any chance that this was simply shader compilation for a game that's not been run by this driver on Linux yet?

tfinnegan937, 2022-11-29:
Good catch. I usually skip the precompilation and let them compile in the background. I'll give that a try and report back. On 515, it was not crashing when I did so, however.

tfinnegan937, 2022-11-29:
I have completely deleted and reinstalled steam and the game. I allowed for shaders to fully precompile. The game still crashes at startup on the latest drivers. I'm not sure if this is on the driver, or the game, though. Is there more information that I can provide to narrow this down?

n3m0-22, 2022-11-29:
This may be relevant. https://github.com/ValveSoftware/Proton/issues/6328

mmstick, 2022-11-30:
Could be related to https://www.phoronix.com/news/DXVK-NVAPI-0.6