pop-os/nvidia-graphics-drivers #185: RTX 3080 Ti has render lag on everything but mouse pointer.

Ruscios, 2023-07-09:
Just installed Pop!_OS 22.04 on a new machine. Windows install works fine so all hardware is functional, motherboard firmware is fully updated.

System Specs:
CPU: AMD Ryzen 7 7700X
GPU: NVIDIA GeForce RTX 3080 Ti

When connected to my monitor from the motherboard HDMI using the built-in Radeon graphics, I have no issues. If I try to switch over to or boot with the GPU connected, I get the same output, except that any movement/changes/updates/typing/etc other than the mouse pointer render at only about 1 fps. The mouse pointer updates normally. 

I've tried using driver 535 (forgot subversion), driver 470.161.03, and currently driver 525.116.04, all with the same result. Also tried Nouveau, output was garbled and unusable.

Based on looking through other issues, I believe this is related to #61 but all configurations mentioned there relate to laptops which have hybrid graphics, this is a desktop build.

mmstick, 2023-07-09:
That issue doesn't apply to desktops because you connect the video cable directly to the GPU on a desktop. In a laptop, some ports may be connected to the dGPU and others the iGPU so they have to send video output through each other with PRIME or reverse PRIME.

Ruscios, 2023-07-09:
I only bring up that issue, because the behavior mentioned in [this reply](https://github.com/pop-os/nvidia-graphics-drivers/issues/61#issuecomment-920974378) to that issue describes the behavior I'm seeing where everything is choppy aside from the mouse cursor.

leviport, 2023-07-10:
Might be worth checking your BIOS settings. There is often a setting that selects whether to use dGPU or iGPU, and I have seen instances where the "automatic" selection does something weird and causes behavior similar to what you're describing.

Ruscios, 2023-07-10:
Changing BIOS primary GPU did not, but changing Primary GPU in Xorg settings in /usr/share/xorg.conf.d/10-nvidia.conf by adding the line `Option "PrimaryGPU" "yes"` does appear to have solved the issue. Additionally, it seems that the iGPU output is not affected in the same way as the dGPU, as it also still functions normally.

This does solve the issue, but I'm not sure if it's intended functionality. Seems like an Xorg issue though, so I'll close it here.

closed 2023-07-10 by Ruscios