pop-os/nvidia-graphics-drivers #187: Updating to the nvidia drivers 535.86.05 causes Steam to hang on Wayland on new Thelio Mira

benuski, 2023-07-27:
**(1) Issue/Bug Description:**
Updating to the nvidia drivers 535.86.05 causes Steam to hang on Wayland on new Thelio Mira. It starts to load as normal, but right when it gets to the share display dialog (when I'm using -pipewire), or when I get to "Loading user data..." (when I'm not using -pipewire), it will hang and progress no further.

Steam works fine on X11, this is only on Wayland.

**(2) Steps to reproduce (if you know):**
Update to nvidia drivers 535.86.05 (occurs both for the Steam flatpak or the steam deb available through the pop repos)
Launch on Wayland (it doesn't matter if I use the pipewire flag or not)

**(3) Expected behavior:**
Steam launches and I am able to use Steam remote play (this is what I was able to do before this nvidia driver update)

**(4) Distribution (run `cat /etc/os-release`):**
NAME="Pop!_OS"
VERSION="22.04 LTS"
ID=pop
ID_LIKE="ubuntu debian"
PRETTY_NAME="Pop!_OS 22.04 LTS"
VERSION_ID="22.04"
HOME_URL="https://pop.system76.com"
SUPPORT_URL="https://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=jammy
UBUNTU_CODENAME=jammy
LOGO=distributor-logo-pop-os

**(5) Gnome Shell version:**
``` 
gnome-shell:
  Installed: 42.5-0ubuntu1pop1~1675984688~22.04~f574f54
  Candidate: 42.5-0ubuntu1pop1~1675984688~22.04~f574f54
  Version table:
     42.9-0ubuntu2 500
        500 http://apt.pop-os.org/ubuntu jammy-updates/main amd64 Packages
 *** 42.5-0ubuntu1pop1~1675984688~22.04~f574f54 1001
       1001 http://apt.pop-os.org/release jammy/main amd64 Packages
        100 /var/lib/dpkg/status
     42.0-2ubuntu1 500
        500 http://apt.pop-os.org/ubuntu jammy/main amd64 Packages 
```

**(6) Pop Shell version (run `apt policy pop-shell` or provide the latest commit if building locally):**

```
pop-shell:
  Installed: 1.1.0~1682616779~22.04~b5accce
  Candidate: 1.1.0~1682616779~22.04~b5accce
  Version table:
 *** 1.1.0~1682616779~22.04~b5accce 1001
       1001 http://apt.pop-os.org/release jammy/main amd64 Packages
       1001 http://apt.pop-os.org/release jammy/main i386 Packages
        100 /var/lib/dpkg/status
```

**(7) Where was Pop Shell installed from:**
Normal repos

**(8) Monitor Setup (2 x 1080p, 4K, Primary(Horizontal), Secondary(Vertical), etc):**
2x1080p running at 144, 1x1080p vertical (portrait right) @ 60hz

**(9) Other Installed/Enabled Extensions:**
No additional externsions installed

**(10) Other Notes:**
Nvidia 3060 Ti, Thelio Mira purchased directly from System76 this month

github-luciano, 2023-08-14:
I can confirm that installing "libnvidia-egl-wayland1" resolved the issue I was experiencing with Steam not opening in the Wayland environment. Additionally, this fix also improved the performance of games running under XWayland, which had become unplayable for me after updating from version 525 to 535. The problems were resolved after I installed the mentioned package.

mmstick, 2023-10-06:
I'll get this added to our packaging soon, along with backporting a newer version of this library as it resolves some oustanding issues with NVIDIA graphics in COSMIC.

jacobgkau, 2024-12-26:
It looks like libnvidia-egl-wayland1 is currently a depends of libnvidia-gl-560, which is depended on by nvidia-driver-560 and nvidia-driver-560-open.

closed 2024-12-26 by jacobgkau