pop-os/nvidia-graphics-drivers #190: Patch package to enable suspend-then-hibernate

cstrahan, 2023-08-30:
Would it be possible to include the changes I propose in https://github.com/pop-os/systemd/issues/15#issuecomment-1153401200 to get suspend-then-hibernate working out of the box?

cstrahan, 2023-12-22:
Happy Holidays!

Figured I'd give this bump. I think it would be awesome for S76 laptops running Pop!_OS to have working suspend-then-hibernate out-of-the-box.

Regular suspend leaves my laptop battery at nearly 0% charge after sitting around for a day or so. When traveling, it's nice to leave the laptop in my bag and forget about it until I need it. It kinda sucks when I need to take care of something real quick and realize that the laptop is dead... so I've resorted to anxiously pulling my laptop out and charging it immediately wherever I go.

Another point in favor of having this work OOtB: suspend is more susceptible to spurious wake ups than hibernate, which means that my S76 laptop can reach dangerously hot temperatures as it tries to cook itself in my laptop bag. Having STH configured to hibernate after 10 minutes of suspend means that I can catch it early if the laptop spuriously wakes up in my bag in the back of the car, and after that it's almost guaranteed that the laptop will happily sit there consuming zero charge until I open the lid when I'm ready to work with it later.

mmstick, 2023-12-22:
@cstrahan If you submit a PR, our QA team can look at it.

cstrahan, 2024-01-16:
@mmstick Cool, thanks!

There have been some new developments on systemd's side, so I may hold off until I've had time to look into them (to avoid unnecessary churn):

- https://github.com/systemd/systemd/issues/27559
- https://github.com/systemd/systemd/pull/30612