pop-os/nvidia-graphics-drivers #192: `Pop!_Shop` wants to downgrade to `535` to `510`

CanntAim, 2023-10-01:
Hey all,

I have a Wild Dog Pro from 2017 sitting on `20.04` version of `PopOS` and I noticed recently that `Pop!_Shop` application seems to _really_ want me to install `510` despite:

A) `510` has issues. _(Doesn't recognize display and defaults to 800x600 stretched)_
B) I'm already on a more recent version of the drivers that work well for me.

[image]

I'm okay with simply waiting till newer drivers are released and switching to them by purging and installing drivers through `apt` or better yet switching them through _additional drivers_ application but just wanted to note this as it appears to be a bug? Not sure if there is a reason for the downgrade but I have been upgrading drivers through `Pop!_Shop` until now and this is first time that I observed it wanting me to go backward.

Thanks for all the work and great operating system!

jacobgkau, 2023-10-02:
This should probably be transferred to https://github.com/pop-os/shop since it's concerning the Pop!_Shop GUI rather than the NVIDIA drivers, although the GUI's received major updates in 22.04 and those aren't being backported to 20.04 (you should upgrade if possible.)

It is somewhat expected for multiple versions to be presented, see issues such as https://github.com/pop-os/nvidia-graphics-drivers/issues/112. The shop doesn't "really want" you to install it, it's just offering it as an option. Notifications and badges will show up for updates (which have "Update" buttons), but not for drivers (which have "Install" buttons.)

jacobgkau, 2023-10-02:
This issue seems similar to https://github.com/pop-os/shop/issues/417. https://github.com/pop-os/shop/pull/332 improved the UX by also showing the currently-installed version alongside older available versions; but that improvement probably applied to the current version of Pop!_OS at the time (which was 21.10) and not 20.04.