pop-os/nvidia-graphics-drivers #195: nvidia display brightness control broken in 545

Vincent-FA, 2023-11-16:
**Distribution:** 22.04 LTS
 
 
 
**Related Application and/or Package Version:** nvidia-driver-545 `545.29.02-1pop0~1699530170~22.04~77a5d6c`

**Issue/Bug Description:** brightness controls for laptop are no longer working.
`nvidia_0: Failed to write brightness to device: Invalid argument`

**Steps to reproduce (if you know):** just boot and try to change the brightness with the keys or the slider in the preferences

**Expected behavior:** changing display brightness

**Other Notes:** This seems to be a known bug for the 545 driver, there is patch for it: https://github.com/NVIDIA/open-gpu-kernel-modules/issues/573
Maybe you can patch it and update your package quicker than upstream? It's very annoying.
I also did not find a way to downgrade the nvidia packages on this OS, they all seem to be transitional packages to the newest, I'd be happy to do it in the meantime if you have a way. Thanks

Vincent-FA, 2023-11-20:
I enabled the hybrid mode in the BIOS (the one documented as being not supported by linux) and the problem disappeared, while being in the hybrid graphics mode in Pop!. It's a good workaround apparently.

mmstick, 2023-11-20:
@Vincent-FA Hybrid mode in firmware is actually the recommended setting for Linux.

Vincent-FA, 2023-11-20:
Good to know after two years of use of the computer... It's sad that DELL indicates it's not supported in the firmware UI, even in recent updates.

IIIphr, 2023-11-21:
Is it possible to have a patch sooner than nvidia? My laptop only has one GPU and it's dedicated, so I don't I can put it on hybrid mode.

mmstick, 2023-11-21:
Only NVIDIA can release driver fixes for the proprietary driver.

SicktorHass, 2023-11-21:
is there a reason why we dont have the current production branch driver version 535.129.03 provided in pop_os shop?

pojunen, 2023-11-22:
Fixed for 545.29.06?
https://www.nvidia.com/Download/driverResults.aspx/216530/en-us/

    **Fixed a regression that prevented setting backlight brightness levels.**
    Fixed a bug that could lead to UI corruption in nvidia-installer on systems with more than one initramfs file per kernel.
    Fixed a bug that caused games built on the Source 2 engine to hang when running under Xwayland.

IIIphr, 2023-11-23:
I just upgraded to 545.29.02 and it didn't fix it for me. Did it fix the problem for you guys?

SicktorHass, 2023-11-27:
@leviport yep can confiorm, brightness control working fine on this version (HP Omen RTX 2070 Super)

IIIphr, 2023-11-27:
Thanks for taking the time. I've done what you said and pop os told me to restart, so I did. After that, the problem is still not solved for me.
Right now, when I use the slider on pop os shell for brightness, or the keyboard keys for the same function, the actual brightness of display is not changing.

BUT

Before this update, I found out that the `/sys/class/backlight/nvidia_0/brightness` is changing when I change the brightness using the slider or the buttons, however the `/sys/class/backlight/nvidia_0actual_brightness` is NOT and is stuck at 100 (maximum).
Now, both of them change and are the same number, seems great! If only the actual backlight of my display would work xd

IIIphr, 2023-11-27:
Although, do I need to also update these?
```
$ flatpak update
Looking for updates…

        ID                                               Branch  Op  Remote   Download
 1.     org.freedesktop.Platform.GL.nvidia-545-29-06     1.4     i   flathub  < 325.6 MB
 2.     org.freedesktop.Platform.GL32.nvidia-545-29-06   1.4     i   flathub  < 325.6 MB
```

LucasFA, 2023-11-27:
Works for me, toom though I did upgrade from `nvidia-driver-535-server` instead of `nvidia-driver-545` as the backlight worked properly on that version. Purged, autoremoved and installed.

retrodaredevil, 2023-11-28:
> I enabled the hybrid mode in the BIOS (the one documented as being not supported by linux) and the problem disappeared, while being in the hybrid graphics mode in Pop!. It's a good workaround apparently.

Wanted to say that this fix works for me, however one of the reasons I usually kept "Discrete Graphics" checked in the BIOS is because my "Dynamic Graphics" setting sometimes causes me to be unable to reach the login screen when booting up my computer. So for me I have the option of my computer always boots without brightness control, or my computer sometimes boots with brightness control.

> It looks like #196 might fix this. It would be nice if someone with the bug can confirm whether this update resolves it. To test,
> 
> ```
> sudo apt-manage add popdev:nvidia-545.29.06
> sudo apt update && sudo apt upgrade
> ```
> 
> To remove that testing branch (it will begin to 404 once the branch on Github is deleted):
> 
> ```
> sudo apt-manage remove popdev-nvidia-545-29-06
> ```

I tried this and it seems that I'm getting a 404. Did the branch get deleted within the past 24 hours? Does anyone know when this fix is going to get into the default Pop!_OS repos?

mmstick, 2023-11-28:
This branch was merged. It is in `master` now.

leviport, 2023-11-28:
`master` is a staging branch. I will make a release PR to release them out of `master` staging.

leviport, 2023-11-28:
It has been released. You should be seeing the update soon, if not now.

Reminder to anyone who added the `nvidia-545-29-06` staging branch to remove it if it 404's:
```
sudo apt-manage remove popdev-nvidia-545-29-06
```

IIIphr, 2023-11-29:
I can't manage to get it working on my laptop. Do I have to make another issue and describe the problem? Or it can be discussed here?

leviport, 2023-11-29:
Reopened. Please share hardware details.

IIIphr, 2023-11-29:
Thank you so much, I'm sorry.

I've asked about the problem before ([here](https://unix.stackexchange.com/questions/761922/pop-os-brightness-stuck-at-maximum) and [here](https://askubuntu.com/questions/1368249/display-brightness-problem?_gl=1*qssmzk*_ga*MTA4MTQ2NTkxOS4xNzAwNDE2NDEy*_ga_S812YQPLT2*MTcwMTI3Njg4Ny4xOS4xLjE3MDEyNzY4OTYuMC4wLjA.))

I have an ASUS ROG GL503VS, which has Intel Corei7-7700HQ CPU, Nvidia Geforce GTX 1070 GPU (It's the mobile version I think, and the only GPU on my laptop, and it is dedicated of course), 1920x1080 144hz GeSync display (I don't know if it helps)
I'm running Pop OS 22.04, and kernel:
```
$ uname -r
6.5.6-76060506-generic
```

Right now if I run `nvidia-smi`:
```
$ nvidia-smi
Wed Nov 29 20:28:41 2023       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 545.29.06              Driver Version: 545.29.06    CUDA Version: 12.3     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce GTX 1070        Off | 00000000:01:00.0  On |                  N/A |
| N/A   57C    P8              11W / 115W |    472MiB /  8192MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A      2048      G   /usr/lib/xorg/Xorg                          167MiB |
|    0   N/A  N/A      2157      G   /usr/bin/gnome-shell                         34MiB |
|    0   N/A  N/A      4057      G   ...local/share/Steam/ubuntu12_32/steam        2MiB |
|    0   N/A  N/A      4522      G   ...re/Steam/ubuntu12_64/steamwebhelper      118MiB |
|    0   N/A  N/A      4596      G   ...eam/logs/cef_log.txt --shared-files       28MiB |
|    0   N/A  N/A     18539      G   ...,WinRetrieveSuggestionsOnlyOnDemand       23MiB |
|    0   N/A  N/A     21516      G   ...RunOnOsLogin,WebAssemblyTrapHandler       91MiB |
+---------------------------------------------------------------------------------------+
```

With the last update, `/sys/class/backlight/nvidia_0/brightness` and `/sys/class/backlight/nvidia_0/actual-brightness` are changing accordingly, both by using the UI slider and the fn keys. Everything works fine, the keyboard backlight works just like windows, the only problem is that the actual backlight of my display is no changing.

I've tried setting `acpi_backlight=vendor`, or `nvidia.NVreg_EnableBacklightHandler=1`, both using `sudo kernelstud -a` but to no avail. Inside of `/sys/class/backlight`:
```
/sys/class/backlight$ ls -al
total 0
drwxr-xr-x  2 root root 0 Nov 29 15:50 .
drwxr-xr-x 83 root root 0 Nov 29 15:50 ..
lrwxrwxrwx  1 root root 0 Nov 29 20:24 nvidia_0 -> ../../devices/pci0000:00/0000:00:01.0/0000:01:00.0/backlight/nvidia_0
```

I've tried using `xbacklight` but I don't recall on which version of driver. I can try it again if it's necessary. If there are any other information neeeded, I'd happily give it to you.

Oh and also:
```
/sys/class/backlight$ system76-power graphics
nvidia
/sys/class/backlight$ system76-power graphics hybrid
setting graphics to hybrid
daemon returned an error message: "does not have switchable graphics"

```

LucasFA, 2023-11-29:
I think I recall one instance in which I fixed this issue for my laptop with a 1060 by just purging and reinstalling the drivers. It was more necessary in Fedora, but I think I recall fixing it in Pop once. Worth a shot in case you haven't tried. 

Another time I went on a hunt to locate driver files from a previous installation, too.

IIIphr, 2023-11-29:
Yeah I already did that for this update, after I reverted back from the test branch, I purged everything nvidia related and installed them again.

IIIphr, 2023-12-08:
Hi again xd.
I recently found the xorg configuration files and again, trying the solutions that I found on the internet was not successful for me. This is my nvidia configuration file in xorg:
```
/usr/share/X11/xorg.conf.d$ cat 10-nvidia.conf 
Section "OutputClass"
    Identifier "nvidia"
    MatchDriver "nvidia-drm"
    Driver "nvidia"
    Option "AllowEmptyInitialConfiguration"
    ModulePath "/usr/lib/x86_64-linux-gnu/nvidia/xorg"
EndSection
```
I've tried setting `"EnableBrightnessControl=1"` (adding it at the end of the `Option` line) but it didn't work. I don't know if this file needs any other modifications or not, or even if I added the option in the right place. 
Also, there are other files in this directory:
```
/usr/share/X11/xorg.conf.d$ ls -al
total 36
drwxr-xr-x 2 root root 4096 Dec  8 10:37 .
drwxr-xr-x 5 root root 4096 Oct  6 18:16 ..
-rw-r--r-- 1 root root   92 Feb 24  2023 10-amdgpu.conf
-rw-r--r-- 1 root root  206 Dec  8 10:37 10-nvidia.conf
-rw-r--r-- 1 root root  143 Jul 26 19:13 10-pop-dpms.conf
-rw-r--r-- 1 root root 1350 Oct 23 20:00 10-quirks.conf
-rw-r--r-- 1 root root   92 Jul 11  2022 10-radeon.conf
-rw-r--r-- 1 root root 1429 Feb 11  2022 40-libinput.conf
-rw-r--r-- 1 root root 3458 Apr  6  2022 70-wacom.conf
```
But since most other things already recognize nvidia on my device, I think I chose the correct file. But I can't be sure xd
Does anybody here have anything for me to try? I would really appreciate it.

IIIphr, 2024-01-03:
Hi all, sorry for bringin this up again but hopefully it'll be closed this time.

I did some more research and found [this](https://bbs.archlinux.org/viewtopic.php?id=284359) about the igpu thing, and as radiomike mentions, I need the igpu enabled in order to have a properly functioning backlight.

But then I came across [this](https://rog-forum.asus.com/t5/rog-gaming-notebooks/integrated-graphics-and-missing-options-in-bios/td-p/563293) and it seems like I can't enable igpu from bios, I've updated the bios from asus support website and it doesn't have an option, so my guess is that igpu is disabled by manufacturer and I can't do anything about it.

Am I right? I don't know if the think radiomike said about cards is right or not, if it is then It makes sense and we can close this since it isn't related to drivers and stuff anymore :)

leviport, 2024-01-03:
I doubt the iGPU is disabled. I bet the BIOS option was just hidden in favor of using OS-level controls instead. That machine was designed at a time when hybrid graphics was quite a bit more clunky, and it's much improved these days.

IIIphr, 2024-01-03:
Thanks a lot, you've (and also other people here!) helped a lot already :)
I'll try them, and probably update my stack question since this is probably not the proper place for it, sorry for that xd.
I think we can close this issue since my problem is not related, and the original issue is solved :D

leviport, 2024-01-03:
Sounds good, I'll close then. If you want, feel free to hop into Pop chat for further discussion: https://chat.pop-os.org/pop-os/

I'm in there with the same username and avatar.

closed 2024-01-03 by leviport