pop-os/nvidia-graphics-drivers #198: Pop!_Shop showing install button for old driver version under Updates & Installed Software

sampellino, 2023-12-12:
[image]

I have the 545.29.06 NVIDIA drivers installed, but the pop shop is wanting me to install the 470.199.02 driver. There is no way to remove it from the Updates & Installed Software menu.

Version info:
[image]

mmstick, 2023-12-12:
Not a bug. This lists all supported drivers.

sampellino, 2023-12-12:
If that's true, why is it not showing all supported older drivers to choose from? Why only one specific older version?

mmstick, 2023-12-12:
These are the only drivers that we've packaged which remain in the repositories.

sampellino, 2023-12-12:
Ahh, makes sense. Closing issue.

closed 2023-12-12 by sampellino